MiczFlor/RPi-Jukebox-RFID #912: 🐛 | RFID Reader Service dead after a while

jo-me, 2020-04-20:
## Bug

### What I did

Left the working box on over the night doing nothing.

### What happened

On the next morning, the box did not respond to any of the RFID tags.
Enabling "DEBUG_rfid_trigger_play_sh" did not produce any logs, so it seems like the service crashed or stopped working for some reason.
Syslog also did not show any logs when scanning. 

I restarted the "RFID Reader" service via web ui and then it worked again.
Also logs came from the rfid trigger script.

### I expected this to happen

working rfid 

### Further information that might help

.

## Software

### Base image and version

latest buster 

### Branch / Release

master 

### Installscript

buster

## Hardware

### RaspberryPi version

2
### RFID Reader

RC522

### Soundcard

Hifiberry Miniamp

### Other notable hardware

OnoffShim

jo-me, 2020-04-20:
I can reproduce this by setting the second swipe action to "only system commands" and putting a card on the reader without removing it.

The debug.log will show the "Do nothing" logs freuqently but sometimes it hangs for a short time and continues and sometimes the log output just stops without recovering from it. I then have to go and restart the rfid service. through the UI or systemctl command.

In the situation where the log output stops, the systemd service seems to be still running (checked via  `sudo systemctl status phoniebox-rfid-reader`)

I'm using the Reader.py.experimental version for the RC522.

jo-me, 2020-04-20:
So, after debugging around for a bit, I found that the rfid scanning did not hang due to the service or script execution but due to an upstream problem with the pi-rc522 library that is used for the RC522 scanners. In the `Reader.py` it was stuck in the `self.device.wait_for_tag()` call which caused the hangups.

There is an issue over in the pi-rc522 project that is pretty much identical to mine: https://github.com/ondryaso/pi-rc522/issues/48
The proposed fix is a change in that same function and has already been merged to master.

**The issue is that the library version installed by the installer script is not identical to the master branch of the pi-rc522 project.**

I copied the code of the function over from the other project and let the card sit on the reader for over an hour without hangups. They usually occurred within the first minute or so. 

Not sure why other people don't seem to have the same issues with the rc522 reader...

s-martin, 2020-04-20:
On develop branch this is already fixed, please see PR #840.

So there the local copy is dropped and the most recent version of pi-rc522 is used.

We should release version 2.0 (#890) "soon", nevertheless I can't give you a time of arrival.

In the meantime you could either install from develop (which could have other issue) or manually port the changes of #840 to your installation.

s-martin, 2020-04-20:
Ok, just checked and I'm not sure, if packed version on pypi (which we use) is incorporating the latest updates from the GitHub repo.

s-martin, 2020-04-20:
We should check, if we should install from the GitHub repo directly.

MiczFlor, 2020-05-02:
I close this, because the changes were merged.

closed 2020-05-02 by MiczFlor