MiczFlor/RPi-Jukebox-RFID #913: 🚀 | Help - Coders |

Piehti, 2020-04-20:
Hey!

I'm just adding some extra features to my phoniebox and wanted to add some own command.
Where can I find the hooks for the %YOURCOMMAND% variables in rfig_trigger_play.conf.

I'd just like to change some to my favours (lights, videos, pictures... but without MQTT and stuff).

It's specially for video-playback, web-gui showing and stuff.

Best regards
Peter

that-coding-human, 2020-12-13:
@Piehti , you mentioned:

> Ps.: You propably want to add 3 custom fields for custom command on systemlevel in the webinterace in further versions.

What do you mean with "3 custom fields for custom command"? Can you elaborate on it a little more? Looks like you already have a clear idea how this should work.
I am currently thinking about a proposal for how to achieve this as well.

s-martin, 2024-05-10:
For anyone interested in implementing this, you should target the new V3 (future3 branches).

V3 uses RPC commands (see [here](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/develop/documentation/builders/card-database.md)), which could be extended for the desired functionality.