MiczFlor/RPi-Jukebox-RFID #914: 🐛 | Playlist cannot be started a second time if second swipe is set to do nothing

jo-me, 2020-04-20:
## Bug

### What I did

Set second swipe option to "only system commands". 

Swipe a card, 
Wait until playlist finishes.
Swipe card again

### What happened

Nothing, because the system thinks it is a second swipe and carries out the associated action => Nothing.

### I expected this to happen

After a playlist is finished I would expect a swipe of the same card to be recognized as first swipe and start the playlist again. 
As far as I can tell the playlist name is compared for this and thus no action is taken if it is the same as before. So the playlist name should be deleted once it is finished.

### Further information that might help

## Software

### Base image and version

buster

### Branch / Release

master
### Installscript

buster

## Hardware

### RaspberryPi version

Pi2

### RFID Reader

RC522

s-martin, 2020-04-20:
I agree this should be fixed.

jo-me, 2020-04-21:
So my current workaround is to manipulate the LASTPLAYLIST variable if I detect an idling MPC.

 ```
    LASTPLAYLIST=$(cat $PATHDATA/../settings/Latest_Playlist_Played)
    # workaround start
    currentSong=`mpc current`
    if [[ -z "$currentSong" ]]
    then
        #end of playlist (EOPL) reached. Ignore last played playlist
        LASTPLAYLIST=""
    fi
    # workaround end
```

This leads to an endless loop of the playlist if the card is left on the reader.
What is missing is a detection whether the card has been removed in between so that the playlist will end and stop after one loop if the card is left on the reader and only begin if it is re-swiped.

MiczFlor, 2020-05-02:
@jo-me I like that workaround :) the loop is a bit of a scary thought. Are you concerned for CPU / power bank life? But if there was a small sleep of a few seconds?

MiczFlor, 2020-05-14:
Hi @jo-me 
your workaround, if I understand correctly now, creates a loop if one uses the hack to allow RFID cards on the Phoniebox (the Tonies way, so to speak). Other than that, it should fix your bug, no?

jo-me, 2020-05-14:
Well, yeah except for the part that the removal of the card is currently not detected. This would be necessary to avoid the loops.

MiczFlor, 2020-05-19:
Hi @jo-me where did you add your hack. I am currently thinking to add the following at this line:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/358fb207b37853cadd8275b878f67aee6c203a57/scripts/rfid_trigger_play.sh#L404
~~~
elif [ "$SECONDSWIPE" == "NOAUDIOPLAY" ]
        then
            # The following involves NOT playing the playlist, so we set:
            PLAYPLAYLIST=no
            # Special case for NOAUDIOPLAY because once the playlist has finished, 
            # it needs to be noted by the system that the second swipe is like a *first* swipe.
            currentSong=`mpc current`
            if [[ -z "$currentSong" ]]; then
                #end of playlist (EOPL) reached. Ignore last played playlist
                PLAYPLAYLIST=yes
            fi
~~~
Could you try that?

jo-me, 2020-05-19:
I added it right after line 333 when the LASTPLAYLIST variable is set.

Will try it out, but don't have a working box right now as i'm just in the middle of case building :)

jo-me, 2020-06-16:
I installed the latest version yesterday and it seems to be working. 
Thanks for including that fix!

closed 2020-06-16 by jo-me