MiczFlor/RPi-Jukebox-RFID #915: 🚀 | Set a Startup-Volume that is set on each startup

BerniPi, 2020-04-20:
## Feature Description

### What functionality would you like to see in your phoniebox?

I would like to set a Volume, that is set on each startup.

### How do you envision the feature to work from a users perspective?

There is an option in Settings to set the Startup-Volume (same as max volume). this Volume is set on each startup

### Further information that might help

I am trying to set up a startup script to get this functionality.

BerniPi, 2020-04-29:
I now added a command to the startup-sound.service in /etc/systemd/system/phoniebox-startup-sound.service
But it's a bit tricky: I use the playout_controls.sh (to later add the startup-sound capability), but when the script runs to early, it doesn't get exexcuted. Meanwhile it has to be before the startup-sound. And there has to be a short time between the two commands, because the volume gets faded.

I changed the script to this:
`[Unit]
Description=Phoniebox Startup Sound
After=multi-user.target

[Service]
User=pi
Group=pi
Type=oneshot
RemainAfterExit=true
WorkingDirectory=/home/pi/RPi-Jukebox-RFID
ExecStartPre=/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=setvolume -v=50
ExecStart=/bin/sleep 2
ExecStartPost=/usr/bin/mpg123 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3
`
Now i have to look into playout_controls.sh and how the settings get saved and read from the settings-folder and add a startup-volume option to the file.

s-martin, 2020-04-29:
Great, could you open a PR?

BerniPi, 2020-04-29:
Isn't it too early?
I drastically changed the time, the startup sound gets played from "After=alsa-state.service" to "After=multi-user.target". Maybe there is another solution.
It also would be great to set the startup-volume in the settings folder and not in the service file.

BerniPi, 2020-05-01:
#923

veloxidSchweiz, 2020-05-02:
In order to do this the mpd/mopidy service needs to be available ( the reason why @BerniPi change the time when it should be played. I do realized in my system that it sometimes takes some time untill everything is ready.
In the `gpio_control` I am actually trying to connect to the mpd using a mpd-client and only when i can connect i enable the LED. (see https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/84b02a8e05b32c86517b4dad07fc0e94b883a500/components/gpio_control/GPIODevices/led.py#L44-L46)

Maybe one could think about a similar script.... (one could create a play_after_mpd_start.sh, which asks for connection)

BerniPi, 2020-05-02:
I added the code last night: #923 
Sorry about the great number of commits, I'm new to git(hub).

I thougt about setting the new volume with amixer, but the volumepercentages are different.
What about setting the  new volume at shutdown?

Is it by intention, that the percentage sign is missing in the volumecircle?

MiczFlor, 2020-05-02:
@veloxidSchweiz I changed the startup sound service in the latest `develop` branch. Now there is a script called (`scripts/startup_scripts.sh`) instead. This might make it easier to stall the startup sound and what else there might be needed after boot based on the ready state of such components.

closed 2020-05-02 by BerniPi