MiczFlor/RPi-Jukebox-RFID #919: 🐛 | RFID Card to "toggle wifi" doesn't work

manajoe, 2020-04-23:
## Bug
I assigned a card ID to toggle wifi, but I can only toggle once (e.g. switch it on, but not off again)

### What I did
I'm running on version 2.0 rc7 classic (installed via buster install script). I'm trying to add a card ID to toggle wifi.
I found the existing function "togglewifi" in scripts/playout_controls.sh. 
I added the following lines to scripts/rfid_trigger_play.sh:
`$TOGGLEWIFI)
$PATHDATA/playout_controls.sh -c=togglewifi
;;
`
And these lines to settings/rfid_trigger_play.conf
`TOGGLEWIFI="0002173904"`
Then I swiped card ID 0002173904

### What happened
- Wifi is off.
- I swipe the card.
- Wifi is turned on.
/var/log/messages shows:
`raspberrypi rfkill: unblock set for type wifi
raspberrypi kernel: [ 6944.925978] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
raspberrypi kernel: [ 6944.925999] brcmfmac: power management disabled
raspberrypi kernel: [ 6946.316686] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready`
- after waiting several minutes, I swipe the card again.
- Wifi is NOT turned off.
/var/log/messages shows:
`raspberrypi rfkill: unblock set for type wifi`

Debug log shows the following after both swipes:
`########### SCRIPT rfid_trigger_play.sh (2020-04-23.20:34:56) ##
Card ID '0002173904' was used
########### SCRIPT playout_controls.sh (2020-04-23.20:34:57) ##
VAR COMMAND: togglewifi
VAR VALUE: 
Attempting to play: /home/pi/RPi-Jukebox-RFID/shared/audiofolders/
Type of play $VALUE: 
Path not found /home/pi/RPi-Jukebox-RFID/shared/audiofolders/`

### I expected this to happen

First swipe to activate Wifi (if it is initially off), second swipe to deactivate Wifi

If I run the script from the CLI, everything works fine:
`RPi-Jukebox-RFID/scripts/playout_controls.sh -c=togglewifi
Wifi will now be deactivated`
For further testing, I assigned separate card IDs for ENABLEWIFI and DISABLEWIFI, they both work as expected.

### Further information that might help
First swipe to activate Wifi which works:

> Apr 23 21:10:46 raspberrypi daemon_rfid_reader.py[3637]: Wifi will now be activated

> Apr 23 21:10:46 raspberrypi rfkill: unblock set for type wifi
> Apr 23 21:10:46 raspberrypi kernel: [ 8521.006526] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
> Apr 23 21:10:46 raspberrypi kernel: [ 8521.006545] brcmfmac: power management disabled
> Apr 23 21:10:46 raspberrypi systemd[1]: Starting Load/Save RF Kill Switch Status...
> Apr 23 21:10:46 raspberrypi daemon_rfid_reader.py[3637]: /home/pi/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh: line 326: [: too many arguments
> Apr 23 21:10:46 raspberrypi systemd[1]: Started Load/Save RF Kill Switch Status.
> Apr 23 21:10:48 raspberrypi dhcpcd[15087]: wlan0: carrier acquired
> Apr 23 21:10:48 raspberrypi kernel: [ 8522.379770] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
> Apr 23 21:10:48 raspberrypi dhcpcd[15087]: wlan0: IAID eb:5d:79:c5
> Apr 23 21:10:48 raspberrypi dhcpcd[15087]: wlan0: adding address fe80::ea43:b43c:8b14:3d48
> Apr 23 21:10:48 raspberrypi avahi-daemon[367]: Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::ea43:b43c:8b14:3d48.
> Apr 23 21:10:48 raspberrypi avahi-daemon[367]: New relevant interface wlan0.IPv6 for mDNS.
> Apr 23 21:10:48 raspberrypi avahi-daemon[367]: Registering new address record for fe80::ea43:b43c:8b14:3d48 on wlan0.*.
> Apr 23 21:10:48 raspberrypi dhcpcd[15087]: wlan0: probing address 192.168.13.17/24
> Apr 23 21:10:48 raspberrypi dhcpcd[15087]: wlan0: soliciting an IPv6 router
> Apr 23 21:10:51 raspberrypi systemd[1]: systemd-rfkill.service: Succeeded.
> Apr 23 21:10:52 raspberrypi dhcpcd[15087]: wlan0: using static address 192.168.13.17/24
> Apr 23 21:10:52 raspberrypi avahi-daemon[367]: Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.13.17.
> Apr 23 21:10:52 raspberrypi dhcpcd[15087]: wlan0: adding route to 192.168.13.0/24
> Apr 23 21:10:52 raspberrypi dhcpcd[15087]: wlan0: adding default route via 192.168.13.1
> Apr 23 21:10:52 raspberrypi avahi-daemon[367]: New relevant interface wlan0.IPv4 for mDNS.
> Apr 23 21:10:52 raspberrypi avahi-daemon[367]: Registering new address record for 192.168.13.17 on wlan0.IPv4.
> Apr 23 21:11:01 raspberrypi dhcpcd[15087]: wlan0: no IPv6 Routers available

Second swipe to deactivate Wifi, which fails:

> Apr 23 21:11:07 raspberrypi daemon_rfid_reader.py[3637]: Wifi will now be activated
> Apr 23 21:11:07 raspberrypi rfkill: unblock set for type wifi
> Apr 23 21:11:07 raspberrypi daemon_rfid_reader.py[3637]: /home/pi/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh: line 326: [: too many arguments

By the way, line 326 of rfid_trigger_play.sh looks like this:
`if [ ! -z "$FOLDER" -a ! -z ${FOLDER+x} -a -d "${AUDIOFOLDERSPATH}/${FOLDER}" ]; then`

## Software

### Base image and version

`2020-02-13-raspbian-buster-lite.img`

### Branch / Release

master

### Installscript

/master/scripts/installscripts/buster-install-default.sh

## Hardware

### RaspberryPi version

3B Rev 1.2

### RFID Reader

16c0:27db HXGCoLtd Keyboard

### Soundcard

onboard

### Other notable hardware
```
GPIO Buttons:
shut = Button(3, hold_time=2)
vol0 = Button(13,pull_up=True)
volU = Button(16,pull_up=True,hold_time=0.3,hold_repeat=True)
volD = Button(19,pull_up=True,hold_time=0.3,hold_repeat=True)
next = Button(26,pull_up=True)
prev = Button(20,pull_up=True)
halt = Button(21,pull_up=True)
#reco = Button(6, pull_up=True) # Choose GPIO to fit your hardware
#play = Button(12,pull_up=True) # Choose GPIO to fit your hardware

shut.when_held = def_shutdown
vol0.when_pressed = def_vol0
volU.when_pressed = def_volU
#When the Volume Up button was held for more than 0.3 seconds every 0.3 seconds he will call a ra$
volU.when_held = def_volU
volD.when_pressed = def_volD
#When the Volume Down button was held for more than 0.3 seconds every 0.3 seconds he will lower t$
volD.when_held = def_volD
next.when_pressed = def_next
prev.when_pressed = def_prev
halt.when_pressed = def_halt
#reco.when_pressed = def_recordstart
#reco.when_released = def_recordstop
#play.when_pressed = def_recordplaylatest
```

s-martin, 2020-04-25:
@Franzformator, do you have any ideas? You were mentioned in `playout_controls.sh` for the togglewifi function.

MiczFlor, 2020-05-03:
Hi @manajoe 
do you by chance also have a folder linked to the RFID card?

manajoe, 2020-05-03:
@MiczFlor Thanks for your suggestion, I double checked again, definetely not.

But I think I found the problem somewhere else:

Under "togglewifi" in _RPi-Jukebox-RFID/scripts/playout_controls.sh_ I removed this line:
`       ##if [[ $(rfkill list wifi | grep -i "Soft blocked: no")  > 0 ]]`

and changed it to this:
`if [[ $(rfkill list wifi | grep -i "Soft blocked: no")  != "" ]]`

This worked for me, I tested on my above mention RPi 3B and another newer Model 3B+ successfully.

Strange though, that the command `RPi-Jukebox-RFID/scripts/playout_controls.sh -c=togglewifi` worked from CLI before and still does... 

Maybe someone else can confirm this behavior?

MiczFlor, 2020-05-12:
Note to self: 
* possibly check if wifi up or down by checking IF IP address available for wlan0
* read... https://raspberrytips.com/disable-wifi-raspberry-pi/

PR #969 (2020-05-22): Added check for ip within togglewifi function
As requested from @MiczFlor I tried to implement the check IP into togglewifi to fix #919 .

The problem here is, I cannot test this setup since I don't have a spare RFID reader.
closed 2020-05-22 by MiczFlor