MiczFlor/RPi-Jukebox-RFID #920: 🐛 | Skip Samba configuration dialog during installation by default?

fredg02, 2020-04-23:
## Bug

### What I did
I used the install script (buster-install-default.sh) and wanted to run a non-interactive installation (see also issue #858).
### What happened
The following screen pops up during the installation:
[image]
### I expected this to happen
After the initial set of questions during the installation, no further interaction should be required.
If the installation runs in automatic mode (upcoming PR), no interaction should be required at all.
### Further information that might help
debconf-set-selections can be used to skip this configuration dialog.
* `echo "samba-common samba-common/dhcp boolean false" | sudo debconf-set-selections` => same as if you would select "No" in the dialog. 
* `echo "samba-common samba-common/dhcp boolean true" | sudo debconf-set-selections` => same as if you would select "Yes" in the dialog. 

Here are my questions:
* Should this dialog be skipped by default for any type of installation (interactive and non-interactive)?
* Should this dialog only be skipped during a non-interactive installation?
* What should be the default answer? Yes or No?
(I don't think this dialog is even relevant for most users, but I might be wrong.)
### Base image and version
2020-02-13-raspbian-buster-lite.img
### Branch / Release
develop
### Installscript
scripts/installscripts/buster-install-default.sh
## Hardware
### RaspberryPi version, etc
does not matter

s-martin, 2020-04-25:
From my point of view we could default this to "No" directly in the script, I personally don't see the use case for WINS.

We then should document this behavior somewhere.

s-martin, 2020-05-01:
But I would like to check with @MiczFlor, if I'm missing a use case for WINS here.

MiczFlor, 2020-05-08:
It's been added. Thanks for the hint:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/18de7c9be16a840afac4fc980a05cc550c5bb10f/scripts/installscripts/buster-install-default.sh#L1090

closed 2020-05-08 by MiczFlor