MiczFlor/RPi-Jukebox-RFID #921: 🚀 | Notification about IP of RaspberryPi

BerniPi, 2020-04-29:
## Feature Description
I know that it's recommended to use a static ip. But i also want to connect my phoniebox to the networks of grandparents and to hotspots from mobiles.
I can add the wifi-settings from my phone's hotspot and the phoniebox connects, but i cannot access to the webinterface without knowing the ip. Connecting with the hostname works at home, but not on my android phone.

### What functionality would you like to see in your phoniebox?
The phoniebox notify me about his ip: There is a possibility that the ip gets spoken, but thats not practicable. A good solution is, to mail the ip at startup or at buttonpress.

### How do you envision the feature to work from a users perspective?
The phoniebox starts and I get a notification about the ip.

### Further information that might help
I know that this is very special and i don't think that it shoud be added to phoniebox as code, but maybe anyone else is interested or you want to add it to the wiki.

Solution: It's fairly simple. I found some scripts, but only this one worked: https://github.com/oliverscheer/send-email-with-device-ip-address
I execute it with a systemd-service at startup.

s-martin, 2020-05-01:
I would see this as an optional features as some people may have a mail agent disabled to improve overall startup performance.

If you provide a PR we gladly integrate it.

BerniPi, 2020-05-01:
Looks like MiczFlor is already working on it: https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/3a65a6c3abe8ef0e498ac421656f524130ece66b

veloxidSchweiz, 2020-05-02:
maybe an alternative would be a text to voice output.. .
it would be a pretty small script:
install 
`apt-get install espeak`
and then:
`hostname -I | espeak`

veloxidSchweiz, 2020-05-02:
Looing in the commit,
One should think about a more generic event system,...
maybe an option that one can generate an event 'ip_change' and then one can connect handlers with this event...
This would be the pre-work to later on allow easy implementation of plugins and every one can install the things he like.
But this would definitely need some more thoughts and specially some architectual decissions.
@s-martin what do you think?

s-martin, 2020-05-02:
I think a general event system with configurable sinks (e.g. mail, display, log) would be a good idea.
Personally I lack the necessary Python experience to implement such a feature.

MiczFlor, 2020-05-02:
Hi @BerniPi @s-martin 
I finished a version that reads out the IP address. I will push this to develop, have it tested on Pi3 and it works. While testing I found a few issues with the web UI settings page, which might have come from developing on the docker image. The shell script global write config needs to be run as root from the webUI. I will push this at the same time.

The Settings page has the option to switch it on and off. So if you move to a different Wifi, the process would be:

* go to settings and switch it on
* add new wifi
* boot in new wifi
* the IP is read out
* go to settings and switch it off

I worked on installing the mail version, but have a mail server configured on the Pi would really make the install a *nightmare* :) for average users.

What I will do now: push my changes and then manually merge @BerniPi pull request. There will be a bunch of conflicts. So I do this now and test is on the Pi

MiczFlor, 2020-05-02:
Ah... and another thing: the service startup sound will cease to exist. I created a startup-scripts.sh file which we can add on. This will be launched when booting. I will make the changes in @BerniPi pull request, too.

closed 2020-05-02 by MiczFlor

MiczFlor, 2020-05-03:
Hi @BerniPi 
just added the RFID option to the latest develop merge.