MiczFlor/RPi-Jukebox-RFID #928: install script for `develop` branch installs `master` branch

MiczFlor, 2020-05-03:
After using the install script for the `develop` branch, Phoniebox works as expected. Good.
But checking the branch with `git branch` I am on `master` not `develop`.
@s-martin @ZyanKLee what do you think. Shouldn't the develop install be in the develop world?

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/4c7c9c7bc9eddf726346af96a7df9ce6cad56f6f/scripts/installscripts/buster-install-default.sh#L696

s-martin, 2020-05-03:
The idea was that you need to set a env variable before executing the install script, i.e.  `GIT_BRANCH=develop bash ./buster-install-default.sh`.

MiczFlor, 2020-05-03:
True, that's a good point (that I knew and forgot...). I have to check the install script again. My understanding was:
- if you do not provide that variable (as it is correctly done in the wiki for develop) it will switch to master
- the version number will be generated including the branch
But, I did the install with the conf file (non-interactive) and get the version number 2.0 %GIT_COMMIT% %GIT_BRANCH% in the WebUI

MiczFlor, 2020-05-03:
Hi @s-martin 
All works as expected. My bad.
I thought the first time when installing with the config file non-interactive the version number was wrong. Now that didn't happen.

closed 2020-05-03 by MiczFlor