MiczFlor/RPi-Jukebox-RFID #931: 🐛 | BUG SUMMARY

borsic77, 2020-05-03:
## Bug

### What I did

Trying to use RPi jukebox with a Waveshare PN532 NFC hat, SPI configuration. I followed the one-script install guide, and used Reader.py.experimental. RegisterDevice.py shows MFRC522 and RDM6300. 

### What happened

When I select MFRC522 I can run the python sample code from Waveshare, and it detects my RFID card.
https://www.waveshare.com/wiki/PN532_NFC_HAT
However, in the web-interface the RFid card does not show up under the Card ID tab. I am guessing that I have to change some settings in Reader.py?

### I expected this to happen

Detecting the RFid cards in the web interface.

### Further information that might help

tail -n 500 /var/log/syslog
May  3 16:06:13 raspberrypi avahi-daemon[390]: Withdrawing address record for fe80::ba27:ebff:fe5f:8103 on wlan0.
May  3 16:06:13 raspberrypi avahi-daemon[390]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::ba27:ebff:fe5f:8103.
May  3 16:06:13 raspberrypi avahi-daemon[390]: Interface wlan0.IPv6 no longer relevant for mDNS.
May  3 16:06:13 raspberrypi lightdm[479]: Error getting user list from org.freedesktop.Accounts: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Accounts was not provided by any .service files
May  3 16:06:13 raspberrypi polkitd[450]: started daemon version 0.105 using authority implementation `local' version `0.105'
May  3 16:06:13 raspberrypi dbus-daemon[362]: [system] Successfully activated service 'org.freedesktop.PolicyKit1'
May  3 16:06:13 raspberrypi systemd[1]: Started Authorization Manager.
May  3 16:06:13 raspberrypi systemd[1]: Received SIGRTMIN+21 from PID 174 (plymouthd).
May  3 16:06:13 raspberrypi systemd[1]: Received SIGRTMIN+21 from PID 174 (plymouthd).
May  3 16:06:13 raspberrypi systemd[1]: plymouth-quit-wait.service: Succeeded.
May  3 16:06:13 raspberrypi systemd[1]: Started Hold until boot process finishes up.
May  3 16:06:13 raspberrypi systemd[1]: Started Light Display Manager.
May  3 16:06:13 raspberrypi systemd[1]: plymouth-start.service: Succeeded.
May  3 16:06:13 raspberrypi systemd[1]: Started Getty on tty1.
May  3 16:06:13 raspberrypi systemd[1]: Reached target Login Prompts.
May  3 16:06:14 raspberrypi systemd[1]: Started The PHP 7.3 FastCGI Process Manager.
May  3 16:06:14 raspberrypi raspi-config[398]: Checking if shift key is held down: No. Switching to ondemand scaling governor.
May  3 16:06:14 raspberrypi systemd[1]: Started LSB: Switch to ondemand cpu governor (unless shift key is pressed).
May  3 16:06:14 raspberrypi systemd[1]: Started Disk Manager.
May  3 16:06:14 raspberrypi udisksd[350]: Acquired the name org.freedesktop.UDisks2 on the system message bus
May  3 16:06:14 raspberrypi systemd[1]: Started Lighttpd Daemon.
May  3 16:06:15 raspberrypi daemon_rfid_reader.py[456]: Please run RegisterDevice.py first
May  3 16:06:15 raspberrypi systemd[1]: systemd-rfkill.service: Succeeded.
May  3 16:06:15 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Main process exited, code=exited, status=1/FAILURE
May  3 16:06:15 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
May  3 16:06:15 raspberrypi exim4[454]: Starting MTA: exim4.
May  3 16:06:15 raspberrypi systemd[1]: Started LSB: exim Mail Transport Agent.
May  3 16:06:15 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Service RestartSec=100ms expired, scheduling restart.
May  3 16:06:15 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Scheduled restart job, restart counter is at 1.
May  3 16:06:15 raspberrypi systemd[1]: Stopped Phoniebox RFID-Reader Service.
May  3 16:06:15 raspberrypi systemd[1]: Started Phoniebox RFID-Reader Service.
May  3 16:06:15 raspberrypi daemon_rfid_reader.py[830]: Please run RegisterDevice.py first
May  3 16:06:15 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Main process exited, code=exited, status=1/FAILURE
May  3 16:06:15 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
May  3 16:06:16 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Service RestartSec=100ms expired, scheduling restart.
May  3 16:06:16 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Scheduled restart job, restart counter is at 2.
May  3 16:06:16 raspberrypi systemd[1]: Stopped Phoniebox RFID-Reader Service.
May  3 16:06:16 raspberrypi systemd[1]: Started Phoniebox RFID-Reader Service.
May  3 16:06:16 raspberrypi daemon_rfid_reader.py[835]: Please run RegisterDevice.py first
May  3 16:06:16 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Main process exited, code=exited, status=1/FAILURE
May  3 16:06:16 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
May  3 16:06:16 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Service RestartSec=100ms expired, scheduling restart.
May  3 16:06:16 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Scheduled restart job, restart counter is at 3.
May  3 16:06:16 raspberrypi systemd[1]: Stopped Phoniebox RFID-Reader Service.
May  3 16:06:16 raspberrypi systemd[1]: Started Phoniebox RFID-Reader Service.
May  3 16:06:17 raspberrypi lightdm[853]: Error getting user list from org.freedesktop.Accounts: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Accounts was not provided by any .service files
May  3 16:06:17 raspberrypi systemd[1]: Created slice User Slice of UID 1000.
May  3 16:06:17 raspberrypi systemd[1]: Starting User Runtime Directory /run/user/1000...
May  3 16:06:17 raspberrypi systemd[1]: Started User Runtime Directory /run/user/1000.
May  3 16:06:17 raspberrypi systemd[1]: Starting User Manager for UID 1000...
May  3 16:06:17 raspberrypi daemon_rfid_reader.py[847]: Please run RegisterDevice.py first
[... 468 lines cut ...]
May 03 16:06:22 raspberrypi mopidy[472]: INFO     Starting Mopidy core
May 03 16:06:22 raspberrypi mopidy[472]: INFO     Audio output set to "alsasink"
May 03 16:06:22 raspberrypi mopidy[472]: INFO     Starting Mopidy frontends: Iri
May 03 16:06:22 raspberrypi mopidy[472]: INFO     Starting Iris 3.43.0
May 03 16:06:22 raspberrypi mopidy[472]: INFO     MPD server running at [::ffff:
May 03 16:06:22 raspberrypi mopidy[472]: INFO     HTTP server running at [::fff
-->

## Software

### Base image and version

Raspbian GNU/Linux 10

### RaspberryPi version

3 B+

### RFID Reader

Waveshield PN532 NFC Hat in SPI mode

###

Thank you for your help!

fredg02, 2020-05-15:
I've added basic support for PN532 RFID readers via I2C. So you might be able to get your RFID reader to work with that as well. Check out the develop branch and see https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/rfid-reader/PN532 for more info.

If you want/need to use SPI you will most likely need to do some hacking yourself.

BTW: I don't consider this to be a bug. This should be a feature-request.

MiczFlor, 2020-05-19:
Hi @borsic77 
thanks for the input. You write *"in the web-interface the RFid card does not show up under the Card ID tab"* and I am not sure what you expect in that tab.
In the web UI this tab allows you to connect RFID tags with audio folders or system commands. Swiping the card should display the RFID ID in the form. The card itself is not shown there.

s-martin, 2021-02-28:
Hey @borsic77, is this still an issue?

charlesvestal, 2022-01-28:
I can confirm the Waveshare PN532 hat doesn't work out of the box over I2C. 

https://www.waveshare.com/wiki/PN532_NFC_HAT

While running run_register_rfid_reader.py I did see the following error 

/bin/sh: 1: ./setup.inc.sh: Permission denied

which may be related.

charlesvestal, 2022-01-28:
I can confirm the waveshare hat DOES work out of the box over I2C when configured with the pins and jumpers as shown here:

https://www.waveshare.com/wiki/PN532_NFC_HAT

I think the ticket confusion here is on the expectation of the cards screen. I too expected it to show cards as swiped.  I suggest updating the empty state message to "No cards registered. Add one below to get started" or similar.

s-martin, 2022-01-29:
Labeled also for future 3.

s-martin, 2023-12-14:
Confirmed it works both with recent 2.x versions and 3.x

closed 2023-12-14 by s-martin