MiczFlor/RPi-Jukebox-RFID #935: 🚀 | Sleep function with reducing volume

Mechi144, 2020-05-07:
## Feature Description

### What functionality would you like to see in your phoniebox?

My son loves the Phoniebox very much and I would like to use the box as a sleep aid.
I would like him to be able to listen to music while falling asleep, which slowly becomes quieter.

### How do you envision the feature to work from a users perspective?

At the moment there is a stop playout timer which stops the player after a certain time. Would it be possible to extend this with a function that slowly reduces the volume to zero over the set period of time until it stops playing?
If the function is activated, the phoniebox should slowly reduce the volume over the remaining sleep period.

### Further information that might help

 The activation could possibly be done via a checkbox next to the already existing Stop playout timer in the web app or via RFID card.
Afterwards, the volume should return to the previously set level so that the speaker is not muted the next time it is used.

MiczFlor, 2020-05-08:
Hi @Mechi144 
a very nice thought. Not so trivial to implement :) the audio playout is controlled by a player that does not support this feature in itself. I will think about it.
All the best, micz

Piehti, 2020-05-10:
+1 -  while I already like the autosleep.
Braindump: Why not sending intervalled vol- decreases. User could even specify the time slots passing by before the next level down. Stupidly thought (sorry if stupid, im just a t(h)inker not a real coder):

```
playoutcontrols.sh -c=voldown
sleep 100
playoutcontrols.sh -c=voldown
sleep 100
playoutcontrols.sh -c=voldown
sleep 100
playoutcontrols.sh -c=voldown
sleep 100
playoutcontrols.sh -c=voldown
sleep 100
playoutcontrols.sh -c=voldown
sleep 100
```

While the sleep interval could / should be customizable.

I would try myself, but time... dat time....

MiczFlor, 2020-05-19:
Hi @Piehti 
that should actually work. I like the thinking behind it. I will see if I find the coding time to convert the thinking into code :)

BerniPi, 2020-06-08:
> As I am not a very experienced coder I'm struggling with the calculated VOLSTEP. If the division is 0 (e.g. if volume is 20 and minutes are 30) I try to set it to 1. But in other situations (volume 80 and minutes 25) the step of "3" would be too small and the volume would only reach 26...
Maybe someone has an idea?

How do you get 26? 80-25*3=5 What's my mistake?

The first thing i thought about, was to not calculate the VOLSTEPS, but the period of time between the commands. But minutes is the smallest interval for the at command.
You already round up if the value is smaller than 1. Why not always round up?
Just always add 1 to the quotient of VOLPERCENT and VOLSTEP. You don't need the check for zero then. But with your example you will reach 0 5 minutes early.

MiczFlor, 2020-06-09:
The way I would do it is in a loop that does the following

* get current volume percent
* divide current vol% by ten, save in varStep
* start loop ten itterations
    * get current volume percent volCurrent
    * volCurrentNew = volCurrent - varStep
    * set volume `playout_controls.sh -c=setvolume -v=${volCurrentNew}`
* loop ends
* shutdown silent `playout_controls.sh -c=shutdownsilent`

manajoe, 2020-06-09:
Thanks for your hints
@BerniPi : sorry, i exchanged the values, the biggest problem was volume 85 and 30 minutes where a leftover would have been 25
I followed @MiczFlor and now always have a fixed amount of 10 steps and schedule the "at" command every x minutes (where x is "value" divided by "10", so for value 30 it's every 3 minutes).

What do you think?
https://github.com/MiczFlor/RPi-Jukebox-RFID/compare/develop...manajoe:shutdown_vol_reduce?expand=1

closed 2021-01-23 by MiczFlor