MiczFlor/RPi-Jukebox-RFID #938: 🐛 | deavtivating repeat does not work

oswaldo-be, 2020-05-10:
## Bug

### What I did

### What happened

### I expected this to happen

### Further information that might help

## Software

### Base image and version

### RFID Reader

not relevant...
### Soundcard

### Other notable hardware

Or maybe I set something wrong?
In that way can anybody help me please?
Thank you very much in advance!

oswaldo-be, 2020-05-10:
hmm now that I deactivated it one time per mpc repeat 0 I can activate and deactivate it in the GUI again and it works...

Piehti, 2020-05-10:
So this issue is closed?
I would have suggested to check the rights on the *.conf *.scripts as to restart the mpd service by chattering down a: sudo service mpd restart. Non persistent problems seem to vanish a lot by mpd restart.

oswaldo-be, 2020-05-10:
It can be closed. Thank you

closed 2020-05-12 by s-martin