MiczFlor/RPi-Jukebox-RFID #946: 🐛 | BUG SUMMARY

sambold, 2020-05-14:
## Bug

I implemented a stop on removal function that (kind of) works. If I put the card on the player, it starts to play. As soon as I remove it, it pauses. 

Unfortunately I am left with a minor (but still annoying) bug: when the playlist ends or when I push prev on the first song/next on the last song, the functionality of play/pause seem to be switched. So if I remove the card, the music starts to play. If I place the card on the phoniebox it stops/pauses.If the playlist ends a second time or if I use a different tag, the functionality is switched again and everything works fine again.

I think this could be related to the fact that "playerpause" in playout_controls.sh is set to "mpc toggle" somehow. But I am not really sure.

If you had any idea how I could solve this problem, I really would appreciate it :)

### What I did

I made some adjustment to the following scripts (but I think they have nothing to do with the problem): 
- pirc522/rfid.py
- Reader.py
- daemon_rfid_reader.py

Here are the changes in some more detail

#### pirc522/rfid.py: added a timeout parameter to function
def wait_for_tag(self,timeout):
        # enable IRQ on detect
        start_time = time.time()
        self.init()
        self.irq.clear()
        self.dev_write(0x04, 0x00)
        self.dev_write(0x02, 0xA0)
        # wait for it
        waiting = True
        while waiting and time.time()-start_time < timeout:
            self.dev_write(0x09, 0x26)
            self.dev_write(0x01, 0x0C)
            self.dev_write(0x0D, 0x87)
            waiting = not self.irq.wait(0.1)
        self.irq.clear()
        self.init()

#### Reader.py: added the timeout parameter to the function call
self.device.wait_for_tag(2)

#### daemon_rfid_reader.py: implementing the stop on removal functionality
#!/usr/bin/env python2

import subprocess
import os
from Reader import Reader

reader = Reader()

#get absolute path of this script
dir_path = os.path.dirname(os.path.realpath(__file__))

print dir_path

last_cardid = ""
removed = True
while True:
    cardid = reader.reader.readCard()
    if cardid != None:
        removed = False
        if cardid != last_cardid:
            # new card
            last_cardid = cardid
            try:
                # start the player scritp and pass on the cardid
                subprocess.call([dir_path + '/rfid_trigger_play.sh --cardid=' + cardid], shell=True)
            except OSError as e:
                print "Execution of play failed."
    elif cardid == None and removed == False:
        # card removed after play
        removed = True
        last_cardid = ""
        try:
            # pause the player scrip
            subprocess.call([dir_path + '/playout_controls.sh -c=playerpause'], shell=True)
        except OSError as e:
            print "Execution of Pause failed."

### What happened

After the playlist ends (or if I push prev/next on first/last song) the functionality switches. That means if I remove the card, the music starts to play and if I put the card on the box it pauses.

If the playlist ends once again the functionality is back to normal. So if I remove the card, the music pauses. If I put the card on the box, the music starts to play.

### I expected this to happen

The functionality should not switch after a playlist ends.

### Base image and version

OS Distribution
Raspbian
Description
Raspbian GNU/Linux 9.3 (stretch)
Release
9.3
Codename
stretch

### Branch / Release

up-to-date with origin/master

## Hardware

### RaspberryPi version

Raspberry Pi 3 Model B Rev 1.2

### RFID Reader

Mifare RC522 RFID Reader

### Other notable hardware

I edited the gpio-buttons.py so that a special button can execute two different functions (push/hold), but the changes had nothing to do with prev/next

prev = Button(20,pull_up=True)
next = Button(26,pull_up=True) 

def def_next():
    check_call("./scripts/playout_controls.sh -c=playernext", shell=True)
def def_prev():
    check_call("./scripts/playout_controls.sh -c=playerprev", shell=True)

s-martin, 2020-05-17:
Could you add your code changes as a pull request?

sambold, 2020-06-03:
Sorry that it took me that long to answer. First I could not find the time. Then I ran into the problem that there is a "new" phoniebox version based on buster and it seems that the folder structure has changed a little bit. All of this together leads to up to the result that I am not able to provide an adequate/working pull request (right now I am struggling with RegisterDevice.py, because it won't find the pirc522 library. But that is a different problem. Anyways ... I am working on it ...)

s-martin, 2020-06-04:
For the pirc522 reader you now need to execute 
`components/rfid-reader/RC522/setup_rc522.sh`
before you run RegisterDevice.py

s-martin, 2020-06-04:
Additional info: we now use pirc522 directly from the repo https://github.com/ondryaso/pi-rc522

So you can’t change `pirc522/rfid.py` anymore. But I think there was a similar patch applied to the original code, see https://github.com/ondryaso/pi-rc522/commit/ac0ae47eae033fe52778b85c43d4d1bb557efbde

Toqqi, 2020-10-21:
I'm currently looking into the same issue regarding a "stop on removal" feature using pi-rc522.

Regarding your comment on the relation to "playerpause" in playout_controls.sh being set to "mpc toggle":
You could use "playerpauseforce", which is set to "mpc pause" in playout_controls.sh
I have tested this and it does not toggle play when executed on a paused track.
You have to provide the delay with the -v parameter for it to work.

E.g.
`subprocess.call([dir_path + '/playout_controls.sh -c=playerpauseforce -v=0.1'], shell=True)`

sambold, 2020-12-01:
hoi ... once again it took me forever to try/implement your suggestions. The good news: (almost) everything works perfectly now. At least the stop-on-removal-function works like a charm. Now I just have to improve the sound quality (probably with a cheap soundcard). 
Unfortunately I am not sure if I can remember all the steps that I took fixing my problem, but I think I did the following: 
1. I installed the new phoniebox version based on buster
2. As before I  added a timeout parameter to the rfid.py-file. To do so I had to identify the new path of rfid.py which is now located in /usr/local/lib/python3.7/dist-packages/pirc522/rfid.py. Probably that is not a good/clean solution, but I think it should work until the next time I run an update (and then I have to redo it again or something like that). 
3. As before I increased the antenna gain by setting antenna_gain to 0x07 (rfid.py)
4. As before I used the self.device.wait_for_tag() call in Reader.py/Reader.py.experimental with the timeout-parameter (in my case I used a value of 5 to compensate for situations where pirc fails to read the ID. This happens quite often, but won't last for too long. With the parameter set to 5 the audio only stops, if the reader can't read an ID for 5 seconds, which is only the case, if I remove the tag completely)
4. To overcome the issue where the control functions (play/pause) were switched, I just had to change playerpause to playerpauseforce as Toqqi suggested 
Thanks for your support :)

MiczFlor, 2020-12-01:
Hi @sambold 
glad to see you got it to work. I am unsure what to do with your description. It would be really good to get this clean solution implemented:
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1122
I am also pinging @Toqqi and @urge-at - seems there are three very capable brains circling aroud the same issue. Possibly between the three of you, this is a piece of cake to implement "properly"?

sambold, 2020-12-01:
Hi, I am always a little bit scared of experimenting with the phoniebox-code, because usually before i fix things, I start to break them ^^. Anyways I

s-martin, 2022-12-24:
#1122 was implemented a long time ago.

Closing for now as there was no activity for a long time.
Feel free to reopen a new issue, if the issue persists with current version 2.4 (or beta 3.2).

closed 2022-12-24 by s-martin