MiczFlor/RPi-Jukebox-RFID #950: Question: Experience with Pimoroni Pirate Audio

drocx, 2020-05-16:
Hello,
I discovered [Pirate Audio](https://shop.pimoroni.com/collections/pirate-audio) a few days ago. 

Has anyone Experience with this device?

It would be the perfect addition for the phoniebox:
- Playback Control Buttons without soldering.
- I2S DAC for better audio
- A high-res color display for Cover Art

s-martin, 2020-05-16:
We have a Readme, see https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/audio/PirateAudioHAT/README.md

fredg02, 2020-05-17:
I'm using a Pirate Audio HAT with Phoniebox. Sound should work fine after running the setup script (https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/audio/PirateAudioHAT/setup_pirateAudioHAT.sh). I had no time to setup and test the display and the buttons yet.

drocx, 2020-05-17:
Hello,

i have received my Pirate Audio HAT and everything is working (Sound, Display and Buttons).

Here are my installation steps:

1. Install the phoniebox develop
2. Run `components/audio/PirateAudioHAT/setup_pirateAudioHAT.sh`
3. Install plugins
 `sudo pip3 install Mopidy-PiDi pidi-display-pil pidi-display-st7789 mopidy-raspberry-gpio`
4. Edit mopidy configuration 
`sudo nano /etc/mopidy/mopidy.conf`

Add the following sections:
```
[raspberry-gpio]
enabled = true
bcm5 = play_pause,active_low,150
bcm6 = volume_down,active_low,150
bcm16 = next,active_low,150
bcm20 = volume_up,active_low,150

[pidi]
enabled = true
display = st7789
```
5. Enable access for modipy user
`sudo usermod -a -G spi,i2c,gpio,video mopidy`

6. Reboot

Maybe someone can upgrade the setup_pirateAudioHAT.sh script for a all in one installation.

Have fun with Pirate Audio 🏴‍☠️ 🤘

s-martin, 2020-05-17:
Thanks for the howto. I agree we should add this to the install script.

drocx, 2020-05-17:
I checked the setup_pirateAudioHAT.sh script. 
Maybe line 24 should be change from `gpio=25=op,dhc` to `gpio=25=op,dh` as mentioned [here](https://github.com/pimoroni/pirate-audio/tree/master/mopidy#manual-setup).
But I don‘t know what the additional `c` changes with the gpio control.

fredg02, 2020-05-17:
> I checked the setup_pirateAudioHAT.sh script.
> Maybe line 24 should be change from `gpio=25=op,dhc` to `gpio=25=op,dh` as mentioned [here](https://github.com/pimoroni/pirate-audio/tree/master/mopidy#manual-setup).
> But I don‘t know what the additional `c` changes with the gpio control.

This was most likely a typo or copy&paste error. I will fix the script accordingly.

drocx, 2020-05-18:
@fredg02 I have fixed the typo. 
Maybe you can extend the script by the steps I mentioned (plug-ins, mopidy.conf, Access rights)?
Unfortunately I am not very familiar with writing Linux scripts.

fredg02, 2020-05-18:
Thanks. I've added the steps you were mentioning. AFAICT SPI also needs to be enabled for this to work.

drocx, 2020-05-18:
Ok. I have updated my [post](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/950#issuecomment-629754181):
`sudo raspi-config nonint do_spi 0`

s-martin, 2020-05-22:
Fixed with PR #966

closed 2020-05-22 by s-martin