MiczFlor/RPi-Jukebox-RFID #951: Phoniebox starts playing last playlist after poweron

RalfAlbers, 2020-05-16:
Phoniebox starts playing my last playlist after poweron. Is there a way to wait on a swipe before starts playing?

I'm running last develop branch on my Pi 3b+ with an rc522.

s-martin, 2020-05-17:
Do you run classic or Spotify edition?

RalfAlbers, 2020-05-17:
The classic edition

s-martin, 2020-05-17:
Can you post the output of `tail -n 1000 /var/log/syslog`?

BerniPi, 2020-05-17:
@RalfAlbers: How do you power off the Phoniebox? Do you use the Shutdown in the Webinterface or a hardware button?

RalfAlbers, 2020-05-17:
I've tested a little bit. I can see this behavior only when the shutdown is triggered by "sudo reboot" / "sudo poweroff". When I shutdown the box via the WebUI, then no playback is resumed. In the debug.log I see no unusual entries. I've attached the log.

[debug.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4640419/debug.log)

A minute after swipe the card I've restarted the box.

It also doesn't matter if the last playlist is a Livestream or local mp3's. 

@s-martin: Syslog is also attached.
[syslog.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4640438/syslog.txt)

jo-me, 2020-06-18:
Is this maybe related to the rfid daemon being started without cleanups happening on shutdown?
I noticed that the current playlist is started (or resumed?) when restarting the rfid daemon manually.

s-martin, 2020-06-18:
@jo-me that sounds very likely.

In that case I don't think there's much we can do, because we can't hook into `sudo reboot` to trigger a cleanup.

jo-me, 2020-06-18:
Why not clean on startup then?

s-martin, 2020-06-18:
> Why not clean on startup then? 

Could be a possibility. @MiczFlor extended the startup script a while ago, so maybe this could also cleanup.

MiczFlor, 2020-06-18:
@s-martin @jo-me 
This is the startup script:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/scripts/startup-scripts.sh
As you can see, the move to run a script, not just the startup sound has allowed us to add a few things on startup, like rescan the mpd library, setting the startup volume.
This is also where the *rfid daemon being started without cleanups happening on shutdown* issue could be added. I am not sure though, what that means :) What would a cleanup do? What's the script?

MrReSc, 2020-12-11:
Is there already a solution for this problem? I have the same behaviour. See issue #1189

s-martin, 2022-12-24:
Closing for now as there was no activity for a long time.
Feel free to reopen a new issue, if the issue persists with current version 2.4 (or beta 3.2).

closed 2022-12-24 by s-martin