MiczFlor/RPi-Jukebox-RFID #952: 🚀 | CMDMAXVOLxx only until next boot?

manajoe, 2020-05-16:
## Feature Description
Non-Reboot persistent Maximum Volume. MaxVol is helpful for the box as a "sleeping aid", but the next day full power should be allowed again :-)

### What functionality would you like to see in your phoniebox?

I can set a maximum volume via RFID with CMDMAXVOLxx in rfid_trigger_play.conf. I'd like that to only be active until the next reboot. Right now this settings persists, as far as I understand, because "Max_Volume_Limit"-file is created under RPi-Jukebox-Settings.
Or maybe explained the other way round: I'd like to have MAXVOL100 at startup, regardless if there were different settings before.

### Further information that might help

I know I could create a second RFID-Card to set the MAXVOL back to 100. Or put a command in any of the startup-services. But maybe that request suits this issue: https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/923#issue-412331091 and could be implemented similarly.

Manajoe

MiczFlor, 2020-05-19:
Hi @manajoe 
Thanks for your suggestion. Adding a temporary feature is possible, but in my opinion the work will outrun the benefit. There are two options that can be assigned to cards: the maximum volume and changing the current volume.

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/settings/rfid_trigger_play.conf.sample

I recommend using
~~~
CMDVOL75="%CMDVOL75%"
~~~
instead of 
~~~
CMDMAXVOL75="%CMDMAXVOL75%"
~~~
to temporarily change the volume - and move the box far enough away so they can't touch the buttons :)

closed 2020-05-19 by MiczFlor

manajoe, 2020-05-29:
Hi @MiczFlor,
thanks for your suggestion, but I tried to implement it nevertheless :-)
As this is my first attempt to contribute more than a translation, I didn't dare to create a PR right now.
It's mostly copy/paste and slight adjustment to existing code. If anybody is interested, you can check my changes here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/compare/develop...manajoe:maxvolumetemp

I created a separate function "MaxVolTemp", which will be reset after the next boot by the startup-script.sh.
In my tests it worked even in combination with classical "MaxVol". You can still set a MaxVol to 80 and afterwards a MaxVolTemp to 50, so during this session it's not possible to set volume higher than 50. After a reboot, you still have MaxVol 80.

Maybe this could even be used for this https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/935 , if one manages to repeat it (cron?)...
Would prevent the kids from raising the volume manually if it was meant to lower automatically as - at least my - kids could try to :-)