MiczFlor/RPi-Jukebox-RFID #954: WLAN not working after installation

OnkelOnk, 2020-05-16:
Hello,

I tried to setup a Phoniebox. I connect the Raspberry 4 to WiFi and run the installation script. After I finished and the system is rebooted it is not able to connect to the WiFi anymore. I copied the accces information directly from the web frontend of my router. So I‘m 100% shure that there are correct. 

The only thing that I did not know before is the country code for my WLan. I used DE because I‘m from Germany. Where do I find the correct country code?

I would be super happy if you could help me to solve this issue.

Best Regards, Dirk

fredg02, 2020-05-17:
> The only thing that I did not know before is the country code for my WLan. I used DE because I‘m from Germany. Where do I find the correct country code?

DE is the correct country code for Germany.

If you managed to setup the WiFi before running the installation script, I would recommend to skip the WiFi configuration during the installation. If you absolutely need a static IP, check the following documentation: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#set-a-static-ip-address-for-your-rpi

OnkelOnk, 2020-05-17:
Thanks for the quick answer. Setup without WiFi worked. I was able to upload an mp3 file and start playing it by using an rfid tag. Unfortunately I ran into the next problem. I tried to setup a Spotify album. I used the Spoitify menu of the phoniebox webfrontend to get the url and added it like described in the tutorial video. I can see the cover of the album at the phoniebox home screen but it doesn’t start playing. Not by tag neither by pressing the play button. 
[image]

s-martin, 2020-05-17:
So audio already worked with local mp3?

Did you install from master or develop branch?

Can you post the output of `journalctl -u mopidy`?

OnkelOnk, 2020-05-17:
Yes Mp3 output worked finde. Audible worked. I ninstalled Phoniebox via the recommended one line installer: (copied from command line hinstory)

cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh

This is the output you asked for: 

-- Logs begin at Sun 2020-05-17 13:26:31 CEST, end at Sun 2020-05-17 13:51:19 CEST. --
May 17 13:26:35 raspberrypi systemd[1]: Starting Mopidy music server...
May 17 13:26:35 raspberrypi systemd[1]: Started Mopidy music server.
May 17 13:26:39 raspberrypi mopidy[501]: /usr/lib/python2.7/dist-packages/mopidy/ext.py:202: PkgResourcesDeprecationWarning: Parameters to load are deprecated.  Call .resolve and .require separately.
May 17 13:26:39 raspberrypi mopidy[501]:   extension_class = entry_point.load(require=False)
May 17 13:26:40 raspberrypi mopidy[501]: INFO     Starting Mopidy 2.2.2
May 17 13:26:40 raspberrypi mopidy[501]: INFO     Loading config from builtin defaults
May 17 13:26:40 raspberrypi mopidy[501]: INFO     Loading config from /etc/mopidy/mopidy.conf
May 17 13:26:40 raspberrypi mopidy[501]: INFO     Loading config from command line options
May 17 13:26:40 raspberrypi mopidy[501]: INFO     Enabled extensions: iris, mpd, http, stream, spotify, m3u, softwaremixer, file, local-images, local
May 17 13:26:40 raspberrypi mopidy[501]: INFO     Disabled extensions: none
May 17 13:26:40 raspberrypi mopidy[501]: INFO     Starting Mopidy mixer: SoftwareMixer
May 17 13:26:40 raspberrypi mopidy[501]: INFO     Mixer volume set to 30
May 17 13:26:40 raspberrypi mopidy[501]: INFO     Starting Mopidy audio
May 17 13:26:40 raspberrypi mopidy[501]: INFO     Starting Mopidy backends: SpotifyBackend, StreamBackend, M3UBackend, FileBackend, LocalBackend
May 17 13:26:40 raspberrypi mopidy[501]: INFO     Loaded 2 local tracks using json
May 17 13:26:40 raspberrypi mopidy[501]: INFO     Audio output set to "alsasink"
May 17 13:26:40 raspberrypi mopidy[501]: ERROR    Spotify login error: <ErrorType.OTHER_PERMANENT: 10>
May 17 13:26:42 raspberrypi mopidy[501]: ERROR    OAuth token refresh failed: Unknown error.
May 17 13:26:42 raspberrypi mopidy[501]: ERROR    Failed to load Spotify user profile
May 17 13:26:42 raspberrypi mopidy[501]: INFO     Started spotify backend
May 17 13:26:42 raspberrypi mopidy[501]: INFO     Loaded 0 playlists
May 17 13:26:42 raspberrypi mopidy[501]: INFO     Starting Mopidy core
May 17 13:26:42 raspberrypi mopidy[501]: INFO     Refresh Playlists took 12ms
May 17 13:26:42 raspberrypi mopidy[501]: INFO     Starting Mopidy frontends: IrisFrontend, MpdFrontend, HttpFrontend
May 17 13:26:42 raspberrypi mopidy[501]: INFO     Starting Iris 3.43.0
May 17 13:26:42 raspberrypi mopidy[501]: INFO     MPD server running at [::ffff:0.0.0.0]:6600
May 17 13:26:42 raspberrypi mopidy[501]: INFO     Starting GLib mainloop
May 17 13:26:42 raspberrypi mopidy[501]: INFO     HTTP server running at [::ffff:0.0.0.0]:6680
May 17 13:38:20 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:39:51 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:39:52 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:39:54 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:39:55 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:39:55 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:39:55 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:39:55 raspberrypi mopidy[501]: INFO     Failed to lookup "spotify:album:1ooR7lBK5YtmqPN0nXAxLg": Session must be logged in and online to load objects: <ConnectionState.LOGGED_OUT: 0>
May 17 13:39:55 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:39:55 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:39:55 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:39:56 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:39:56 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:39:57 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:40:02 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:40:07 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:40:10 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:40:10 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:40:10 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:40:10 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:40:10 raspberrypi mopidy[501]: INFO     Failed to lookup "spotify:album:1ooR7lBK5YtmqPN0nXAxLg": Session must be logged in and online to load objects: <ConnectionState.LOGGED_OUT: 0>
May 17 13:40:10 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:40:10 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
[... cut ...]
May 17 13:50:54 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:50:59 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:51:04 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:51:09 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:51:14 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600
May 17 13:51:19 raspberrypi mopidy[501]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600

OnkelOnk, 2020-05-17:
I first thought that I used the wrong login credetials but I can see my own playlists in the spotify menue of the phoniebox. Therby I think my credentials are correct. 

Thanks for your help.

s-martin, 2020-05-17:
Try `sudo systemctl restart mopidy`.

OnkelOnk, 2020-05-17:
Worked :). But after a restart of the PI I have to run the command again in order to make it work. Is there a permant way to fix that?

s-martin, 2020-05-17:
We have currently problem #893

In this issue is a workaround described you can use, if you have always a WiFi connection when you use your Phoniebox.

OnkelOnk, 2020-05-17:
WiFi only is no problem for me. I ran `sudo raspi-config` and use the menu "3 Boot Options" -> B2 "Wait for Network at Boot" -> Yes. It is not exacly waht is described in #893 but probably because the manue changed. 

The result is that I don't get an error anymore when I run `journalctl -u mopidy ` after the start of the PI. 

-- Logs begin at Sun 2020-05-17 17:01:01 CEST, end at Sun 2020-05-17 17:11:02 CEST. --
May 17 17:01:16 raspberrypi systemd[1]: Starting Mopidy music server...
May 17 17:01:16 raspberrypi systemd[1]: Started Mopidy music server.
May 17 17:01:20 raspberrypi mopidy[644]: /usr/lib/python2.7/dist-packages/mopidy/ext.py:202: PkgResourcesDeprecationWarning: Parameters to load are deprecated.  Call .resolve and .require separately.
May 17 17:01:20 raspberrypi mopidy[644]:   extension_class = entry_point.load(require=False)
May 17 17:01:21 raspberrypi mopidy[644]: INFO     Starting Mopidy 2.2.2
May 17 17:01:21 raspberrypi mopidy[644]: INFO     Loading config from builtin defaults
May 17 17:01:21 raspberrypi mopidy[644]: INFO     Loading config from /etc/mopidy/mopidy.conf
May 17 17:01:21 raspberrypi mopidy[644]: INFO     Loading config from command line options
May 17 17:01:21 raspberrypi mopidy[644]: INFO     Enabled extensions: iris, mpd, http, stream, spotify, m3u, softwaremixer, file, local-images, local
May 17 17:01:21 raspberrypi mopidy[644]: INFO     Disabled extensions: none
May 17 17:01:21 raspberrypi mopidy[644]: INFO     Starting Mopidy mixer: SoftwareMixer
May 17 17:01:21 raspberrypi mopidy[644]: INFO     Mixer volume set to 30
May 17 17:01:21 raspberrypi mopidy[644]: INFO     Starting Mopidy audio
May 17 17:01:21 raspberrypi mopidy[644]: INFO     Starting Mopidy backends: SpotifyBackend, StreamBackend, M3UBackend, FileBackend, LocalBackend
May 17 17:01:22 raspberrypi mopidy[644]: INFO     Loaded 2 local tracks using json
May 17 17:01:22 raspberrypi mopidy[644]: INFO     Audio output set to "alsasink"
May 17 17:01:22 raspberrypi mopidy[644]: INFO     Logged in to Spotify in online mode
May 17 17:01:22 raspberrypi mopidy[644]: INFO     Logged into Spotify Web API as theonkelonk
May 17 17:01:22 raspberrypi mopidy[644]: INFO     Started spotify backend
May 17 17:01:22 raspberrypi mopidy[644]: INFO     Starting Mopidy core
May 17 17:01:22 raspberrypi mopidy[644]: INFO     Starting Mopidy frontends: IrisFrontend, MpdFrontend, HttpFrontend
May 17 17:01:22 raspberrypi mopidy[644]: INFO     Starting Iris 3.43.0
May 17 17:01:22 raspberrypi mopidy[644]: INFO     MPD server running at [::ffff:0.0.0.0]:6600
May 17 17:01:22 raspberrypi mopidy[644]: INFO     HTTP server running at [::ffff:0.0.0.0]:6680
May 17 17:01:22 raspberrypi mopidy[644]: INFO     Starting GLib mainloop
May 17 17:02:03 raspberrypi mopidy[644]: INFO     Loaded 187 playlists
May 17 17:02:24 raspberrypi mopidy[644]: INFO     Refresh Playlists took 61753ms

Unfortunately it is not working after the restart with no errors in the log. I have to restart manualy again to make it work. After `sudo systemctl restart mopidy` via shell it works.

**Summary:** I made the change to the boot setup. The errors in the log are gone but I still have to restart the server manualy to make it work.

s-martin, 2022-12-24:
Closing for now as there was no activity for a long time.
Feel free to reopen a new issue, if the issue persists with current version 2.4 (or beta 3.2).

closed 2022-12-24 by s-martin