MiczFlor/RPi-Jukebox-RFID #956: 🚀 | Create HotSpot if no network connection is available

RalfAlbers, 2020-05-17:
## Feature Description

### What functionality would you like to see in your Phoniebox?

If you are on a trip with your box an no known SSID is available, by now you have to open the box an connect with monitor an keyboard to the box. If an HotSpot is created by the Box you can easily configure the new WLAN-Connection with your Smartphone.

### Further information that might help

I found a project ( https://www.raspberryconnect.com/projects/65-raspberrypi-hotspot-accesspoints/157-raspberry-pi-auto-wifi-hotspot-switch-internet ) that is providing such a function. Maybe this can help.

MiczFlor, 2020-05-19:
Hi @RalfAlbers 
I have been playing with this hot spot issue about a year ago on stretch. And at some point I wanted to throw the Pi out of the window :) it did not work reliably. This might have changed. 
My script is still here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/scripts/installscripts/stretch-install-default-HotspotAddOn.sh
If you have some time, please feel free to add, edit and test it. I might come back to this when I have spare time.
Thanks, all the best, micz

MiczFlor, 2020-05-22:
In the `develop` branch, we just merged the code of @Groovylein :
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/967
any testers out there who want to have a shot? And post feedback here?
Please note: for the time being there is a special install script for the hotspot version:
https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/scripts/installscripts

RalfAlbers, 2020-05-28:
during my testings with the last develop-branch I had issues with the wlan connectivity after configure a connection via WebUI. 
My Step (network-connection is WLAN only):
- Setup 2020-02-13-raspbian-buster-lite on SD with `wpa_supplicant.conf` (psk is wpa_passphrase encrypted)
- install last develop-branch with autohotspot with oneliner `cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-default-with-autohotspot.sh; chmod +x buster-install-default-with-autohotspot.sh; GIT_BRANCH=develop bash ./buster-install-default-with-autohotspot.sh`
- reboot
- WiFi Connection in WebUI is empty, so I configure my current WiFi
- reboot
- The box is not able to connect to my wlan anymore, until I setup the wlan cleartext-pass in `wpa_supplicant.conf`. Positiv: Autohotspot make a connection with wifi possible :-)

In the logs I can not find anything why the connection with wpa_passphrase fails.
Any ideas?

Groovylein, 2020-05-28:
@RalfAlbers ,
can you maybe provide the output of `sudo /usr/bin/autohotspot`?

Piehti, 2020-06-04:
I got a Phonie with Accesspoint on it. Works like a charm. But it has no killswitch.
Yet I got a Phonie as VPN-Router running which actually has a killswitch somewhere.

Not sure when I'll have time to look into it, but I can propably reverse and transfer the killswitch to "no wifi, then activate hotspot".

Yet I'm not sure if it wasn't more clever just to keep the hotspot running 24/7 when Wi-Fi runs, and bridge the connection from wlan0 <> wlan1. Like that you could eighter connect through a Wi-Fi-Router or directly into the Pi-Wi-Fi.

Unfortunatly not all PI's can do that (AFAIK). Most will need a second Wi-Fi-Dongle.

> I have been playing with this hot spot issue about a year ago on stretch. And at some point I wanted to throw the Pi out of the window :)
@MiczFlor: I suffered the same feelings for weeks until the AP worked OR / AND the VPN with no DNS leaking. ^^;

 
> In the logs I can not find anything why the connection with wpa_passphrase fails.
> Any ideas?
@RalfAlbers I'd be interested in your:

/etc/dhcpcd.conf
/etc/hostapd/hostapd.conf
/etc/dnsmasq.conf
/etc/resolv.conf

You might want to strike out sensible data. I could imagine, several settings are overridden several times due higher priorized services. Propably the killswitch forgets some points to change or got denyed - like when changing resolv.conf manually.

Best regards
Pete

RalfAlbers, 2020-06-06:
@Groovylein & @Piehti I've attached the requested files. In the `dhcpcd.conf` I've comment the line `nohook wpa_supplicant` to had a faster connection to my AP. In my testing the setting has no impact of the connectivity via wpa_passphrase.

[autohotspot.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4739776/autohotspot.txt)
[dhcpcd.conf.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4739777/dhcpcd.conf.txt)
[dnsmasq.conf.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4739778/dnsmasq.conf.txt)
[hostapd.conf.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4739779/hostapd.conf.txt)
[iwconfig.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4739780/iwconfig.txt)
[resolv.conf.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4739781/resolv.conf.txt)

Piehti, 2020-06-09:
Sry for keep ya waiting mate. I really wish I had morr time for this Praject. Well exams arr through next week. Then ill be right back.

Groovylein, 2020-07-14:
@RalfAlbers ,
sorry for letting you wait.
I had this problem too, so I disabled the power management. The web interface is enabling the powermanagement after connecting to the WLAN. 😞 I need to investigate, how the WLAN change happen.

See pull req.:
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/967/commits/be323ea434df2bc030bc7881707b1edbb2e4c3e7

What you can do and retry:
`sudo iw wlan0 set power_save off`

Erechtheus, 2020-08-25:
Hey folks! I also tried to install the HotSpot function but 
`sudo /usr/bin/autohotspot`
returns:
`failed to start hostapd.service unit hostapd.service is masked`.

I tried to resolve this by:

> sudo systemctl unmask hostapd
> sudo systemctl enable hostapd
> sudo systemctl start hostapd

but it's still not working. Any ideas? I am working on a Raspberry Pi Zero W.

Groovylein, 2020-08-25:
Good morning @Erechtheus,

That's interesting. I've done the implementation months ago but I can not imagine having this problem. What does the status of the service say?

Erechtheus, 2020-08-25:
Dear @Groovylein thanks for answering so quickly. Attached you can find some outputs:

`sudo /usr/bin/autohotspot`

> SSid's in range:  BSS 78:81:02:ab:12:86(on wlan0) SSID: EasyBox-444242 BSS 2c:3a:fd:c8:ba:fc(on wlan0) SSID: Valle Gran Rey
> Device Available Check try  0
> Device Available, checking SSid Results
> No SSid found, assessing WiFi status
> Creating Hotspot
> RTNETLINK answers: File exists
> Job for hostapd.service failed because the control process exited with error code.
> See "systemctl status hostapd.service" and "journalctl -xe" for details.

`sudo systemctl status hostapd.service`

> ● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
>    Loaded: loaded (/lib/systemd/system/hostapd.service; disabled; vendor preset: enabled)
>    Active: activating (auto-restart) (Result: exit-code) since Sun 2020-08-23 21:03:45 CEST; 802ms ago
>   Process: 7210 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=exited, status=1/FAILURE)

`journalctl -xe`

```
> -- Support: https://www.debian.org/support
> -- 
> -- The unit hostapd.service has entered the 'failed' state with result 'exit-code'.
> Aug 23 21:03:59 raspberrypi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
> -- Subject: A start job for unit hostapd.service has failed
> -- Defined-By: systemd
> -- Support: https://www.debian.org/support
> -- 
> -- A start job for unit hostapd.service has finished with a failure.
> -- 
> -- The job identifier is 57975 and the job result is failed.
> Aug 23 21:04:01 raspberrypi systemd[1]: hostapd.service: Service RestartSec=2s expired, scheduling restart.
> Aug 23 21:04:01 raspberrypi systemd[1]: hostapd.service: Scheduled restart job, restart counter is at 643.
> -- Subject: Automatic restarting of a unit has been scheduled
> -- Defined-By: systemd
> -- Support: https://www.debian.org/support
> -- 
> -- Automatic restarting of the unit hostapd.service has been scheduled, as the result for
> -- the configured Restart= setting for the unit.
> Aug 23 21:04:01 raspberrypi systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
> -- Subject: A stop job for unit hostapd.service has finished
> -- Defined-By: systemd
> -- Support: https://www.debian.org/support
> -- 
> -- A stop job for unit hostapd.service has finished.
> -- 
> -- The job identifier is 58036 and the job result is done.
> Aug 23 21:04:01 raspberrypi systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator...
> -- Subject: A start job for unit hostapd.service has begun execution
> -- Defined-By: systemd
> -- Support: https://www.debian.org/support
> -- 
> -- A start job for unit hostapd.service has begun execution.
> -- 
> -- The job identifier is 58036.
> Aug 23 21:04:01 raspberrypi hostapd[7279]: Configuration file: /etc/hostapd/hostapd.conf
> Aug 23 21:04:01 raspberrypi hostapd[7279]: Cannot enable IEEE 802.11d without setting the country_code
> Aug 23 21:04:01 raspberrypi hostapd[7279]: 1 errors found in configuration file '/etc/hostapd/hostapd.conf'
> Aug 23 21:04:01 raspberrypi hostapd[7279]: Failed to set up interface with /etc/hostapd/hostapd.conf
> Aug 23 21:04:01 raspberrypi hostapd[7279]: Failed to initialize interface
> Aug 23 21:04:01 raspberrypi systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
> -- Subject: Unit process exited
> -- Defined-By: systemd
> -- Support: https://www.debian.org/support
> -- 
> -- An ExecStart= process belonging to unit hostapd.service has exited.
> -- 
> -- The process' exit code is 'exited' and its exit status is 1.
> Aug 23 21:04:01 raspberrypi systemd[1]: hostapd.service: Failed with result 'exit-code'.
> -- Subject: Unit failed
> -- Defined-By: systemd
> -- Support: https://www.debian.org/support
> -- 
> -- The unit hostapd.service has entered the 'failed' state with result 'exit-code'.
> Aug 23 21:04:01 raspberrypi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
> -- Subject: A start job for unit hostapd.service has failed
> -- Defined-By: systemd
> -- Support: https://www.debian.org/support
> -- 
> -- A start job for unit hostapd.service has finished with a failure.
> -- 
> -- The job identifier is 58036 and the job result is failed.
```

Groovylein, 2020-08-25:
Hey, 
It looks like there is an error in

`/etc/hostapd/hostapd.conf`

Looks like the country code is not valid, but I am not sure. 

Did you change anything in this config file?

aberfisch, 2020-08-25:
Would be great to have in addition a captive portal which opens after Wifi join directly the phoniebox page. Would be pretty user-friendly. Cheers! QR Code Scan --> wifi join --> captive portal splash with forward to phoniebox page.

Erechtheus, 2020-08-25:
@Groovylein good call! 

First, indeed the country code was false (i.e., empty). I assume the reason is that I did not configure the WIFI in the Jukebox, but rather used the WIFI configuration of the operating system?

Second, no I did not modify the config file.

Groovylein, 2020-08-25:
@Erechtheus,
I need look into the instalation script but currently I am on vacation and don't have my laptop with me 😅

Does the script now works? If not can you provide once again information from the journal?

Cheers

Erechtheus, 2020-08-26:
Dear @Groovylein !
No worries. I am also on vacation starting today. My last status was that I can see the HotSpot network, but I cannot connect with any device to the Wifi. Mobile was waiting for an IP-address and my laptop did not provide me with more information. But I will also look into this again. Thank you so far..

s-martin, 2023-05-02:
Closing as this feature was implemented a long time ago.

closed 2023-05-02 by s-martin