MiczFlor/RPi-Jukebox-RFID #961: Phoniebox stuck on loop list

marmacman, 2020-05-18:
Our phoniebox has been working fantastic for half a year, this week it began to loop every playlist that we are listening to.  When i try to turn off loop list on the webapp it just keeps reverting to loop list.  Anyone have any ideas.  Thank you!  

version 2.0 rc6
Editiion Classic.

s-martin, 2020-05-18:
We have been fixing some things with repeat function recently (see develop branch), but I'm not sure why this would happen suddenly to you.

Can you post results from `tail -n 500 /var/log/syslog`?

manajoe, 2020-05-18:
I also had that issue recently, that I couldn't end the loop.
If I remember correctly I klicked the "loop button" in the web interface multiple times and set it to "loop track" (or "loop list", but definetely _NOT_ to "off", as I had tried several times before) and reloaded the web interface. Afterwards I think the problem was solved.
Alternatively, maybe that helps?
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/938#issuecomment-626312131

marmacman, 2020-05-19:
Thanks for the help!  It seems to have cleared up the problem for now!  Strange that it should suddenly become an issue?  I am still learning about all these kind of things.  The phoniebox was my introduction into raspberry pi/linux/coding.

closed 2020-05-19 by marmacman