MiczFlor/RPi-Jukebox-RFID #962: 🐛 | BUG SUMMARY

MiczFlor, 2020-05-19:
## Bug

Installing latest `develop` and/or `master` branch with USB RFID reader and the service does not start after reboot. 

The hardware has been used before on a Pi 2B and worked before on the Pi 3B I am testing on right now.

It's a bit embarrassing for me to post this bug... I should fix it, not post it, but have no clue what's going on...

### What I did

* Fresh SD card with Buster, run the buster install script, tried both: master and develop
* Registered the USB RFID reader with the script - it showed up as `Barcode Reader`
* the devixe.txt file shows the correct name

### What happened

The RFID reader did not work, in more detail: the rfid-reader service can be enabled but then throws an error that the device is not connected (see below).

* the error exit of `scripts/daemon-rfid-reader.py` shows the correct device name (see below results of systemctl status)
  * `May 19 13:35:42 raspberrypi daemon_rfid_reader.py[28391]: Could not find the device "Barcode Reader". Make sure is connected`
  * This error message comes from `Reader.py`

* the RFID reader was plugged in and beeped when I swiped a card. The card ID shows up correctly in the terminal

### I expected this to happen

Phoniebox working with USB RFID reader out of the box

### Further information that might help
~~~
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ sudo systemctl restart phoniebox-rfid-reader
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ sudo systemctl status phoniebox-rfid-reader
● phoniebox-rfid-reader.service - Phoniebox RFID-Reader Service
   Loaded: loaded (/etc/systemd/system/phoniebox-rfid-reader.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-05-19 13:35:41 CEST; 729ms ago
 Main PID: 28391 (python3)
    Tasks: 1 (limit: 2200)
   Memory: 5.1M
   CGroup: /system.slice/phoniebox-rfid-reader.service
           └─28391 python3 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py

May 19 13:35:41 raspberrypi systemd[1]: Started Phoniebox RFID-Reader Service.
May 19 13:35:42 raspberrypi daemon_rfid_reader.py[28391]: Could not find the device "Barcode Reader". Make sure is connected
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ sudo systemctl status phoniebox-rfid-reader
● phoniebox-rfid-reader.service - Phoniebox RFID-Reader Service
   Loaded: loaded (/etc/systemd/system/phoniebox-rfid-reader.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-05-19 13:35:44 CEST; 29s ago
  Process: 28404 ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py (code=exited, status=1/FAILURE)
 Main PID: 28404 (code=exited, status=1/FAILURE)

May 19 13:35:44 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Service RestartSec=100ms expired, scheduling restart.
May 19 13:35:44 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Scheduled restart job, restart counter is at 5.
May 19 13:35:44 raspberrypi systemd[1]: Stopped Phoniebox RFID-Reader Service.
May 19 13:35:44 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Start request repeated too quickly.
May 19 13:35:44 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
May 19 13:35:44 raspberrypi systemd[1]: Failed to start Phoniebox RFID-Reader Service.
~~~

## Software

* Buster
* 2.0 - 0262de2 - develop

### Installscript
 
`scripts/installscripts/buster-install-default.sh`

## Hardware

Pi 3 B
USB RFID reader `Barcode Reader`

MiczFlor, 2020-05-19:
... no comment ... just stupidity on my part :)

MiczFlor, 2020-05-19:
Looks like in line 42, the original `develop` is like this:
~~~
                deviceName = f.read().rstrip()
~~~
and reverting this line to `master`, it works. Looking like this:
~~~
                deviceName = f.read()
~~~
In the device txt file there is a line break at the end of the RFID device name. Apparently that is needed to make the script work. I removed the trimming and committed the file anew.

closed 2020-05-19 by MiczFlor

s-martin, 2020-05-19:
Wow, sometimes even whitespaces are dangerous :)

fredg02, 2020-05-19:
@MiczFlor  @s-martin Hmm. That's really weird. `RegisterDevice.py` does not add a white space or line break after the name of the device in `deviceName.txt` and it works fine.

When I tested it, I used `echo` to write to `deviceName.txt` (see here: https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/rfid-reader/PN532/setup_pn532.sh#L44) which **added a line break. This broke my installation**. Therefore I added `rstrip()` in #943 so users would not need to care about a white space or line break.