MiczFlor/RPi-Jukebox-RFID #973: Vol Up Down is not working

htpzc, 2020-05-22:
## Bug
Volume Up and Down brings error

### What I did
Fresh Install with Master Branch.
No Spotify. 
Hold it simple
GPIO Buttons are default configured.
All other buttons work fine (forward, back, play/pause)

### What happened
Everything works fine, but vol up down buttons dont work

### I expected this to happen
vol turns up and down

### Further information that might help
Push Button
/var/log/syslog
May 22 20:15:43 raspberrypi gpio-buttons.py[472]: expr: syntax error: unexpected argument '11'
May 22 20:15:43 raspberrypi gpio-buttons.py[472]: ./scripts/playout_controls.sh: line 244: [: -le: unary operator expected
May 22 20:15:43 raspberrypi gpio-buttons.py[472]: OK MPD 0.21.4
May 22 20:15:43 raspberrypi gpio-buttons.py[472]: OK

Push other button
May 22 20:17:03 raspberrypi gpio-buttons.py[472]: expr: syntax error: unexpected argument '('
May 22 20:17:03 raspberrypi gpio-buttons.py[472]: ./scripts/playout_controls.sh: line 281: [: -ge: unary operator expected
May 22 20:17:03 raspberrypi gpio-buttons.py[472]: OK MPD 0.21.4
May 22 20:17:03 raspberrypi gpio-buttons.py[472]: OK

## Software

### Base image and version
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
### Branch / Release

s-martin, 2020-05-24:
Are you using the spotify edition or the classic edition?

We released 2.0 just moments ago, this release should fix many, many issues, so you could give that a try.

topas-rec, 2020-05-29:
I updated my packages in apt. Afterwards my volume buttons also don't work anymore.
I reinstalled Phoniebox software through the buster script without success. (Regarding volume control)
I have the same errors in system log. (With slightly different line numbers. My error is in line 238 of playout controls and slightly different unexpected argument errors. My unexpected argument is '3' and a parenthesis - like above)

Changing the volume in the web app also does not work.

How can I debug this?

Manually changing PCM in alsamixer of course works.

topas-rec, 2020-05-29:
I can try to make a clean install tomorrow.
Otherwise I'll have to roll back my sdcard backup image from before the update.

dnmtch, 2020-05-30:
Hi
i'm playing around with a new installation of version 2 on a raspbery pi 3. unfortunately i can't get my already successfully used rotary encoders to work.

- installation on Buster lite led to several "bugs" 
- installation on buster desktop worked, (nano ~/.config/phoniebox/gpio_settings.ini can be executed successfully)

-------------------
[install_log.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4704829/install_log.txt)

topas-rec, 2020-05-30:
I'm still in the progress of a fresh installation so wait for that, but I found many issues that show the issue: `unary operator expected` in the `playout_controls.sh`
See: https://github.com/MiczFlor/RPi-Jukebox-RFID/search?q=unary+operator+expected&type=Issues

Can that be related? Where is this thing coming from?

topas-rec, 2020-05-30:
Fresh installed finished and issue present (I did not check if that affects volume change, but i guess it does):

Volume up error message in syslog:
```
May 30 07:27:03 raspberrypi gpio-buttons.py[547]: expr: syntax error: unexpected argument '3'
May 30 07:27:03 raspberrypi gpio-buttons.py[547]: ./scripts/playout_controls.sh: line 238: [: -le: unary operator expected
May 30 07:27:03 raspberrypi gpio-buttons.py[547]: OK MPD 0.21.4
May 30 07:27:03 raspberrypi gpio-buttons.py[547]: OK
```
When pressing volume down:
```
May 30 07:27:08 raspberrypi gpio-buttons.py[547]: expr: syntax error: unexpected argument '('
May 30 07:27:08 raspberrypi gpio-buttons.py[547]: ./scripts/playout_controls.sh: line 276: [: -ge: unary operator expected
May 30 07:27:08 raspberrypi gpio-buttons.py[547]: OK MPD 0.21.4
May 30 07:27:08 raspberrypi gpio-buttons.py[547]: OK

```

s-martin, 2020-05-30:
I suspect a problem here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/7ab02865569df8dcf55c582bc066d03e1f128658/scripts/playout_controls.sh#L236

I don’t see were `AUDIOVOLCHANGESTEP` is defined?

@topas-rec could you also execute `echo -e status\\nclose | nc -w 1 localhost 6600 | grep -o -P '(?<=volume: ).*'` and post the result here?

I don’t know, if this part of the script is only executed with rotary knobs or if this issue exists for regular buttons also.

dnmtch, 2020-05-30:
If that helps I can offer to do fresh installations on a rasperry pi 3. My setup consists of 2 encoders and 1-n buttons.

i always get stuck with the configuration of gpio_settings.ini 
https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/gpio_control

[image]

topas-rec, 2020-05-30:
> @topas-rec could you also execute echo -e status\\nclose | nc -w 1 localhost 6600 | grep -o -P '(?<=volume: ).*' and post the result here?

That is not much (Tried it twice):
```
pi@raspberrypi:~ $ echo -e status\\nclose | nc -w 1 localhost 6600 | grep -o -P '(?<=volume: ).*'
pi@raspberrypi:~ $ echo -e status\\nclose | nc -w 1 localhost 6600 | grep -o -P '(?<=volume: ).*'
pi@raspberrypi:~ $ 
```

> I suspect a problem here:
RPi-Jukebox-RFID/scripts/playout_controls.sh
Line 236 in 7ab0286
 VOLPERCENT=`expr ${VOLPERCENT} + \( ${AUDIOVOLCHANGESTEP} \* ${VALUE} \)` 
I don’t see were AUDIOVOLCHANGESTEP is defined?

Might be. I have not enough knowledge of shell scripts to follow up on this.

> In addition I think we need to find out which dependencies are needed from Raspian Desktop, so they could be installed, if the user uses a Raspian Lite installation (which should be sufficient for Phoniebox).

That's an interesting idea. I want to install the desktop version next to see if the lite version (which I use) is the cause for this.

s-martin, 2020-05-30:
> That is not much (Tried it twice):
>
> ```
> pi@raspberrypi:~ $ echo -e status\\nclose | nc -w 1 localhost 6600 | grep -o -P '(?<=volume: ).*'
> pi@raspberrypi:~ $ echo -e status\\nclose | nc -w 1 [...]

Ok, I think there’s an issue. This should return a (volume) value between 0 and 100.

> > I suspect a problem here:
> > RPi-Jukebox-RFID/scripts/playout_controls.sh
> > Line 236 in [7ab0286](https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/7ab02865569df8dcf55c582bc066d03e1f128658)
> > VOLPERCENT=`expr ${VOLPERCENT} + \( ${AUDIOVOLCHANGESTEP} \* ${VALUE} \)`
> > I don’t see were AUDIOVOLCHANGESTEP is defined?
> 
> Might be. I have not enough knowledge of shell scripts to follow up on this.
> 
> > In addition I think we need to find out which dependencies are needed from Raspian Desktop, so they could be installed, if the user uses a Raspian Lite installation (which should be sufficient for Phoniebox).
> 
> That's an interesting idea. I want to install the desktop version next to see if the lite version (which I use) is the cause for this.
> BTW: I did not use the lite version for the first time. So lite version worked with Phoniebox software before 2.0.

Ok, so I guess lite vs. desktop version is probably not an issue.

topas-rec, 2020-05-30:
> Ok, so I guess lite vs. desktop version is probably not an issue.
So it does not help if I test the desktop version? If not I would restore my backup.

topas-rec, 2020-05-31:
Reverted to my backup where volume keys worked.
After updateing all installed packages the volume buttons don't work anymor with the issues mentioned above.
The following packages were part of the update. (I know that is a lot, but perhaps that helps)
```
The following NEW packages will be installed:
  busybox initramfs-tools initramfs-tools-core klibc-utils libklibc linux-base pigz rpi.gpio-common
The following packages will be upgraded:
  apt apt-transport-https apt-utils base-files bind9-host binutils binutils-arm-linux-gnueabihf
  binutils-common bluez bluez-firmware curl distro-info-data dphys-swapfile e2fsprogs exim4-base
  exim4-config exim4-daemon-light firmware-atheros firmware-brcm80211 firmware-libertas
  firmware-misc-nonfree firmware-realtek fuse git git-man guile-2.2-libs iputils-ping libapt-inst2.0
  libapt-pkg5.0 libbind9-161 libbinutils libboost-atomic1.67.0 libboost-iostreams1.67.0
  libboost-regex1.67.0 libboost-system1.67.0 libboost-thread1.67.0 libcom-err2 libcups2 libcurl3-gnutls
  libcurl4 libdns-export1104 libdns1104 libext2fs2 libfuse2 libgnutls-dane0 libgnutls30 libicu63 libidn2-0
  libisc-export1100 libisc1100 libisccc161 libisccfg163 libldap-2.4-2 libldap-common liblwres161
  libmariadb3 libmysofa0 libopenjp2-7 libpam-chksshpwd libpam-modules libpam-modules-bin libpam-runtime
  libpam-systemd libpam0g libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0 libpython3.7
  libpython3.7-dev libpython3.7-minimal libpython3.7-stdlib libraspberrypi-bin libraspberrypi-dev
  libraspberrypi-doc libraspberrypi0 libss2 libssh-gcrypt-4 libssl1.1 libsystemd0 libudev1 libunbound8
  mariadb-common openssh-client openssh-server openssh-sftp-server openssl php7.3 php7.3-cgi php7.3-cli
  php7.3-common php7.3-fpm php7.3-json php7.3-opcache php7.3-readline pi-bluetooth python-rpi.gpio
  python3-rpi.gpio python3.7 python3.7-dev python3.7-minimal raspberrypi-bootloader raspberrypi-kernel
  raspberrypi-kernel-headers raspberrypi-sys-mods raspi-config rpi-eeprom rpi-eeprom-images rpi-update ssh
  sudo systemd systemd-sysv tzdata udev wpasupplicant
115 upgraded, 8 newly installed, 0 to remove and 0 not upgraded.
```

topas-rec, 2020-05-31:
Tested the desktop version without success.
So volume buttons are no longer working in the current version (which is live).

topas-rec, 2020-05-31:
> > > @topas-rec could you also execute echo -e status\nclose | nc -w 1 localhost 6600 | grep -o -P '(?<=volume: ).*' and post the result here?
> > 
> > 
> > That is not much (Tried it twice):
> > ```
> > pi@raspberrypi:~ $ echo -e status\\nclose | nc -w 1 localhost 6600 | grep -o -P '(?<=volume: ).*'
> > pi@raspberrypi:~ $ echo -e status\\nclose | nc -w 1 localhost 6600 | grep -o -P '(?<=volume: ).*'
> > pi@raspberrypi:~ $ 
> > ```
> 
> Ok, I think there’s an issue. This should return a (volume) value between 0 and 100. @MiczFlor, I don’t know where the value is really polled from and which service/application should run here to make this work.
> In any case I this we should have a check and an error message here, so a meaningful value is used.
> 
> > > I suspect a problem here:
> > > RPi-Jukebox-RFID/scripts/playout_controls.sh
> > > Line 236 in [7ab0286](https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/7ab02865569df8dcf55c582bc066d03e1f128658)
> > > VOLPERCENT=`expr ${VOLPERCENT} + \( ${AUDIOVOLCHANGESTEP} \* ${VALUE} \)`
> > > I don’t see were AUDIOVOLCHANGESTEP is defined?
> > 
> > 
> > Might be. I have not enough knowledge of shell scripts to follow up on this.
> > > In addition I think we need to find out which dependencies are needed from Raspian Desktop, so they could be installed, if the user uses a Raspian Lite installation (which should be sufficient for Phoniebox).
> > 
> > 
> > That's an interesting idea. I want to install the desktop version next to see if the lite version (which I use) is the cause for this.
> > BTW: I did not use the lite version for the first time. So lite version worked with Phoniebox software before 2.0.
> 
> Ok, so I guess lite vs. desktop version is probably not an issue.

Issue is here also from my point of view:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/03e63de6669c49548c9246185f36554f245f7347/scripts/playout_controls.sh#L234

When I `echo -e status\\nclose | nc -w 1 localhost 6600` it says

```
OK MPD 0.21.4
repeat: 0
random: 0
single: 0
consume: 0
playlist: 2
playlistlength: 0
mixrampdb: 0.000000
state: stop
OK
```

when nothing is playing and else

```
OK MPD 0.21.4
repeat: 0
random: 0
single: 0
consume: 0
playlist: 4
playlistlength: 42
mixrampdb: 0.000000
state: play
song: 0
songid: 1
time: 14:149
elapsed: 14.354
bitrate: 320
duration: 149.368
audio: 44100:24:2
nextsong: 1
nextsongid: 2
OK
```

Nothing about volume which is piped to grep to extract the volume. There was volume information before I guess. (See https://stackoverflow.com/questions/30323721/mpd-receive-actual-mpd-status-with-linux-bash-script)

Assuming this is related to an `mpd` version update - how do we get the volume information from mpd?

topas-rec, 2020-05-31:
I think I got further and solved this on my machine.

- fact: mpd has no volume status when this issue is present
- I found somewhere that this is caused by no or wrong mixer setup in mpd
- I looked at my alsamixer which usually shows only the `PCM` mixer control
- After updating all packages through apt it shows a mixer control named `Headphone ` and not `PCM` anymore

[image]

I looked into mpd.conf and there was `PCM` set in audio output mixer section. I changed that to `Headphone` and restartet mpd through `sudo systemctl restart mpd`.

Now the volume change works without issues.

Can someone confirm this solution who has this issue? (All packages have to be up to date)

s-martin, 2020-06-01:
Thanks for investigating this!

I can also confirm that on my dev Pi I now have "Headphone".

I agree that probably an update of MPD changed this. 

We should to check, if we now need to support both cases.

s-martin, 2020-06-01:
Looks like this is a change of the Pi audio behavior itself:
https://retropie.org.uk/forum/topic/26434/usb-audio-broke-after-latest-updates/2
https://www.raspberrypi.org/blog/latest-raspberry-pi-os-update-may-2020/

topas-rec, 2020-06-01:
> Thanks for investigating this!
> 
> I can also confirm that on my dev Pi I now have "Headphone".
> This is mpd version 0.21.5
> 
> I agree that probably an update of MPD changed this.
> 
> We should to check, if we now need to support both cases.

I don't know where to implement handling these cases and I'm not the best bash programmer. Weather we need both cases (PCM and headphone) or just handling the up to date Raspbian version (Headphone), can you do this?

s-martin, 2020-06-01:
According to https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working **Section b)** you need to change the output at 2 places.

Could you check, if this works for you when you change the two places to `Headphone` and then give me feedback?

Then I would update the install script and the wiki docs.

s-martin, 2020-06-01:
@topas-rec additional question:
Which Pi do you have?

I experienced this new behavior on my Pi4, but I currently don't have an older model to test.

topas-rec, 2020-06-01:
Mine is a `Pi Model 3B V1.2`

s-martin, 2020-06-01:
Ok, so we can assume that a fix is needed for more models than Pi4

MiczFlor, 2020-06-02:
I am looking into this now, upgrading my Raspbian Lite to the latest version and then will install a new OS using the latest `lite` version.
~~~
Raspberry Pi OS (32-bit) Lite
Minimal image based on Debian Buster
Version:May 2020
Release date:2020-05-27
Kernel version:4.19
Size:432 MB
~~~

MiczFlor, 2020-06-02:
I can verify that PCM has disappeared and is now called Headphone. I updated the installer script for the `develop` branch. Weirdly, the command `amixer scontrols` does not show *Headphones*. But `alsamixer` offers it as a sound card.
Installing a new OS now and see what it does with the new install script...

MiczFlor, 2020-06-02:
I can also confirm that `echo -e status\\nclose | nc -w 1 localhost 6600` does return the volume in percent anymore. Ok, need to look into this much much more. 
I love my job :) there is always something falling off the other end of the table when you adjust the table cloth at this end... ;)
Won't be able to fix this today.

MiczFlor, 2020-06-02:
I've been investigating a bit further. And here comes the fun part: 
(first my question @htpzc : did you try with HDMI plugged in? Did you have a speaker / headset connected to the minijack?)

Once I detached the HDMI cable AND plugged in a set of speakers into the minijack, everything worked as expected:

* setting startup volume in settings to 75% -> reboot and worked
* changing volume in the web app up and down worked (display and actual audio out from 75 to 0 and back to 100)
* read the volume from command line:
~~~
$ echo -e status\\nclose | nc -w 1 localhost 6600 | grep -o -P '(?<=volume: ).*'
43
~~~
* reading the volume level with the Phoniebox script
~~~
$ ./RPi-Jukebox-RFID/scripts/playout_controls.sh -c=getvolume
43
~~~
This was done with `Headphone` as the iFace name:
~~~
$ sudo cat /etc/mpd.conf | grep Head
	mixer_control	"Headphone"
~~~
and:
~~~
$ sudo cat RPi-Jukebox-RFID/settings/global.conf | grep Head
AUDIOIFACENAME="Headphone"
~~~
Not sure how to document this best during the installation and/or in the manual.

htpzc, 2020-06-02:
sorry for not answering. I am in holidays for the next 20 days. 
No i did not Connect hdmi. 
My minijack is the usb adapter which is recommended on the homepage. There i have connected the speakers which are recommended at the Homepage ;-)

Sorry, but i cannot test anything until i am back home. Then i will check again.

MiczFlor, 2020-06-03:
enjoy your holidays and switch off :) I can't reproduce the behaviour unfortunately. but someone else might jump in?

dnmtch, 2020-06-03:
> enjoy your holidays and switch off :) I can't reproduce the behaviour unfortunately. but someone else might jump in?

so to help, i take:
- a raspi 3 
- add Button according to Flexible Pinout (vol +/- GPIO 5/6)
- install the script on a fresh Raspian Lite withHDMI cable (without display) AND a set of speakers/headphone plugged in
- record the whole SSH Session
- test if the buttons are working

develp oder master?
cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh

s-martin, 2020-06-03:
Please use `develop`, install with this command:

`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; GIT_BRANCH=develop bash ./buster-install-default.sh`

s-martin, 2020-06-03:
I have a Pi4, no connection to HDMI or audio jack and no additional sound card (it's my dev device), and I only see audio device `Headphone` at `amixer scontrols`, `alsamixer` and `aplay -l`.
I had to change to `Headphone`, otherwise MPD would throw errors.

> This was done with `Headphone` as the iFace name:
> 
> ```
> $ sudo cat /etc/mpd.conf | grep Head
> 	mixer_control	"Headphone"
> ```
> 
> and:
> 
> ```
> $ sudo cat RPi-Jukebox-RFID/settings/global.conf | grep Head
> AUDIOIFACENAME="Headphone"
> ```

@MiczFlor, side question: in addition we have also `settings/Audio_iFace_Name`, how does that correspond to `global.conf`?

dnmtch, 2020-06-03:
**Good News: Buttons are working!**

- raspi 3 with fresh Raspian Lite (2020-05-27-raspios-buster-lite-armhf) with HDMI cable (without display) AND a set of speakers/headphone plugged in
- Button according to Flexible Pinout (vol +/- GPIO 5/6) (default is 16/19)
- record the whole SSH Session [putty.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4726325/putty.log)

- buttons are working!

[image]

dnmtch, 2020-06-03:
If someone could help me with the new "Rotary Knob" Setup (https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/gpio_control) i would realy be glad to document that.

s-martin, 2020-06-04:
> If someone could help me with the new "Rotary Knob" Setup (https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/gpio_control) i would realy be glad to document that.

GPIO Control is not straightforward yet (#991 is supposed to change that). I haven’t used it before (@veloxidSchweiz implemented it), but I try to explain as much as possible.

GPIO Control replaces also gpio-buttons.py and all GPIO devices (except readers) are configured in `~/.config/phoniebox/gpio_settings.ini`. 

Steps necessary:

1. Deactivate old GPIO service: `sudo systemctl deactivate phoniebox-gpio-buttons` and  `sudo systemctl stop phoniebox-gpio-buttons

2. Install GPIO control: `sudo components/gpio_control/install.sh`

3. manually edit `~/.config/phoniebox/gpio_settings.ini` for a button and a rotary knob

4. post results here :)

doliverio, 2020-06-04:
I tried to configure the Rotary Knob as well. With my setup the VolumeControl Device cannot be initialized correctly for some reason.

OS: Raspbian Buster Lite
Hardware: Raspberry Pi Zero + HifiBerry MiniAmp + Rotary Control

**Steps**

```
# Headless install (develop  branch)
cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; GIT_BRANCH=develop bash ./buster-install-default.sh -a

sudo reboot

# old service
sudo systemctl deactivate phoniebox-gpio-buttons
sudo systemctl stop phoniebox-gpio-buttons

# to fix error: 'components/gpio_control/install.sh: line 8: pip: command not found'
sudo update-alternatives --install /usr/bin/pip pip /usr/bin/pip3 1

# cd into the directory. otherwise the install-script is not able to find the example-configs
cd ~/RPi-Jukebox-RFID/components/gpio_control

# Uncomment line 4 in order to be able to execute the installation as sudo (#   exit 1)
nano ./install.sh

# Start installation
sudo ./install.sh

# enable and start service
sudo systemctl enable phoniebox_gpio_control.service
sudo systemctl start phoniebox_gpio_control.service

# modify gpio_settings.ini
sudo nano ~/.config/phoniebox/gpio_settings.ini

# check functionality
cd ~/RPi-Jukebox-RFID/components/gpio_control
python gpio_control.py
```

Output:

    WARNING:root:Could not add Device VolumeControl with [('enabled', 'True'), ('type', 'RotaryEncoderClickable'), ('pinup', '16'), ('pindown', '19'), ('pull_up', 'True'), ('hold_time', '0.3'), ('hold_repeat', 'True'), ('timebase', '0.1 # only for rotary encoder'), ('functioncalldown', 'functionCallVolD'), ('functioncallup', 'functionCallVolU'), ('functioncalltwobuttons', 'functionCallVol0 #only for TwoButtonControl'), ('functioncallbutton', 'functionCallPlayerPause # only for RotaryEncoderClickable')]

s-martin, 2020-06-04:
The service `phoniebox_gpio_control` starts `gpio_control.py`, so I'm not sure, if you cn use both at the same time.
@veloxidSchweiz, do you have an idea?

s-martin, 2020-06-04:
What's the result of `journalctl -u phoniebox_gpio_control.service -f`?

s-martin, 2020-06-04:
I think **type** should just be `RotaryEncoder`, at least I couldn't find any instance of `RotaryEncoderClickable` in the code.

doliverio, 2020-06-04:
> I think type should just be RotaryEncoder, at least I couldn't find any instance of RotaryEncoderClickable in the code.

Perfect, this did the trick. Two updates of the `gpio_settings.ini` are necessary:

1) `Type: RotaryEncoderClickable` => `Type: RotaryEncoder`
2) `timeBase: 0.1 # only for rotary encoder` => `timeBase: 0.1`

Thanks @s-martin !

Piehti, 2020-06-05:
Howdy! 
Any news? Got like the same issue. It's a RasPi 3B+. But I think it also comes up on the RasPi4b. At least I think I had that issues, when I decided to move the PhoeniBox from a KODIbox zu a VPNBox. I thought the issues was related to conflicting MPD usage of KODI / Phonie but might have been update-related there already.

I actually dont wanna resetup that 3B+ and 4B systems, since it took me ages to get the VPN / AP / PiHole etc running correctly. Might try on a RasPi 3B I got laying around later that day.

Stupid question: May this be related to "rpi-update"? Can anyone confirm, that users without rpi-update are concerned? Not sure, but think I did it on all devices - since the RPi 3B+ runs at 1.5ghz and the RP4b at 2.1GHZ ("slightly") OC'ed it.

Help!!!! For now I can only adjust volume on phsical volume rulers and my wife keeps shouting at my and I have to jump behind the TV and.... !!! :D ^^;

Best regards
Pete

s-martin, 2020-06-05:
I think you need to check with
`alsamixer`
`amixer scontrols`
or
`aplay -l`

how your audio devices are now called and which number they may have.

Changed names (e.g. `Headphone`) need to be updated in `/etc/mpd.conf`, `settings/global.conf` and `settings/Audio_iFace_Name`.

If you made changes in `/etc/asound.conf` (see https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working) you maybe need to update the numbers for the device as seen in `aplay -l`.

doliverio, 2020-06-05:
> Stupid question: May this be related to "rpi-update"? Can anyone confirm, that users without rpi-update are concerned?
 
@Piehti Yes, I was affected without making `rpi-update`. 
Line `timeBase: 0.1 # only for rotary encoder` in `gpio_settings.ini` generates this error: 

`ValueError: could not convert string to float: '0.1 # only for rotary encoder`

Fixing this line and changing `Type: RotaryEncoderClickable` to `Type: RotaryEncoder` solved the problem for me.

s-martin, 2020-06-05:
> Stupid question: May this be related to "rpi-update"? Can anyone confirm, that users without rpi-update are concerned? Not sure, but think I did it on all devices - since the RPi 3B+ runs at 1.5ghz and the RP4b at 2.1GHZ ("slightly") OC'ed it.

No, it’s related to a Raspberry Pi update (https://www.raspberrypi.org/blog/latest-raspberry-pi-os-update-may-2020/), which affected all Pi’s

Piehti, 2020-06-05:
> Changed names (e.g. `Headphone`) need to be updated in `/etc/mpd.conf`, `settings/global.conf` and `settings/Audio_iFace_Name`.
>
> If you made changes in `/etc/asound.conf` (see [...]

I got my sound back working, by fixing the suggested stuff to "Headphones".
Unfortunatly I still dont got my volumecontrol back. Ich changed to rotaryclick as decriped above. No effect. What do you mean by "fixing" the line? I just deleted the comments in it - which doesnt help. :/

veloxidSchweiz, 2020-06-05:
could you run the check installation srcipt:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/gpio_control/check_installation.sh

Kreiszahler, 2020-06-05:
```
Problem during installation occured 
   Service not running, please check functionallity by running gpio_control.py 
   in the directory ~/RPi-Jukebox-RFID/components/gpio_control: 
      $ cd ~/RPi-Jukebox-RFID/components/gpio_control
      $ python gpio_control.py
   or check output of journaclctl by:
      $ journalctl -u phoniebox_gpio_control.service -f

PROBLEM IN INSTALLATION 
```

Im not using any GPIO buttons  - aye?

Piehti, 2020-06-05:
Diggin' deeper...

```
File "gpio_control.py", line 6, in <module>
    from GPIODevices import *
File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/__init__.py", line 7, in <module>
    from .led import *
File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/led.py", line 4, in <module>
    import mpd
```

veloxidSchweiz, 2020-06-05:
did you install the requirements, i.e. python-mpd2
otherwise try 
'''pip install -r requirements.txt'''
or just 
'''pip install python-mpd2'''

Piehti, 2020-06-05:
So I ran **sudo apt-get install python-pip** for I had no pip-support. *scratchscratch*
After **pip install -r requirements.txt** + reboot, I got **python gpio_control.py** back running. Still the **check_installation.sh** states the same error.

When hitting the vol up / down, the following error occours:

[image]

I think that is new and I can remember to have had that once, I played with the playoutcontrols when implementing video stuff and messed up something. Before I take us on a ride for ghosts, (because I did something that I didnt want to do nor can remember) I gonna backup the state for later purpose and just reinstall the dev branch to be sure.

Be right back.

Piehti, 2020-06-05:
After I found my samba-shares gone, I re-setup and resynced and chowned / chmodded the box.
**The issue seems to be gone after the installation of the dev branch**.

Thanks for all the support. Sorry for th inconvenience.

dnmtch, 2020-06-06:
> 
> 
> What's the result of `journalctl -u phoniebox_gpio_control.service -f`?

$ journalctl -u phoniebox_gpio_control.service -f
-- Logs begin at Sat 2020-06-06 22:23:40 BST. --
Jun 06 22:33:07 raspberrypi PhonieboxGPIOControl[8263]: Traceback (most recent call last):
Jun 06 22:33:07 raspberrypi PhonieboxGPIOControl[8263]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 6, in <module>
Jun 06 22:33:07 raspberrypi PhonieboxGPIOControl[8263]:     from GPIODevices import *
Jun 06 22:33:07 raspberrypi PhonieboxGPIOControl[8263]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/__init__.py", line 7, in <module>
Jun 06 22:33:07 raspberrypi PhonieboxGPIOControl[8263]:     from .led import *
Jun 06 22:33:07 raspberrypi PhonieboxGPIOControl[8263]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/led.py", line 4, in <module>
Jun 06 22:33:07 raspberrypi PhonieboxGPIOControl[8263]:     import mpd
Jun 06 22:33:07 raspberrypi PhonieboxGPIOControl[8263]: ModuleNotFoundError: No module named 'mpd'
Jun 06 22:33:07 raspberrypi systemd[1]: phoniebox_gpio_control.service: Main process exited, code=exited, status=1/FAILURE
Jun 06 22:33:07 raspberrypi systemd[1]: phoniebox_gpio_control.service: Failed with result 'exit-code'.
Jun 06 22:33:10 raspberrypi systemd[1]: phoniebox_gpio_control.service: Service RestartSec=3s expired, scheduling restart.
Jun 06 22:33:10 raspberrypi systemd[1]: phoniebox_gpio_control.service: Scheduled restart job, restart counter is at 161.
Jun 06 22:33:10 raspberrypi systemd[1]: Stopped Phoniebox GPIO Control Service.
Jun 06 22:33:10 raspberrypi systemd[1]: Started Phoniebox GPIO Control Service.
Jun 06 22:33:10 raspberrypi PhonieboxGPIOControl[8277]: Traceback (most recent call last):
Jun 06 22:33:10 raspberrypi PhonieboxGPIOControl[8277]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 6, in <module>
Jun 06 22:33:10 raspberrypi PhonieboxGPIOControl[8277]:     from GPIODevices import *
Jun 06 22:33:10 raspberrypi PhonieboxGPIOControl[8277]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/__init__.py", line 7, in <module>
Jun 06 22:33:10 raspberrypi PhonieboxGPIOControl[8277]:     from .led import *
Jun 06 22:33:10 raspberrypi PhonieboxGPIOControl[8277]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/led.py", line 4, in <module>
Jun 06 22:33:10 raspberrypi PhonieboxGPIOControl[8277]:     import mpd
Jun 06 22:33:10 raspberrypi PhonieboxGPIOControl[8277]: ModuleNotFoundError: No module named 'mpd'
Jun 06 22:33:10 raspberrypi systemd[1]: phoniebox_gpio_control.service: Main process exited, code=exited, status=1/FAILURE
Jun 06 22:33:10 raspberrypi systemd[1]: phoniebox_gpio_control.service: Failed with result 'exit-code'.
Jun 06 22:33:14 raspberrypi systemd[1]: phoniebox_gpio_control.service: Service RestartSec=3s expired, scheduling restart.
Jun 06 22:33:14 raspberrypi systemd[1]: phoniebox_gpio_control.service: Scheduled restart job, restart counter is at 162.
Jun 06 22:33:14 raspberrypi systemd[1]: Stopped Phoniebox GPIO Control Service.
Jun 06 22:33:14 raspberrypi systemd[1]: Started Phoniebox GPIO Control Service.
Jun 06 22:33:14 raspberrypi PhonieboxGPIOControl[8288]: Traceback (most recent call last):
Jun 06 22:33:14 raspberrypi PhonieboxGPIOControl[8288]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 6, in <module>
Jun 06 22:33:14 raspberrypi PhonieboxGPIOControl[8288]:     from GPIODevices import *
Jun 06 22:33:14 raspberrypi PhonieboxGPIOControl[8288]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/__init__.py", line 7, in <module>
Jun 06 22:33:14 raspberrypi PhonieboxGPIOControl[8288]:     from .led import *
Jun 06 22:33:14 raspberrypi PhonieboxGPIOControl[8288]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/led.py", line 4, in <module>
Jun 06 22:33:14 raspberrypi PhonieboxGPIOControl[8288]:     import mpd
Jun 06 22:33:14 raspberrypi PhonieboxGPIOControl[8288]: ModuleNotFoundError: No module named 'mpd'
Jun 06 22:33:14 raspberrypi systemd[1]: phoniebox_gpio_control.service: Main process exited, code=exited, status=1/FAILURE
Jun 06 22:33:14 raspberrypi systemd[1]: phoniebox_gpio_control.service: Failed with result 'exit-code'.
Jun 06 22:33:17 raspberrypi systemd[1]: phoniebox_gpio_control.service: Service RestartSec=3s expired, scheduling restart.
Jun 06 22:33:17 raspberrypi systemd[1]: phoniebox_gpio_control.service: Scheduled restart job, restart counter is at 163.
Jun 06 22:33:17 raspberrypi systemd[1]: Stopped Phoniebox GPIO Control Service.
Jun 06 22:33:17 raspberrypi systemd[1]: Started Phoniebox GPIO Control Service.
Jun 06 22:33:17 raspberrypi PhonieboxGPIOControl[8300]: Traceback (most recent call last):
Jun 06 22:33:17 raspberrypi PhonieboxGPIOControl[8300]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 6, in <module>
Jun 06 22:33:17 raspberrypi PhonieboxGPIOControl[8300]:     from GPIODevices import *
Jun 06 22:33:17 raspberrypi PhonieboxGPIOControl[8300]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/__init__.py", line 7, in <module>
Jun 06 22:33:17 raspberrypi PhonieboxGPIOControl[8300]:     from .led import *
Jun 06 22:33:17 raspberrypi PhonieboxGPIOControl[8300]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/led.py", line 4, in <module>
[... cut ...]
Jun 06 22:33:27 raspberrypi PhonieboxGPIOControl[8335]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/led.py", line 4, in <module>
Jun 06 22:33:27 raspberrypi PhonieboxGPIOControl[8335]:     import mpd
Jun 06 22:33:27 raspberrypi PhonieboxGPIOControl[8335]: ModuleNotFoundError: No module named 'mpd'
Jun 06 22:33:27 raspberrypi systemd[1]: phoniebox_gpio_control.service: Main process exited, code=exited, status=1/FAILURE
Jun 06 22:33:27 raspberrypi systemd[1]: phoniebox_gpio_control.service: Failed with result 'exit-code'.

s-martin, 2020-06-07:
@dnmtch:

Please try
`sudo python3 -m pip install python-mpd2`

MiczFlor, 2020-06-12:
Because I can not reproduce this behaviour, but many reported the volume up / down does not work anymore, I added the option to switch volume management between `mpd` and `amixer`. Not sure if this does help, but it might help to find the issue - also with external soundcards.

What you need to do: 

* get the latest `develop` branch
* go to the web app and under settings you can select which manager to use:
[image]
* if you now switch on the debugging under settings for playout_controls, you will see `VOLUMEMANAGER:amixer` in the log files where volume is being handled

I hope this will help to narrow down the volume up/down issue

bendreality, 2020-06-18:
Hi all,

I got it to work. It's the first time for me using Linux, so if one (or more) of the steps doesn't make sense, please correct me. :) And don't use this solution just yet. Wait for someone more sophisticated to review what I wrote. 

I had the same issue. I couldn't control the volume via the web interface. But I was sure that the sound card was setup correctly. music could be played via web interface, but the volume control graphic was blank.  Also the softvol was setup correctly. I could play music via the web interface, but had to control volume with alsamixer or amixer. But first, every time I tried to open alsamixer or amixer I got a similar error. 

`
pi@phoniebox:~ $ amixer
amixer: Mixer attach default error: No such file or directory
`

if I run with sudo it worked:

```
pi@phoniebox:~ $ sudo amixer
Simple mixer control 'Master',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 153 [60%]
  Front Right: 153 [60%]

```
So I guessed that it's maybe right management stuff. So I did :

```
ls -l  /usr/bin/amixer
-rwxr-xr-x 1 root root 42517 Jun 18 13:29 /usr/bin/amixer
```
as root I did a:

`chown pi:pi /usr/bin/amixer`

and

`sudo chmod a+X /usr/bin/amixer`

and added the pi user to the audio group (got this from [here](https://wiki.archlinux.org/index.php/Advanced_Linux_Sound_Architecture#User_privileges)):

`usermod -a -G audio pi`

And I Edited the /etc/mpd.conf to:

```
audio_output {
        type            "alsa"
        name            "My ALSA Device"  
        mixer_type      "software"      # <--------- changed from "hardware"
    
}
```

Now it works, but I surely did some stupid stuff along the way. 😆 So maybe someone with a little deeper insight can distillate this "solution" into something a little more profound. Maybe the edit of the mpd.conf is enough, but I did everything in the order I wrote it down. 

So, I hope it helps. :)

themorlan, 2020-07-04:
Hey,
I just installed the development version (+Spotify) on a RPi3b+ on RaspiOS Lite.
After running the install.sh I had to change permissions on `~/.config/phoniebox/`with `sudo chown pi:pi -R phoniebox/`because it was owned by root. Also debug.log was owned by root and threw errors.
After that the rotary knob worked. 
But I'm a little confused cause when turning the knob the change of volume is delayed by approximately 2 seconds. I did not have this behavior on the base version without Spotify. Do other also observe this?

Cali227, 2020-07-09:
Hey, I hope you can help me. I am totaly new here and i have no idea of prgramming. I just wanted to creat a music box for my daughter =)

So i have this problem: When i install the Phoniebox without Spotify, everything works very well. I changed "PCM" to "Headphone" in "Sudo nano etc/mpd.conf" ( I have no soundcard btw).

But i want to use the Version with Spotify and when i chose [y] in sthe One Line installer, the Phnoebox does not work. 
I can upload a song from my computer but there is no sound and the track is not playing.

I have to say, that i dont understand most of your lines above. It would be great if you can write exactly what i have to type/change in the console.

I hope you can help me =)

Tahnk you!!!

topas-rec, 2020-07-09:
@Cali227 this issue is for a volume problem. Next time you should open a new issue for your problem or search if someone already created an issue for your problem (this is a friendly advice)

Spotify version needs special handling for local files. This is explained at the readme file and links to [MUST READ for users of Phoniebox +Spotify Edition](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/docs/SPOTIFY-INTEGRATION.md)

To be honest I cannot find an issue where this was already asked but I had problems with local files too and was able to solve them when I found out that I have to scan the library once.

Perhaps that helps.
**If you have further problems, please try to find an issue that already covers the problem (and perhaps a solution or workaround) or create a new issue. When you have volume problems you might report them here is they match the issue description above in the first comment**

htpzc, 2020-07-12:
Hey Guys, back from holidays :-)

Now got the time and checked what you have written and tested that.
For information again. I am using an Pi Zero W.

Checked:
# cat /proc/asound/modules
 0 snd_bcm2835
 1 snd_usb_audio

# amixer scontrols
Simple mixer control 'Speaker',0
Simple mixer control 'Mic',0
Simple mixer control 'Auto Gain Control',0

No Headphones to see. So what to do.
I checked your Dokumentation (Troubleshooting) again if there is anything i have overseen or which has been newly added.

So what i already did is:
# vi /usr/share/alsa/alsa.conf
defaults.ctl.card 1
defaults.pcm.card 1

# vi /etc/asound.conf
pcm.!default {
    type hw
    card 1
}
ctl.!default {
    type hw
    card 1
}

### But what i not did was changing PCM to:
# vi /etc/mpd.conf
mixer_control	"Speaker"		# optional

After reboot everything worked fine again.

As already written, before, everything worked fine and i did not  have to change the mixer_control parameter in mpd.conf.
Now, maybe after this Pi update you have written, it was necessary to change that parameter.

So for me, it works fine. 
I am having some trouble with noise in the speakers, but that is an other issue and i will start a new issue for that  when i have tested everything out.

So thanks for your trouble investigation. It did not solve my problem but it pushed me to the needed answer ;-)

Marcus

closed 2020-07-12 by htpzc

phreakadelle, 2020-07-21:
I'd my phoniebox up and running since a year. Now i've did an update to use the latest version and also added some GPIO buttons. I've faced the same issue, when i did the installation from scratch. I had a keyboard and screen (hdmi) attached. The volume control didn't work.

Now i just did a new installation from the commandline, without the screen attached and it worked out of the box from the master branch. 

Thanks to @s-martin and @MiczFlor for this awesome project.

Cali227, 2020-07-21:
So you use Spotify?

[2 more replies]