MiczFlor/RPi-Jukebox-RFID #975: 🐛 | No titles on WebUI when adding spotify playlist

Groovylein, 2020-05-23:
## Bug

### What I did

I added a Spotify playlist via the WebUI and connected it to an RFID card

### What happened

The playlist does not have any title

Screenshot will be placed here later

BerniPi, 2020-05-23:
See here: #799

Groovylein, 2020-05-23:
I installed from latest develop branch.

MiczFlor, 2020-05-23:
There were two commits that might cause this behaviour. I have no spotify running, who could find the bug?
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/884
by @lovethisshit 
and
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/953
by @danielhammer

danielhammer, 2020-05-23:
Just checked out latest develop branch, added several public and private playlist and do not face the issue yet. I will try reinstalling from scratch. Maybe some issue with directory permissions comes up on fresh installs? Let's see...

danielhammer, 2020-05-23:
I deleted the previously installed folder of RPi-Jukebox-RFID and reinstalled by using the one-line-script as described here [https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#b-developer-install-script-for-buster](url) and the playlists titles are coming up as expected.

danielhammer, 2020-05-24:
**Lastly I tested:**

- completely fresh Raspian buster image (raspbian buster lite)
- fresh RPi-Jukebox-RFID installation via [one-line-install-script for developers](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#b-developer-install-script-for-buster)
- added several private and public Spotify playlists by URI (format `spotify:playlist:37i9dQZF1DX2TRYkJECvfC`) via "Add new Card" web form.

**Result:**

- Playlist titles and covers get loaded and shown in web UI correctly (Firefox, Safari, Chrome) ✔️
- tested the same for single songs, just to be shore, works fine ✔️

@Groovylein could it be possible, that you are working on a slightly older version of the develop branch? The behaviour you describe in your bug report, is similar to what I observed before integrating my patch that was finally merged in #953 .

Groovylein, 2020-05-24:
Ok thank you for testing @danielhammer .
I will have a look later on which commit I am and compare it.

Groovylein, 2020-05-24:
My mistake I was on an older commit.

closed 2020-05-24 by Groovylein