MiczFlor/RPi-Jukebox-RFID #978: 🐛 | Import RFID links from .csv file failed

RalfAlbers, 2020-05-25:
## Bug
Import of an exported csv fails with "ERROR: There was an error uploading the file, please try again!"

### What I did
Import of an exported csv via WebUI. The CSV contains only RFIDs & Playlists, no systemcommands

### What happened
Import fail with "ERROR: There was an error uploading the file, please try again!"

### I expected this to happen
Import restores the assignment of RFID to playlist

### Further information that might help
/var/log/lighttpd/error.log shows:

> (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: MAILWLANIPYN in /home/pi/RPi-Jukebox-RFID/htdocs/inc.header.php on line 158
> (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: MAILWLANIPADDR in /home/pi/RPi-Jukebox-RFID/htdocs/inc.header.php on line 159
> (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined variable: post in /home/pi/RPi-Jukebox-RFID/htdocs/inc.processCheckCardEditRegister.php on line 111
> (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined variable: post in /home/pi/RPi-Jukebox-RFID/htdocs/inc.processCheckCardEditRegister.php on line 123
> (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  move_uploaded_file(/rfidImport.csv): failed to open stream: Permission denied in /home/pi/RPi-Jukebox-RFID/htdocs/cardRegisterNew.php on line 58
> (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  move_uploaded_file(): Unable to move '/tmp/phpPcW29Q' to '/rfidImport.csv' in /home/pi/RPi-Jukebox-RFID/htdocs/cardRegisterNew.php on line 58
> (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined variable: post in /home/pi/RPi-Jukebox-RFID/htdocs/cardRegisterNew.php on line 248
> (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index:  in /home/pi/RPi-Jukebox-RFID/htdocs/cardRegisterNew.php on line 294

## Software
### Base image and version
2020-02-13-raspbian-buster-lite

### Branch / Release
2.0 - ed888da - master

### Installscript
scripts/installscripts/buster-install-default-with-autohotspot.sh

s-martin, 2020-05-25:
@MiczFlor here https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/596f1152b489b4f3a8b8558820aaffbd7380facd/htdocs/cardRegisterNew.php#L58 it looks like there‘s a typo, I suppose we don’t want to copy that to root `/`.

s-martin, 2020-05-26:
@RalfAlbers:
I opened PR #983 to fix this.
In the meantime you could do the change for yourself.
In the file `/home/pi/RPi-Jukebox-RFID-FORK/htdocs/cardRegisterNew.php`:
In line 58 and 65 change `"/rfidImport.csv"` to `"./rfidImport.csv"`.

RalfAlbers, 2020-05-27:
@s-martin: THX, this works 👍

closed 2020-05-27 by MiczFlor