MiczFlor/RPi-Jukebox-RFID #980: 🐛 | Local folder don't play after update

BerniPi, 2020-05-26:
## Bug

### What I did
Updated my development-phoniebox to latest develop. And kept all settings or left it the same.

### What happened
After updating and rebooting an existing folder don't play anymore. I tested the folder before updating, because i had this problem before.
I uploaded the a new folder through the webinterface and this folder works (same files, different foldername)

### I expected this to happen
The playback works for existing folders.

### Further information that might help
I looked at the permissions. The permissions and owner of the new created folder is different (owner www-data and 777). But although i change the permissions of the old folder, i can't get it to work.

## Software
Phoniebox Classic

### Base image and version
Raspian Buster Lite

### Branch / Release
latest develop

### Installscript
scripts/installscripts/buster-install-default.sh`

## Hardware

### RaspberryPi version
3B

### RFID Reader

None.

### Soundcard
PCM

### Other notable hardware
None

BerniPi, 2020-05-26:
I now changed the Resume option in the Webinterface and now the folder works. I reinstall and investigate further.

s-martin, 2020-05-26:
Can you post the output of `tail -n 500 /var/log/syslog`

BerniPi, 2020-05-26:
I looked into syslog and some other logfiles and didn't find anything interesting to me. Maybe i overlooked something.
I start from the beginning again.

BerniPi, 2020-05-26:
These are the syslog entries: `May 26 13:50:09 phoniebox kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
May 26 13:50:09 phoniebox kernel: [    0.000000] Memory: 939084K/970752K available (8192K kernel code, 653K rwdata, 2220K rodata, 1024K init, 822K bss, 23476K reserved, 8192K cma-reserved)
May 26 13:50:09 phoniebox systemd[1]: Condition check resulted in RPC security service for NFS client and server being skipped.
May 26 13:50:09 phoniebox kernel: [    0.000000] Virtual kernel memory layout:
May 26 13:50:09 phoniebox kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
May 26 13:50:09 phoniebox kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
May 26 13:50:09 phoniebox kernel: [    0.000000]     vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
May 26 13:50:09 phoniebox systemd[1]: Reached target NFS client services.
May 26 13:50:09 phoniebox kernel: [    0.000000]     lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
May 26 13:50:09 phoniebox kernel: [    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
May 26 13:50:09 phoniebox kernel: [    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
May 26 13:50:09 phoniebox systemd[1]: Reached target Remote File Systems (Pre).
May 26 13:50:09 phoniebox kernel: [    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
May 26 13:50:09 phoniebox kernel: [    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 654 kB)
May 26 13:50:09 phoniebox kernel: [    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 823 kB)
May 26 13:50:09 phoniebox kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
May 26 13:50:09 phoniebox systemd[1]: Reached target Remote File Systems.
May 26 13:50:09 phoniebox kernel: [    0.000000] ftrace: allocating 26860 entries in 79 pages
May 26 13:50:09 phoniebox kernel: [    0.000000] rcu: Hierarchical RCU implementation.
May 26 13:50:09 phoniebox kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
May 26 13:50:09 phoniebox systemd[1]: Started Set console font and keymap.
May 26 13:50:09 phoniebox systemd[1]: Started Create Volatile Files and Directories.
May 26 13:50:09 phoniebox systemd[1]: Starting Update UTMP about System Boot/Shutdown...
May 26 13:50:09 phoniebox kernel: [    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
May 26 13:50:09 phoniebox systemd[1]: Starting Network Time Synchronization...
May 26 13:50:09 phoniebox kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
May 26 13:50:09 phoniebox kernel: [    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
May 26 13:50:09 phoniebox kernel: [    0.000019] Switching to timer-based delay loop, resolution 52ns
May 26 13:50:09 phoniebox kernel: [    0.000276] Console: colour dummy device 80x30
May 26 13:50:09 phoniebox systemd[1]: Started Update UTMP about System Boot/Shutdown.
May 26 13:50:09 phoniebox kernel: [    0.000296] console [tty1] enabled
May 26 13:50:09 phoniebox kernel: [    0.000342] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
May 26 13:50:09 phoniebox kernel: [    0.000357] pid_max: default: 32768 minimum: 301
May 26 13:50:09 phoniebox systemd[1]: Started Raise network interfaces.
May 26 13:50:09 phoniebox kernel: [    0.000700] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
May 26 13:50:09 phoniebox kernel: [    0.000715] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
May 26 13:50:09 phoniebox kernel: [    0.001663] CPU: Testing write buffer coherency: ok
May 26 13:50:09 phoniebox kernel: [    0.002123] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
May 26 13:50:09 phoniebox kernel: [    0.002781] Setting up static identity map for 0x100000 - 0x10003c
May 26 13:50:09 phoniebox kernel: [    0.002935] rcu: Hierarchical SRCU implementation.
May 26 13:50:09 phoniebox systemd[1]: Started Network Time Synchronization.
May 26 13:50:09 phoniebox kernel: [    0.003734] smp: Bringing up secondary CPUs ...
May 26 13:50:09 phoniebox kernel: [    0.004627] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
May 26 13:50:09 phoniebox kernel: [    0.005624] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
May 26 13:50:09 phoniebox systemd[1]: Reached target System Initialization.
May 26 13:50:09 phoniebox kernel: [    0.006536] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
May 26 13:50:09 phoniebox kernel: [    0.006658] smp: Brought up 1 node, 4 CPUs
May 26 13:50:09 phoniebox kernel: [    0.006670] SMP: Total of 4 processors activated (153.60 BogoMIPS).
May 26 13:50:09 phoniebox systemd[1]: Listening on triggerhappy.socket.
May 26 13:50:09 phoniebox kernel: [    0.006675] CPU: All CPU(s) started in HYP mode.
May 26 13:50:09 phoniebox kernel: [    0.006680] CPU: Virtualization extensions available.
May 26 13:50:09 phoniebox kernel: [    0.007654] devtmpfs: initialized
May 26 13:50:09 phoniebox kernel: [    0.020703] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
May 26 13:50:09 phoniebox systemd[1]: Listening on mpd.socket.
May 26 13:50:09 phoniebox kernel: [    0.020941] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
May 26 13:50:09 phoniebox kernel: [    0.020961] futex hash table entries: 1024 (order: 4, 65536 bytes)
[... cut ...]
May 26 13:50:56 phoniebox systemd[1]: systemd-update-utmp-runlevel.service: Succeeded.
May 26 13:50:56 phoniebox systemd[1]: Started Update UTMP about System Runlevel Changes.
May 26 13:50:57 phoniebox systemd[1]: Startup finished in 2.004s (kernel) + 44.035s (userspace) = 46.040s.
May 26 13:51:26 phoniebox systemd[1]: phoniebox-idle-watchdog.service: Succeeded.
May 26 13:51:58 phoniebox lighttpd[746]: error adding /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Recordings/2020-04-07_21-27-55.wav: No such directory`

BerniPi, 2020-05-26:
I can get the folders to work, when i change a playback setting in the webinterface.
About the syslog: I also tried an other folder, but the failing playback doesn't get logged.

BerniPi, 2020-05-26:
I got more errors of this kind: 
`May 26 19:58:11 phoniebox lighttpd[766]: cat: /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_Folder_Played: No such file or directory
May 26 19:58:11 phoniebox lighttpd[766]: cat: /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_Playlist_Played: No such file or directory
May 26 19:58:12 phoniebox lighttpd[766]: cat: /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_Folder_Played: No such file or directory
May 26 19:58:26 phoniebox systemd[1]: phoniebox-idle-watchdog.service: Succeeded.
May 26 19:59:02 phoniebox lighttpd[766]: error adding /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Katze 2/1 - ABC, die Katze lief im Schnee.mp3: No such directory
May 26 19:59:05 phoniebox lighttpd[766]: error adding /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Katze 2/1 - ABC, die Katze lief im Schnee.mp3: No such directory
May 26 19:59:16 phoniebox lighttpd[766]: error adding /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Katze 2/1 - ABC, die Katze lief im Schnee.mp3: No such directory
May 26 19:59:19 phoniebox lighttpd[766]: error adding /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Katze 2/2 - Fünf kleine Fische.mp3: No such directory
May 26 19:59:31 phoniebox lighttpd[766]: error adding /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Katze 2/1 - ABC, die Katze lief im Schnee.mp3: No such directory
May 26 20:01:14 phoniebox lighttpd[766]: error adding /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Recordings/2020-04-07_21-27-55.wav: No such directory
May 26 20:01:19 phoniebox lighttpd[766]: error adding /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Recordings/2020-04-07_21-27-55.wav: No such directory`

s-martin, 2020-05-26:
Just for clarification: Are the files mentioned in the error log the files you cannot play?

BerniPi, 2020-05-26:
Yes, but after toggling "Resume" in the webinterface and refreshing. All playlists work.

BerniPi, 2020-05-26:
I'm not sure anymore, if this errors are connected with my issues. Now i don't get any errors in syslog.
I now tried it again: The first folder started working with second click on play.
The other folders don't start working on their own.
Podcasts work directly after install/update.
I only get the three errors
May 26 19:58:11 phoniebox lighttpd[766]: cat: /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_Folder_Played: No such file or directory
May 26 19:58:11 phoniebox lighttpd[766]: cat: /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_Playlist_Played: No such file or directory
May 26 19:58:12 phoniebox lighttpd[766]: cat: /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_Folder_Played: No such file or directory

again.
The files have ownership root:root. When I delete them, they are created with the same ownership again.

MiczFlor, 2020-05-27:
Ownership root I just changed. But they should be 777 anyway, meaning they should be readable and writeable from anywhere.

MiczFlor, 2020-06-02:
Can you try on the command line `mpc rescan` and see if it works then?
I just added this t the startup script to update the db after booting.

BerniPi, 2020-06-03:
Great! Works.

closed 2020-06-03 by BerniPi