MiczFlor/RPi-Jukebox-RFID #982: 🐛 | Single file play (#682) and append file to playlist ( #785) don't work with filenames containing '

BerniPi, 2020-05-26:
## Bug

### What I did
I tried to play a single file.

### What happened
Most files play fine, but files with ' in filename don't. (eg. "Alle Leut'" or "Ein Schneider fing 'ne Maus" 

### I expected this to happen
It works for all files.

### Further information that might help
There is no information in syslog and debug.log (I enabled debugging for Webapp_API and playoutcontrols.

## Software

### Base image and version
Raspian Buster Lite

### Branch / Release
latest develop

### Installscript
buster-install-default

## Hardware

### RaspberryPi version
3B

### RFID Reader

None

### Soundcard
PCM

### Other notable hardware
None

s-martin, 2020-05-26:
So what happens, when you try to play the file? Nothig or an error?

Could you also post the results of `tail -n 500 /var/log/syslog`? Thanks!

BerniPi, 2020-05-26:
There is no reaction / no error.
Results of tail syslog:
`May 26 22:06:30 phoniebox kernel: [    0.000297] console [tty1] enabled
May 26 22:06:30 phoniebox kernel: [    0.000341] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
May 26 22:06:30 phoniebox systemd[1]: Started Load/Save Random Seed.
May 26 22:06:30 phoniebox kernel: [    0.000356] pid_max: default: 32768 minimum: 301
May 26 22:06:30 phoniebox kernel: [    0.000696] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
May 26 22:06:30 phoniebox kernel: [    0.000712] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
May 26 22:06:30 phoniebox kernel: [    0.001662] CPU: Testing write buffer coherency: ok
May 26 22:06:30 phoniebox kernel: [    0.002124] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
May 26 22:06:30 phoniebox kernel: [    0.002782] Setting up static identity map for 0x100000 - 0x10003c
May 26 22:06:30 phoniebox systemd[1]: Started Create System Users.
May 26 22:06:30 phoniebox kernel: [    0.002937] rcu: Hierarchical SRCU implementation.
May 26 22:06:30 phoniebox kernel: [    0.003741] smp: Bringing up secondary CPUs ...
May 26 22:06:30 phoniebox kernel: [    0.004638] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
May 26 22:06:30 phoniebox systemd[1]: Starting Create Static Device Nodes in /dev...
May 26 22:06:30 phoniebox kernel: [    0.005635] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
May 26 22:06:31 phoniebox systemd[1]: Started Flush Journal to Persistent Storage.
May 26 22:06:31 phoniebox kernel: [    0.006545] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
May 26 22:06:31 phoniebox kernel: [    0.006668] smp: Brought up 1 node, 4 CPUs
May 26 22:06:31 phoniebox systemd[1]: Started Create Static Device Nodes in /dev.
May 26 22:06:31 phoniebox kernel: [    0.006680] SMP: Total of 4 processors activated (153.60 BogoMIPS).
May 26 22:06:31 phoniebox kernel: [    0.006686] CPU: All CPU(s) started in HYP mode.
May 26 22:06:31 phoniebox kernel: [    0.006691] CPU: Virtualization extensions available.
May 26 22:06:31 phoniebox kernel: [    0.007664] devtmpfs: initialized
May 26 22:06:31 phoniebox systemd[1]: Reached target Local File Systems (Pre).
May 26 22:06:31 phoniebox kernel: [    0.020700] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4May 26 22:06:31 phoniebox kernel: [    0.020937] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
May 26 22:06:31 phoniebox systemd[1]: Starting udev Kernel Device Manager...
May 26 22:06:31 phoniebox kernel: [    0.020958] futex hash table entries: 1024 (order: 4, 65536 bytes)
May 26 22:06:31 phoniebox systemd[1]: Started udev Kernel Device Manager.
May 26 22:06:31 phoniebox kernel: [    0.021551] pinctrl core: initialized pinctrl subsystem
May 26 22:06:31 phoniebox kernel: [    0.022407] NET: Registered protocol family 16
May 26 22:06:31 phoniebox kernel: [    0.025408] DMA: preallocated 1024 KiB pool for atomic coherent allocations
May 26 22:06:31 phoniebox kernel: [    0.031094] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
May 26 22:06:31 phoniebox kernel: [    0.031102] hw-breakpoint: maximum watchpoint size is 8 bytes.
May 26 22:06:31 phoniebox kernel: [    0.031302] Serial: AMBA PL011 UART driver
May 26 22:06:31 phoniebox kernel: [    0.033645] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
May 26 22:06:31 phoniebox kernel: [    0.050039] raspberrypi-firmware soc:firmware: Attached to firmware from 2020-02-12 12:38, variant start
May 26 22:06:31 phoniebox kernel: [    0.060052] raspberrypi-firmware soc:firmware: Firmware hash is 53a54c770c493957d99bf49762dfabc4eee00e45
May 26 22:06:31 phoniebox systemd[1]: Found device /dev/ttyAMA0.
May 26 22:06:31 phoniebox kernel: [    0.103213] bcm2835-dma 3f007000.dma: DMA legacy API manager at (ptrval), dmachans=0x1
May 26 22:06:31 phoniebox kernel: [    0.104848] SCSI subsystem initialized
May 26 22:06:31 phoniebox kernel: [    0.105072] usbcore: registered new interface driver usbfs
May 26 22:06:31 phoniebox kernel: [    0.105126] usbcore: registered new interface driver hub
May 26 22:06:31 phoniebox kernel: [    0.105221] usbcore: registered new device driver usb
May 26 22:06:31 phoniebox mtp-probe: checking bus 1, device 3: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1"
May 26 22:06:31 phoniebox kernel: [    0.106782] clocksource: Switched to clocksource arch_sys_counter
May 26 22:06:31 phoniebox kernel: [    0.196725] VFS: Disk quotas dquot_6.6.0
May 26 22:06:31 phoniebox mtp-probe: bus: 1, device: 3 was not an MTP device
May 26 22:06:31 phoniebox kernel: [    0.196834] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
May 26 22:06:31 phoniebox kernel: [    0.196984] FS-Cache: Loaded
May 26 22:06:31 phoniebox kernel: [    0.197190] CacheFiles: Loaded
May 26 22:06:31 phoniebox kernel: [    0.207233] NET: Registered protocol family 2
May 26 22:06:31 phoniebox kernel: [    0.208071] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes)May 26 22:06:31 phoniebox kernel: [    0.208108] TCP established hash table entries: 8192 (order: 3, 32768 bytes)
May 26 22:06:31 phoniebox systemd-udevd[160]: Using default interface naming scheme 'v240'.
May 26 22:06:31 phoniebox kernel: [    0.208219] TCP bind hash table entries: 8192 (order: 4, 65536 bytes)
[... cut ...]
May 26 22:07:40 phoniebox lighttpd[777]: cat: /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_Folder_Played: No such file or directory
May 26 22:07:40 phoniebox lighttpd[777]: cat: /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_Playlist_Played: No such file or directory
May 26 22:07:40 phoniebox lighttpd[777]: cat: /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_Folder_Played: No such file or directory
May 26 22:07:46 phoniebox systemd[1]: phoniebox-idle-watchdog.service: Succeeded.`

s-martin, 2020-05-30:
Probably not related to the problem, but you have a lot of errors like

```
May 26 22:06:40 phoniebox daemon_rfid_reader.py[862]: Please run RegisterDevice.py first
May 26 22:06:40 phoniebox systemd[1]: phoniebox-rfid-reader.service: Main process exited, code=exited, status=1/FAILURE
```

Do you use an RFID reader or do you intend to use one?

BerniPi, 2020-05-30:
No, it's my rpi to test. No reader installed.

s-martin, 2020-05-30:
I guess it’s a web interface issue, so I asked for the wrong log, sorry.

Could you post the output of `sudo tail -n 300 /var/log/lighttpd/error.log`?

BerniPi, 2020-06-01:
2020-06-01 14:13:01: (server.c.1759) logfiles cycled UID = 0 PID = 982
2020-06-01 14:13:23: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: MAILWLANIPYN in /home/pi/RPi-Jukebox-RFID/htdocs/inc.header.php on line 158
2020-06-01 14:13:23: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: MAILWLANIPADDR in /home/pi/RPi-Jukebox-RFID/htdocs/inc.header.php on line 159
2020-06-01 14:13:23: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  usort() expects exactly 2 parameters, 1 given in /home/pi/RPi-Jukebox-RFID/htdocs/inc.viewFolderTree.php on line 181
2020-06-01 14:13:23: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  usort() expects exactly 2 parameters, 1 given in /home/pi/RPi-Jukebox-RFID/htdocs/inc.viewFolderTree.php on line 181
2020-06-01 14:13:23: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  usort() expects exactly 2 parameters, 1 given in /home/pi/RPi-Jukebox-RFID/htdocs/inc.viewFolderTree.php on line 181
2020-06-01 14:13:23: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  usort() expects exactly 2 parameters, 1 given in /home/pi/RPi-Jukebox-RFID/htdocs/inc.viewFolderTree.php on line 181
2020-06-01 14:13:23: (mod_fastcgi.c.421) FastCGI-stderr: PHP Warning:  usort() expects exactly 2 parameters, 1 given in /home/pi/RPi-Jukebox-RFID/htdocs/inc.viewFolderTree.php on line 181

s-martin, 2020-06-01:
Could you manually check, if your issue is fixed, if you manually apply the change in #993?

BerniPi, 2020-06-01:
Yes, these errors are gone!

s-martin, 2020-06-01:
So ist the behavior also gone or just the warnings in the log file?

I can't reproduce, because I don't have a working Pi right now.

BerniPi, 2020-06-01:
No, the behaviour is the same. Just the warnings are gone.
I don't have time now to investigate it further and i don't use these functions. So it has no high priority for me.

MiczFlor, 2020-06-02:
Mental note: investigate escaping single and double quotes for mpd / mpc
https://www.musicpd.org/doc/html/protocol.html#escaping-string-values

String values are quoted with single or double quotes, and special characters within those values must be escaped with the backslash (\). Keep in mind that the backslash is also the escape character on the protocol level, which means you may need to use double backslash.

BerniPi, 2020-06-03:
I did some further testing. I think that it's a problem with html.
Debugging works with "normal filenames", but there is no single debugging line from "htdocs/api/playlist/playsinglefile.php ", if i try a filename containing '.
Foldernames containing ' are  problematic too.
[image]

BerniPi, 2020-06-03:
OK. Two changes are necessary:
-Everywhere where folder or filenames are used in the webinterface the command htmlspecialchars (https://www.php.net/manual/de/function.htmlspecialchars.php) has to be added. Then the html-code is corrected.
- mpd/mpc is not a problem, but the bash scripts: https://superuser.com/questions/163515/bash-how-to-pass-command-line-arguments-containing-special-characters

The single file play is working for me now. But i'll go to bed now. I will open a pullrequest soon.

BerniPi, 2020-06-08:
I also looked into making it possible to use single quotes in folder names, but there are also problems with creating folders in webinterface containing other special characters like double quotes or $

PR #1004 (2020-06-09): Fix "Play single file" to play filenames containing single quotes and appendFileToPlaylist
First part to fix #982
closed 2020-06-09 by MiczFlor