MiczFlor/RPi-Jukebox-RFID #985: 🐛 | Missing Level Shifter for LCD HD44780

Hans-Haefner, 2020-05-28:
## Bug

The HD44780 and the I2C Adapter uses a Level of 5V which is for Arduino ok but Raspberry Pi works with 3,3V Levels

### What I did

Connect a Display HD44780 to I2C 

### What happened

After a restart, the characters are sometimes displayed incorrectly

### I expected this to happen

### Further information that might help

Using al Level  Shifter or I2C Logic Level Converter

## Software

https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/displays/HD44780-i2c

### Base image and version

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release

develop

### Installscript

scripts/installscripts/buster-install-default.sh

## Hardware

### RaspberryPi version

3 B Rev 1.2

### RFID Reader

Neuftech USB Reader

### Soundcard

DACBerry One+

### Other notable hardware

vol0 = Button(9,pull_up=True)
stop = Button(11,pull_up=True)
next = Button(7,pull_up=True,hold_time=2.0,hold_repeat=False)
prev = Button(27,pull_up=True,hold_time=2.0,hold_repeat=False)
halt = Button(12,pull_up=True)

vol0.when_pressed = def_vol0
stop.when_pressed = def_stop

next.when_pressed = def_next
next.when_held = def_contrastup
prev.when_pressed = def_prev
prev.when_held = def_contrastdown
halt.when_pressed = def_halt

s-martin, 2020-05-30:
If I understand you correctly, an extra piece of hardware is needed to fix this problem?

Hans-Haefner, 2020-05-30:
I found out that the incorrectly displayed characters are not present despite the level shifter, but such a converter is still needed to prevent damage. The 5V from the I2C adapter of the LCD simply do not match with 3.3V of the I2C of the Raspberry.
Now I have to find out why sometimes some characters are displayed incorrectly at startup.

closed 2020-05-30 by Hans-Haefner

Hans-Haefner, 2020-06-05:
I don't know why this happens but the confused characters appear sporadically after a restart. I am also not sure whether this is only with my system or whether it is a software problem.
[image]