MiczFlor/RPi-Jukebox-RFID #986: One line install script doesnt seem to work anymore

deformatik-debug, 2020-05-29:
## During the installation with the "one line install command" there are several "Fehlschlag beim holen von xxx" notices (see CLI output in the attachment).

### Setup raspian OS buster on SD card. Put SD card in an Raspberry Pi 3 Model B. Connected via PuttY-->SSH2 to Pi3. Updated and Upgraded raspian. Copy and pasted the "one line install command" into the CLI. Started the installation of Phoniebox. 

### Sevreal "Fehlschlag beim holen von xxx" notices during the installation, eventhough internet connection is workking. Have tried the installation 10 times, always with the same result.

[image]

s-martin, 2020-05-29:
Please execute `sudo apt update` before starting the install script. I assume the apt mirrors are not up to date.

deformatik-debug, 2020-05-29:
That was the solution. It's weird, I could have sworn that I did sudo apt update as well as sudo apt upgrade when I first set up the pi two days ago. However, thank you for the quick response.

closed 2020-05-29 by deformatik-debug