MiczFlor/RPi-Jukebox-RFID #988: 🐛 | SSL stream does not play - certificate failed to verify

topas-rec, 2020-05-31:
## Bug

### What I did

Fresh install using Raspbian buster desktop and add stream URL and try to play it.
URL is https://wdr-diemaus-live.icecastssl.wdr.de/wdr/diemaus/live/mp3/128/stream.mp3

### What happened

The stream does not play and the issue when showing mpc error is shown below.

### I expected this to happen

Stream plays. This worked in the previous install. I even cannot get this running in my image backup anymore where it worked before. Maybe it has external causes like a changed certificate? I am lost here.

### Further information that might help

```
pi@raspberrypi:~ $ mpc
volume: n/a   repeat: off   random: off   single: off   consume: off
ERROR: Failed to decode https://wdr-diemaus-live.icecastssl.wdr.de/wdr/diemaus/live/mp3/128/stream.mp3; CURL failed: server certificate verification failed. CAfile: none CRLfile: none
```
PS: Simply setting it to `http` instead of `https` is not possible with this stream.

## Software

### Base image and version

`2020-05-27-raspios-buster-armhf.img`

### Branch / Release

master 2.0

### Installscript

`scripts/installscripts/buster-install-default.sh`

## Hardware

### RaspberryPi version

Pi Model 3B  V1.2

### RFID Reader

`16c0:27db HXGCoLtd Keyboard`

### Soundcard

onboard

### Other notable hardware

```
GPIO Buttons:
    volU = Button(22,pull_up=True,hold_time=0.3,hold_repeat=True)
    volD = Button(27,pull_up=True,hold_time=0.3,hold_repeat=True)
    frwd = Button(17,pull_up=True)
    prev = Button(24,pull_up=True)
    halt = Button(23,pull_up=True)
```

USB Stick that holds the audiofolders, `bind` mounted into `~/RPi-Jukebox-RFID/shared/audiofolders/`

s-martin, 2020-05-31:
I could play the link in a browser, but I reproduced the issue when trying to play the stream in Phoniebox.

Don’t know so far, what’s the issue.

topas-rec, 2020-05-31:
> I could play the link in a browser, but I reproduced the issue when trying to play the stream in Phoniebox.
> 
> Don’t know so far, what’s the issue.

Plays perfect in browser

drocx, 2020-05-31:
Hello,
it‘s maybe a certificate error with the stream (certificate is no longer valid?!)
I used the stream link for months. Yesterday the link works fine. Today I have the same bug. 
I have installed Phoniebox 2.0.

topas-rec, 2020-06-01:
This https stream in comparison works fine:
https://radio.stew.moe/stream/stream256.opus
Just the "Maus"-Stream fails. 

> Hello,
> the certificate for the stream is ok and valid.
> I used the stream link for months. Yesterday the link works fine. Today I have the same bug.
> I have installed Phoniebox 2.0.
> Mopidy Output: Oops, that link could not be found

@drocx thanks for your help. The Maus-Stream worked fine for me, too two days ago.

topas-rec, 2020-06-01:
I asked the guys from mpd.
https://github.com/MusicPlayerDaemon/MPD/issues/887

s-martin, 2020-06-01:
I'm not an expert for SSL, but I assume there's an issue in the chain of certificates for this URL. Maybe one of these needs to be updated in the Pi.

drocx, 2020-06-01:
@topas-rec 
I have updated 2.0 RC to Final Release a few days ago. No issues after the update.
My son hears WDR Maus ever day without any issues. 

After the stream stop working yesterday I installed every available update without success. 

I think it’s an issues with the root certificate that comes with Raspbian.

The stream works on iOS and windows without any Problems.

topas-rec, 2020-06-02:
Everyone having this issue: Can you please try again today with https://wdr-diemaus-live.icecastssl.wdr.de/wdr/diemaus/live/mp3/128/stream.mp3?

I tried to check if it works on a x86 Raspberry pi OS in a VM and it works before and after all packages update though it shows the curl certificate problem
Then I decided to test on the Raspberry Pi itself in the Phoniebox and... it works again.

drocx, 2020-06-02:
@topas-rec It works again. 🥳

closed 2020-06-03 by topas-rec

Sandoz9, 2023-05-09:
Sorry to re-open this topic but what was the solution in the end... having currently the same issue

s-martin, 2023-05-09:
Looks like a root certificate got invalid.

Don't know, if that's the case for your error right now, but if yes this can't be solved on Phoniebox's side.

Sandoz9, 2023-05-09:
I am actually using mpd... on a Raspberry Pi. So was curious to understand how I could potentially resolve this issue. 
Any idea ?

s-martin, 2023-05-09:
Sorry, my explanation was very brief.

The problem this issue was created for was an invalid (outdated) SSL certificate in the chain of certificates on the server side. So it had nothing to do with Phoniebox.

if you could post the link which is not working and also the output of `tail /var/log/syslog -n 200`, maybe we can try to pin down, if your problem is the same.
If yes, then there’s not much you can do except to wait until someone of the server maintainers fixes the certificate.

Sandoz9, 2023-05-10:
>  output of `tail /var/log/syslog -n 200`,

<details><summary>Output</summary>
<p>

May 10 16:30:00 moode systemd[1]: sysstat-collect.service: Succeeded.
May 10 16:30:00 moode systemd[1]: Finished system activity accounting tool.
May 10 16:33:26 moode winbindd[518]: [2023/05/10 16:33:26.493316,  0] ../../sour                                     ce3/rpc_server/rpc_ncacn_np.c:454(rpcint_dispatch)
May 10 16:33:26 moode winbindd[518]:   rpcint_dispatch: DCE/RPC fault in call ls                                     arpc:32 - DCERPC_NCA_S_OP_RNG_ERROR
May 10 16:33:26 moode winbindd[518]: [2023/05/10 16:33:26.496569,  0] ../../sour                                     ce3/rpc_server/rpc_ncacn_np.c:454(rpcint_dispatch)
May 10 16:33:26 moode winbindd[518]:   rpcint_dispatch: DCE/RPC fault in call ls                                     arpc:32 - DCERPC_NCA_S_OP_RNG_ERROR
May 10 16:38:26 moode winbindd[518]: [2023/05/10 16:38:26.496395,  0] ../../sour                                     ce3/rpc_server/rpc_ncacn_np.c:454(rpcint_dispatch)
May 10 16:38:26 moode winbindd[518]:   rpcint_dispatch: DCE/RPC fault in call ls                                     arpc:32 - DCERPC_NCA_S_OP_RNG_ERROR
May 10 16:38:26 moode winbindd[518]: [2023/05/10 16:38:26.499644,  0] ../../sour                                     ce3/rpc_server/rpc_ncacn_np.c:454(rpcint_dispatch)
May 10 16:38:26 moode winbindd[518]:   rpcint_dispatch: DCE/RPC fault in call ls                                     arpc:32 - DCERPC_NCA_S_OP_RNG_ERROR
May 10 16:40:00 moode systemd[1]: Starting system activity accounting tool...
May 10 16:40:00 moode systemd[1]: sysstat-collect.service: Succeeded.
May 10 16:40:00 moode systemd[1]: Finished system activity accounting tool.
May 10 16:43:26 moode winbindd[518]: [2023/05/10 16:43:26.499340,  0] ../../sour                                     ce3/rpc_server/rpc_ncacn_np.c:454(rpcint_dispatch)
May 10 16:43:26 moode winbindd[518]:   rpcint_dispatch: DCE/RPC fault in call ls                                     arpc:32 - DCERPC_NCA_S_OP_RNG_ERROR
May 10 16:43:26 moode winbindd[518]: [2023/05/10 16:43:26.502439,  0] ../../sour                                     ce3/rpc_server/rpc_ncacn_np.c:454(rpcint_dispatch)
May 10 16:43:26 moode winbindd[518]:   rpcint_dispatch: DCE/RPC fault in call ls                                     arpc:32 - DCERPC_NCA_S_OP_RNG_ERROR
May 10 16:48:26 moode winbindd[518]: [2023/05/10 16:48:26.501672,  0] ../../sour                                     ce3/rpc_server/rpc_ncacn_np.c:454(rpcint_dispatch)
May 10 16:48:26 moode winbindd[518]:   rpcint_dispatch: DCE/RPC fault in call ls                                     arpc:32 - DCERPC_NCA_S_OP_RNG_ERROR
May 10 16:48:26 moode winbindd[518]: [2023/05/10 16:48:26.504247,  0] ../../sour                                     ce3/rpc_server/rpc_ncacn_np.c:454(rpcint_dispatch)
May 10 16:48:26 moode winbindd[518]:   rpcint_dispatch: DCE/RPC fault in call ls                                     arpc:32 - DCERPC_NCA_S_OP_RNG_ERROR
May 10 16:50:00 moode systemd[1]: Starting system activity accounting tool...
May 10 16:50:00 moode systemd[1]: sysstat-collect.service: Succeeded.
May 10 16:50:00 moode systemd[1]: Finished system activity accounting tool.
May 10 16:53:26 moode winbindd[518]: [2023/05/10 16:53:26.503838,  0] ../../sour                                     ce3/rpc_server/rpc_ncacn_np.c:454(rpcint_dispatch)
May 10 16:53:26 moode winbindd[518]:   rpcint_dispatch: DCE/RPC fault in call ls                                     arpc:32 - DCERPC_NCA_S_OP_RNG_ERROR
May 10 16:53:26 moode winbindd[518]: [2023/05/10 16:53:26.506443,  0] ../../sour                                     ce3/rpc_server/rpc_ncacn_np.c:454(rpcint_dispatch)
May 10 16:53:26 moode winbindd[518]:   rpcint_dispatch: DCE/RPC fault in call ls                                     arpc:32 - DCERPC_NCA_S_OP_RNG_ERROR
May 10 16:58:26 moode winbindd[518]: [2023/05/10 16:58:26.506432,  0] ../../sour                                     ce3/rpc_server/rpc_ncacn_np.c:454(rpcint_dispatch)
May 10 16:58:26 moode winbindd[518]:   rpcint_dispatch: DCE/RPC fault in call ls                                     arpc:32 - DCERPC_NCA_S_OP_RNG_ERROR
May 10 16:58:26 moode winbindd[518]: [2023/05/10 16:58:26.509504,  0] ../../sour                                     ce3/rpc_server/rpc_ncacn_np.c:454(rpcint_dispatch)
May 10 16:58:26 moode winbindd[518]:   rpcint_dispatch: DCE/RPC fault in call ls                                     arpc:32 - DCERPC_NCA_S_OP_RNG_ERROR
May 10 17:00:00 moode systemd[1]: Starting system activity accounting tool...
May 10 17:00:00 moode systemd[1]: sysstat-collect.service: Succeeded.
May 10 17:00:00 moode systemd[1]: Finished system activity accounting tool.
May 10 17:03:26 moode winbindd[518]: [2023/05/10 17:03:26.508274,  0] ../../sour                                     ce3/rpc_server/rpc_ncacn_np.c:454(rpcint_dispatch)
May 10 17:03:26 moode winbindd[518]:   rpcint_dispatch: DCE/RPC fault in call ls                                     arpc:32 - DCERPC_NCA_S_OP_RNG_ERROR
May 10 17:03:26 moode winbindd[518]: [2023/05/10 17:03:26.511224,  0] ../../sour                                     ce3/rpc_server/rpc_ncacn_np.c:454(rpcint_dispatch)
May 10 17:03:26 moode winbindd[518]:   rpcint_dispatch: DCE/RPC fault in call ls                                     arpc:32 - DCERPC_NCA_S_OP_RNG_ERROR
May 10 17:07:53 moode rngd[373]: stats: bits received from HRNG source: 700064
[... cut ...]
May 10 18:44:48 moode systemd[21291]: Reached target Sockets.
May 10 18:44:48 moode systemd[21291]: Reached target Basic System.
May 10 18:44:48 moode systemd[21291]: Reached target Main User Target.
May 10 18:44:48 moode systemd[1]: Started User Manager for UID 1000.
May 10 18:44:48 moode systemd[21291]: Startup finished in 938ms.
May 10 18:44:48 moode systemd[1]: Started Session 6 of user pi.

</p>
</details> 

Here the link of one of the many radio stations that fail to connect.
http://direct.fipradio.fr/live/fip-midfi.mp3

s-martin, 2023-05-10:
Looks like there was no message in the log corresponding to Phoniebox.

maybe try to use one of the streams, which are not working and then run the tail command.

Sandoz9, 2023-05-10:
I don't use Phoniebox but Moode... And I tried to launch the failing URL before getting the log.

s-martin, 2023-05-10:
Okay, sorry this is the Phoniebox project.

maybe you should ask the question here: https://github.com/moode-player/moode

Sandoz9, 2023-05-11:
Yes, I just thought that was the same problem. I will post on the moode forum. Thank you