MiczFlor/RPi-Jukebox-RFID #991: 🚀 | Integrate GPIO Control correctly

s-martin, 2020-05-31:
Since 2.0 we have added [GPIO Controls](https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/Master/components/gpio_control), many thanks to @veloxidSchweiz.

At the moment it’s not installed yet in the default installation, but is required at least for rotary encoder.

- [ ] Integrate install.sh in regular installation script 
- [ ] Deactivate outdated gpio-buttons and rotary-encoder services and remove old code
- [ ] Update documentation in wiki and `/components/gpio_control`

Information can be found in #900, PR #990 could probably be closed.

jlarsch, 2020-07-01:
given the GPIO issues with 2.0, is there a one line install routine to roll back and re-install the previous version? Unfortunately, my backup broke

arne123, 2020-08-17:
just short comment to the wiki page and future documentation https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Audio-RotaryKnobVolume
To reverse the rotation direction of an Rotary Enocder, you just need to Swap the A a B Pins in a Config File

themorlan, 2020-08-20:
@s-martin 
>     * [ ]  Integrate install.sh in regular installation script

How should this be implemented? As an interactive section in the script, which contains parts of the [https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/gpio_control/install.sh](url)?

s-martin, 2020-08-20:
@themorlan, yes that would be my approach. Similar to the selection of RFID readers already in the script.

themorlan, 2020-08-20:
Whats the current status? Any work already done? 
I'm a novice in shell scripting but I might try to implement it.

s-martin, 2020-08-21:
Not really something done, `develop` branch contains everything implemented so far.

closed 2020-09-15 by s-martin