MiczFlor/RPi-Jukebox-RFID #992: Allow Tonies to be used with the phoniebox🚀 | FEATURE SUMMARY

Erechtheus, 2020-06-01:
## Feature Description
It would be great if we could use the _pretty_ Tonies figurines. As far as I understood, Tonies  use the 13.56MHz frequency band, but custom handshakes. I understand that this feature probably requires some engineering. However, it would be great if we could collect some information. E.g., details about the custom protocoll and so on 

### What functionality would you like to see in your phoniebox?
I would love to see that Tonies can be used with the phoniebox :-)

### How do you envision the feature to work from a users perspective?
Put Tonies on the reader 

### Further information that might help
NA

s-martin, 2020-06-01:
I don't think this would be possible, because AFAIK Tonies are somehow  protected against copying.

So there won't be any documentation available and reverse engineering would be necessary.

Overcoming the copy protection would probably have some legal implications, too.

Maybe @MiczFlor has some more info as he decided to implement his own Phoniebox, so he maybe investigated Toniebox before starting his own implementation.

Erechtheus, 2020-06-01:
I'm also not totally sure about the copyright mechanism implementation. There was a talk at a chaos communication congress (I think last year), which I don't fully recall at this moment. I'll try to watch it later again.

Anyway, I am **not trying to copy**  the content of the figurines, but rather using the ID of the figurines to play my own music. So I just would like to use the pretty figurines :-). AFAIK the biggest problem is that Tonies don't reply to standard RFID-authentication protocol.

s-martin, 2020-06-02:
Ok, I think just using them as a trigger could work, but still somebody would need to reverse engineer that.

MiczFlor, 2020-06-02:
Hi @Erechtheus 
when you swipe the tonie figure over the RFID reader, does Phoniebox actually get the ID from the figure? It should show up in the web app if you go to 'card ID'

drocx, 2020-06-03:
Here are more Infos about the tonies.
You can not get the id from a tonie without a code changes because of custom handshakes.

http://gt-blog.de/

Erechtheus, 2020-06-03:
Hi @MiczFlor, indeed the Tonies use some authentication protocol and do not expose the IDs directly. The RFID-handshake sounds actually not so difficult to implement. However, I need a reader compatible with ISO15693 and my RC522 reader is not.  At the moment, I am trying to find an RFID reader which is ISO15693 compatible. 

Hi @drocx thank you for the information. I'll give it a look :-)

drocx, 2020-06-03:
Try a PN5180 Reader.

MiczFlor, 2020-06-03:
it's quite an effort to include closed source hardware into an open source project :)

Erechtheus, 2020-06-03:
If I understood correctly, the authentication process works as follows:

- Send  0xB2 0x04 to the figurine
- Receive 16-bit response (random number)
- XOR the response with following HEX: 0xB5 0x6E 0xFD 0x7F
- Send 0xB3 0x04 0x04 + XOR-response to the figurine
- Receive UID :) 

I think this [script](https://gist.github.com/ceres-c/cbf437f9af9e946c96550fd7e1e77cef) does all the magic. The script is provided by one of the CCC guys (see below). Requires a CR95HF.

Helpful links:

[CCC talk](https://media.ccc.de/v/36c3-108-hacking-an-nfc-toy-with-the-chameleonmini)
[Code snipped](https://de.ifixit.com/Antworten/Ansehen/513422/nfc+Chips+f%C3%BCr+tonies+kaufen)
https://gist.github.com/ceres-c/cbf437f9af9e946c96550fd7e1e77cef

Erechtheus, 2020-06-04:
> Try a PN5180 Reader.

@drocx Can you recommend any specific hardware using this integrated circuit? The only boards I found cost >30€.

drocx, 2020-06-05:
@Erechtheus The PN8180 Reader is available at amazon for 7 Euro. 
I don't have this reader but its recommend [here](http://gt-blog.de/inside-tonie-figurine/).

s-martin, 2021-09-02:
#644

s-martin, 2023-12-29:
Maybe this is helpful:

https://youtu.be/DNufX-tss5M?si=jfyHIeWFi4Cs-DxS