MiczFlor/RPi-Jukebox-RFID #994: Sound only from HDMI but not from audio jack

Lumpengnom, 2020-06-01:
Hi,
I installed the phoniebox software on a fresh install of raspian buster with the one-line install script.
It is on a raspberry pi 2.
I then uploaded some music to it via the web interface from another coputer.
It plays music but only if I plug my headphones into the hdmi monitor connected to the phoniebox. 
If I plug th headphones into the audio jack it does not play any music.
I allready right clicked the speaker icon in the raspberry and set it to analog but nothing changed. Music can still be heard when plugged into the hdmi monitor but not when plugged into the audio jack.

The audo jack does work because if i play a file directrly on the the raspberry, for example with vlc player it works.

I suspect it has something to do with mpd but I can not figure out what it is.

Does anybody know what the problem is?

Lumpengnom, 2020-06-01:
Ok, after trying some more it appears to work now.
I went to:
`sudo nano /etc/mpd.conf`

If you scroll down you get to place where it says "audio_output"

Here I uncommented the "device" line and set it from "hw:0,0" to "hw:1,0"

s-martin, 2020-06-02:
For reference the Raspberry Pi changed the audio behavior in the last update in May. Please see also #973 (the last comments).

s-martin, 2020-06-02:
Closing this as it's fixed. Please feel free to reopen the issue, if you still experience this issue.

closed 2020-06-02 by s-martin