MiczFlor/RPi-Jukebox-RFID #995: Spotify doesn't work anymore Spotify login error: <ErrorType.OTHER_PERMANENT: 10>

swissmaster1, 2020-06-02:
My Phoniebox was running now for half a year without a problem. But since 1 or 2 weeks spotify doesn't work anymore. When I restart Mopidy I get fallowing messages from syslog:
`Jun  2 21:40:54 box mopidy[2788]: INFO     Starting Mopidy 2.2.2
Jun  2 21:40:54 box mopidy[2788]: INFO     Loading config from builtin defaults
Jun  2 21:40:54 box mopidy[2788]: INFO     Loading config from /etc/mopidy/mopidy.conf
Jun  2 21:40:54 box mopidy[2788]: INFO     Loading config from command line options
Jun  2 21:40:54 box mopidy[2788]: INFO     Enabled extensions: iris, mpd, http, stream, spotify, m3u, softwaremixer, file, local-images, local
Jun  2 21:40:54 box mopidy[2788]: INFO     Disabled extensions: none
Jun  2 21:40:54 box mopidy[2788]: INFO     Starting Mopidy mixer: SoftwareMixer
Jun  2 21:40:54 box mopidy[2788]: INFO     Mixer volume set to 30
Jun  2 21:40:54 box mopidy[2788]: INFO     Starting Mopidy audio
Jun  2 21:40:54 box mopidy[2788]: INFO     Starting Mopidy backends: SpotifyBackend, StreamBackend, M3UBackend, FileBackend, LocalBackend
Jun  2 21:40:55 box mopidy[2788]: INFO     Audio output set to "alsasink"
Jun  2 21:40:55 box python3[243]: mpd error: Connection refused
Jun  2 21:40:55 box mopidy[2788]: ERROR    Spotify login error: <ErrorType.OTHER_PERMANENT: 10>
Jun  2 21:40:56 box python3[243]: mpd error: Connection refused
Jun  2 21:40:56 box mopidy[2788]: INFO     Loaded 1359 local tracks using json
Jun  2 21:40:57 box mopidy[2788]: INFO     Logged into Spotify Web API as 116859334
Jun  2 21:40:57 box mopidy[2788]: INFO     Started spotify backend
Jun  2 21:40:57 box mopidy[2788]: INFO     Starting Mopidy core
Jun  2 21:40:57 box mopidy[2788]: INFO     Starting Mopidy frontends: IrisFrontend, MpdFrontend, HttpFrontend
Jun  2 21:40:57 box mopidy[2788]: INFO     Starting Iris 3.43.0
Jun  2 21:40:57 box mopidy[2788]: INFO     MPD server running at [::ffff:0.0.0.0]:6600
Jun  2 21:40:57 box mopidy[2788]: INFO     Starting GLib mainloop
Jun  2 21:40:57 box mopidy[2788]: INFO     HTTP server running at [::ffff:0.0.0.0]:6680`

I search already for some hours but wasn't able to find the reason. Restart of Mopidy doesn't change anything.
I then thought, ok let's do a clean install with raspian buster and the current phoniebox release. As far as I remember the current release uses Mopidy 3. But I still have the same problem and the same error Spotify login error: <ErrorType.OTHER_PERMANENT: 10>. The Spotify logindata is correct and is working.  I tried as well with a new client_id etc. from https://mopidy.com/ext/spotify/#authentication. But this doesn't change anything.
In Iris web interface I can see all the album covers but when I click to play something I get "Mopidy: Failed to add some tracks".

Did something change with spotify integration?

s-martin, 2020-06-02:
It's possible that there's currently a problem with Spotify in general, so maybe it goes away.

**But** as far as I know Mopidy 2 uses an outdated login method to Spotify, which was deprecated a while ago and maybe now turned off.

Phoniebox 2.0 was updated to Mopidy 3 which should **still** work.

So you could try to install 2.0 and see if this works. Be aware that 2.0 has currently some issues with rotary knobs (see #991) , if you use these.

Make sure you backup your music, playlists, RFID files, etc.

swissmaster1, 2020-06-02:
I already tried an update. Got same error. As well with a clean install of buster and Phoniebox 2.0. Same error.

s-martin, 2020-06-02:
That’s strange. What‘s the output of `journalctl -u mopidy`, when using the updated version?

swissmaster1, 2020-06-02:
Jun 02 23:38:47 box systemd[1]: Starting Mopidy music server...
Jun 02 23:38:47 box systemd[1]: Started Mopidy music server.
Jun 02 23:38:57 box mopidy[493]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Jun 02 23:38:57 box mopidy[493]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Jun 02 23:38:57 box mopidy[493]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Jun 02 23:38:57 box mopidy[493]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Jun 02 23:38:57 box mopidy[493]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Jun 02 23:38:57 box mopidy[493]: INFO     [MainThread] mopidy.config Loading config from command line options
Jun 02 23:38:57 box mopidy[493]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Jun 02 23:38:58 box mopidy[493]: INFO     [MainThread] mopidy.__main__ Enabled extensions: mpd, local, spotify, softwaremixer, iris, http, m3u, stream, file
Jun 02 23:38:58 box mopidy[493]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Jun 02 23:39:01 box mopidy[493]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Jun 02 23:39:01 box mopidy[493]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Jun 02 23:39:01 box mopidy[493]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Jun 02 23:39:01 box mopidy[493]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Jun 02 23:39:01 box mopidy[493]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
**Jun 02 23:39:02 box mopidy[493]: ERROR    [SpotifyEventLoop] spotify.session Spotify login error: <ErrorType.OTHER_PERMANENT: 10>**
**Jun 02 23:39:02 box mopidy[493]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as 123456789**
**Jun 02 23:40:10 box mopidy[493]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 137 Spotify playlists**
Jun 02 23:40:10 box mopidy[493]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Jun 02 23:40:10 box mopidy[493]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, IrisFrontend, MpdFrontend
Jun 02 23:40:10 box mopidy[493]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Jun 02 23:40:10 box mopidy[493]: INFO     [IrisFrontend-13] mopidy_iris.core Starting Iris 3.45.0
Jun 02 23:40:10 box mopidy[493]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Jun 02 23:40:10 box mopidy[493]: INFO     [MainThread] mopidy.commands Starting GLib mainloop

swissmaster1, 2020-06-02:
IRIS Webinterface says Spotify connected and authorized. I can see all my playlists.

swissmaster1, 2020-06-02:
Oh man!!! I may be found the error.
AdGuard is maybe blocking something.

swissmaster1, 2020-06-02:
[image]

That was it! Now it's working. Omg :-o

closed 2020-06-02 by swissmaster1