MiczFlor/RPi-Jukebox-RFID #1000: problems with spotify in version 2.0🐛 | BUG SUMMARY

martin2345, 2020-06-05:
## Bug
I can play local files and stream links like webradio.
I cannot play any spotify stream.

### What I did

### What happened

## Hardware

### RaspberryPi version

<
4B
-->

s-martin, 2020-06-05:
Could you post the output of `tail -n 500 /var/log/syslog`?

martin2345, 2020-06-05:
Sure I can,thanks for the quick reply

Jun  5 18:31:19 raspberrypi systemd-fsck[291]: fsck.fat 4.1 (2017-01-24)
Jun  5 18:31:19 raspberrypi systemd-fsck[291]: 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Jun  5 18:31:19 raspberrypi systemd-fsck[291]:  Automatically removing dirty bit.
Jun  5 18:31:19 raspberrypi systemd-fsck[291]: Performing changes.
Jun  5 18:31:19 raspberrypi systemd-fsck[291]: /dev/mmcblk0p1: 238 files, 104465/516190 clusters
Jun  5 18:31:19 raspberrypi systemd[1]: Started File System Check on /dev/disk/by-partuuid/ab4f2b40-01.
Jun  5 18:31:19 raspberrypi systemd[1]: Mounting /boot...
Jun  5 18:31:19 raspberrypi systemd-udevd[189]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Jun  5 18:31:19 raspberrypi systemd[1]: Mounted /boot.
Jun  5 18:31:19 raspberrypi systemd[1]: Reached target Local File Systems.
Jun  5 18:31:19 raspberrypi systemd[1]: Condition check resulted in Commit a transient machine-id on disk being skipped.
Jun  5 18:31:19 raspberrypi systemd[1]: Condition check resulted in Load AppArmor profiles being skipped.
Jun  5 18:31:19 raspberrypi systemd[1]: Starting Set console font and keymap...
Jun  5 18:31:19 raspberrypi systemd[1]: Starting Create Volatile Files and Directories...
Jun  5 18:31:19 raspberrypi systemd[1]: Starting Preprocess NFS configuration...
Jun  5 18:31:19 raspberrypi systemd[1]: Starting Tell Plymouth To Write Out Runtime Data...
Jun  5 18:31:19 raspberrypi systemd[1]: Starting Raise network interfaces...
Jun  5 18:31:19 raspberrypi systemd[1]: Started Set console font and keymap.
Jun  5 18:31:19 raspberrypi systemd[1]: nfs-config.service: Succeeded.
Jun  5 18:31:19 raspberrypi systemd[1]: Started Preprocess NFS configuration.
Jun  5 18:31:19 raspberrypi systemd[1]: Condition check resulted in RPC security service for NFS client and server being skipped.
Jun  5 18:31:19 raspberrypi systemd[1]: Condition check resulted in RPC security service for NFS server being skipped.
Jun  5 18:31:19 raspberrypi systemd[1]: Reached target NFS client services.
Jun  5 18:31:19 raspberrypi systemd[1]: Reached target Remote File Systems (Pre).
Jun  5 18:31:19 raspberrypi systemd[1]: Reached target Remote File Systems.
Jun  5 18:31:19 raspberrypi systemd[1]: Started Create Volatile Files and Directories.
Jun  5 18:31:19 raspberrypi systemd[1]: Starting Update UTMP about System Boot/Shutdown...
Jun  5 18:31:19 raspberrypi systemd[1]: Starting Network Time Synchronization...
Jun  5 18:31:19 raspberrypi systemd[1]: Started Update UTMP about System Boot/Shutdown.
Jun  5 18:31:19 raspberrypi systemd[1]: Received SIGRTMIN+20 from PID 186 (plymouthd).
Jun  5 18:31:19 raspberrypi systemd[1]: plymouth-read-write.service: Succeeded.
Jun  5 18:31:19 raspberrypi systemd[1]: Started Tell Plymouth To Write Out Runtime Data.
Jun  5 18:31:19 raspberrypi systemd[1]: Started Raise network interfaces.
Jun  5 18:31:19 raspberrypi systemd-timesyncd[326]: System clock time unset or jumped backwards, restoring from recorded timestamp: Fri 2020-06-05 18:31:18 CEST
Jun  5 18:31:19 raspberrypi systemd[1]: Started Network Time Synchronization.
Jun  5 18:31:19 raspberrypi systemd[1]: Reached target System Initialization.
Jun  5 18:31:19 raspberrypi systemd[1]: Listening on D-Bus System Message Bus Socket.
Jun  5 18:31:19 raspberrypi systemd[1]: Listening on Avahi mDNS/DNS-SD Stack Activation Socket.
Jun  5 18:31:19 raspberrypi systemd[1]: Listening on triggerhappy.socket.
Jun  5 18:31:19 raspberrypi systemd[1]: Reached target Sockets.
Jun  5 18:31:19 raspberrypi systemd[1]: Reached target Basic System.
Jun  5 18:31:19 raspberrypi systemd[1]: Starting Disk Manager...
Jun  5 18:31:19 raspberrypi systemd[1]: Started Regular background program processing daemon.
Jun  5 18:31:19 raspberrypi systemd[1]: Starting Check for Raspberry Pi EEPROM updates...
Jun  5 18:31:19 raspberrypi cron[370]: (CRON) INFO (pidfile fd = 3)
Jun  5 18:31:19 raspberrypi systemd[1]: Starting Avahi mDNS/DNS-SD Stack...
Jun  5 18:31:19 raspberrypi systemd[1]: Condition check resulted in Copy user wpa_supplicant.conf being skipped.
Jun  5 18:31:19 raspberrypi systemd[1]: Starting dhcpcd on all interfaces...
Jun  5 18:31:19 raspberrypi systemd[1]: Starting Deferred execution scheduler...
Jun  5 18:31:19 raspberrypi systemd[1]: Condition check resulted in Turn on SSH if /boot/ssh is present being skipped.
Jun  5 18:31:19 raspberrypi cron[370]: (CRON) INFO (Running @reboot jobs)
Jun  5 18:31:19 raspberrypi systemd[1]: Started Manage Sound Card State (restore and store).
Jun  5 18:31:19 raspberrypi systemd[1]: Started D-Bus System Message Bus.
Jun  5 18:31:19 raspberrypi alsactl[377]: alsactl 1.1.8 daemon started
Jun  5 18:31:19 raspberrypi systemd[1]: Condition check resulted in getty on tty2-tty6 if dbus and logind are not available being skipped.
Jun  5 18:31:19 raspberrypi rpi-eeprom-update[371]: BCM2711 detected
Jun  5 18:31:19 raspberrypi rpi-eeprom-update[371]: Dedicated VL805 EEPROM detected
Jun  5 18:31:19 raspberrypi systemd[1]: Starting LSB: Switch to ondemand cpu governor (unless shift key is pressed)...
Jun  5 18:31:19 raspberrypi systemd[1]: Starting dphys-swapfile - set up, mount/unmount, and delete a swap file...
Jun  5 18:31:19 raspberrypi systemd[1]: Starting rng-tools.service...
Jun  5 18:31:19 raspberrypi udisksd[369]: udisks daemon version 2.8.1 starting
[... cut ...]
Jun  5 18:35:34 raspberrypi mopidy[511]: INFO     [MpdSession-26] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:59504
Jun  5 18:35:39 raspberrypi mopidy[511]: INFO     [MpdSession-27] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:59508
Jun  5 18:35:44 raspberrypi mopidy[511]: INFO     [MpdSession-28] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:59512
Jun  5 18:35:50 raspberrypi mopidy[511]: INFO     [MpdSession-29] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:59516

s-martin, 2020-06-05:
can you try `sudo systemctl restart mopidy` and if it doesn't work post `tail -n 500 /var/log/syslog` again?

martin2345, 2020-06-05:
Thanks, but unfortunately it didn't affect the problem. I attach the logfile you asked for:
 tail -n 500 /var/log/syslog:
Jun  5 19:24:07 raspberrypi bash[388]: LANG="en-UK"
Jun  5 19:24:07 raspberrypi bash[388]: VERSION="2.0 - ed888da - master"
Jun  5 19:24:07 raspberrypi bash[388]: 0
Jun  5 19:24:07 raspberrypi systemd[1]: Starting Deferred execution scheduler...
Jun  5 19:24:07 raspberrypi rng-tools[399]: Starting Hardware RNG entropy gatherer daemon: rngd.
Jun  5 19:24:07 raspberrypi rsyslogd: imuxsock: Acquired UNIX socket '/run/systemd/journal/syslog' (fd 3) from systemd.  [v8.1901.0]
Jun  5 19:24:07 raspberrypi thd[403]: Found socket passed from systemd
Jun  5 19:24:07 raspberrypi systemd[1]: Condition check resulted in Copy user wpa_supplicant.conf being skipped.
Jun  5 19:24:07 raspberrypi rsyslogd:  [origin software="rsyslogd" swVersion="8.1901.0" x-pid="382" x-info="https://www.rsyslog.com"] start
Jun  5 19:24:08 raspberrypi rngd[425]: rngd 2-unofficial-mt.14 starting up...
Jun  5 19:24:08 raspberrypi systemd[1]: Starting dhcpcd on all interfaces...
Jun  5 19:24:08 raspberrypi systemd[1]: Starting Avahi mDNS/DNS-SD Stack...
Jun  5 19:24:08 raspberrypi systemd[1]: Condition check resulted in getty on tty2-tty6 if dbus and logind are not available being skipped.
Jun  5 19:24:08 raspberrypi systemd[1]: Starting Login Service...
Jun  5 19:24:08 raspberrypi rngd[425]: entropy feed to the kernel ready
Jun  5 19:24:08 raspberrypi kernel: [    6.099562] random: crng init done
Jun  5 19:24:08 raspberrypi kernel: [    6.099583] random: 7 urandom warning(s) missed due to ratelimiting
Jun  5 19:24:08 raspberrypi systemd[1]: Started D-Bus System Message Bus.
Jun  5 19:24:08 raspberrypi avahi-daemon[438]: Found user 'avahi' (UID 108) and group 'avahi' (GID 113).
Jun  5 19:24:08 raspberrypi dhcpcd[430]: dev: loaded udev
Jun  5 19:24:08 raspberrypi avahi-daemon[438]: Successfully dropped root privileges.
Jun  5 19:24:08 raspberrypi avahi-daemon[438]: avahi-daemon 0.7 starting up.
Jun  5 19:24:08 raspberrypi systemd[1]: Starting WPA supplicant...
Jun  5 19:24:08 raspberrypi systemd[1]: Starting Disk Manager...
Jun  5 19:24:08 raspberrypi systemd[1]: Reached target System Time Synchronized.
Jun  5 19:24:08 raspberrypi systemd[1]: Started Daily rotation of log files.
Jun  5 19:24:08 raspberrypi systemd[1]: Started Daily apt download activities.
Jun  5 19:24:08 raspberrypi systemd[1]: Started Daily apt upgrade and clean activities.
Jun  5 19:24:08 raspberrypi systemd[1]: Started Daily man-db regeneration.
Jun  5 19:24:08 raspberrypi systemd[1]: Started Clean PHP session files every 30 mins.
Jun  5 19:24:08 raspberrypi systemd[1]: Reached target Timers.
Jun  5 19:24:08 raspberrypi dphys-swapfile[373]: want /var/swap=100MByte, checking existing: keeping it
Jun  5 19:24:08 raspberrypi systemd[1]: Started System Logging Service.
Jun  5 19:24:08 raspberrypi systemd[1]: Started triggerhappy global hotkey daemon.
Jun  5 19:24:08 raspberrypi kernel: [    6.260172] 8021q: 802.1Q VLAN Support v1.8
Jun  5 19:24:08 raspberrypi dhcpcd[430]: forked to background, child pid 468
Jun  5 19:24:08 raspberrypi systemd[1]: Started Check for v3d driver.
Jun  5 19:24:08 raspberrypi systemd[1]: Started rng-tools.service.
Jun  5 19:24:08 raspberrypi systemd[1]: Started Save/Restore Sound Card State.
Jun  5 19:24:08 raspberrypi systemd[1]: Started Deferred execution scheduler.
Jun  5 19:24:08 raspberrypi systemd[1]: Started dhcpcd on all interfaces.
Jun  5 19:24:08 raspberrypi avahi-daemon[438]: Successfully called chroot().
Jun  5 19:24:08 raspberrypi avahi-daemon[438]: Successfully dropped remaining capabilities.
Jun  5 19:24:08 raspberrypi avahi-daemon[438]: No service file found in /etc/avahi/services.
Jun  5 19:24:08 raspberrypi avahi-daemon[438]: Network interface enumeration completed.
Jun  5 19:24:08 raspberrypi avahi-daemon[438]: Server startup complete. Host name is raspberrypi.local. Local service cookie is 3368794369.
Jun  5 19:24:08 raspberrypi udisksd[462]: udisks daemon version 2.8.1 starting
Jun  5 19:24:08 raspberrypi dbus-daemon[443]: [system] Successfully activated service 'org.freedesktop.systemd1'
Jun  5 19:24:08 raspberrypi wpa_supplicant[461]: Successfully initialized wpa_supplicant
Jun  5 19:24:08 raspberrypi systemd[1]: Started WPA supplicant.
Jun  5 19:24:08 raspberrypi systemd[1]: Reached target Network.
Jun  5 19:24:08 raspberrypi systemd[1]: Started Phoniebox GPIO Buttons Service.
Jun  5 19:24:08 raspberrypi systemd[1]: Reached target Network is Online.
Jun  5 19:24:08 raspberrypi systemd[1]: Starting Lighttpd Daemon...
Jun  5 19:24:08 raspberrypi systemd[1]: Starting LSB: exim Mail Transport Agent...
Jun  5 19:24:08 raspberrypi kernel: [    6.486408] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
Jun  5 19:24:08 raspberrypi systemd[1]: Starting /etc/rc.local Compatibility...
Jun  5 19:24:08 raspberrypi systemd[1]: Started Phoniebox Rotary Encoder Service.
Jun  5 19:24:08 raspberrypi udisksd[462]: failed to load module crypto: libbd_crypto.so.2: cannot open shared object file: No such file or directory
Jun  5 19:24:08 raspberrypi udisksd[462]: failed to load module mdraid: libbd_mdraid.so.2: cannot open shared object file: No such file or directory
Jun  5 19:24:08 raspberrypi systemd[501]: phoniebox-rotary-encoder.service: Failed to execute command: No such file or directory
[... cut ...]
Jun  5 19:25:51 raspberrypi mopidy[565]: INFO     [MpdSession-46] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36240
Jun  5 19:25:56 raspberrypi mopidy[565]: INFO     [MpdSession-47] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36244
Jun  5 19:26:00 raspberrypi systemd[1]: phoniebox-idle-watchdog.service: Succeeded.
Jun  5 19:26:01 raspberrypi mopidy[565]: INFO     [MpdSession-48] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36248
pi@raspberrypi:~ $

s-martin, 2020-06-06:
Your Mopidy-Spotify package seems old.

Could you try `sudo pip3 install --upgrade Mopidy-Spotify`?

martin2345, 2020-06-06:
Well this is odd as I flashed a new os on the SD before installing phoniebox 2.0. The only thing I loaded back in were audiofolder and rfid shortcuts.
I did what you proposed but its not changing anything sadly. Should I do a complete new installation again? It said that now mopidy 4.0.1 is installed. Shouldn't it be 3.xx as I read somewhere else here?

s-martin, 2020-06-06:
Mopidy itself is version 3, the extension Mopidy-spotify is supposed to be version 4.

Did you restart mopidy with `sudo systemctl restart mopidy`?

Could you post the logs again?

martin2345, 2020-06-06:
Thanks so much. You literally made a child laughing loud for a few minutes when he realised the box is working again 😉.
It didn't work immediately though. I clicked rescan folders after rebooting,but then it worked. Do you still need the log? Thanks very much for the support!!! Danke und ein schönes Wochenende! I wonder what led to the wrong version but from my side I could close the issue.

s-martin, 2020-06-06:
Glad it works.

rescan the folders also restarts mopidy.

Could you post a screenshot of the rescan button?

Please post the log again as there sometimes still issues.

martin2345, 2020-06-06:
[image]

Hope that is what you meant. I'll post the log as soon as possible. When I link it to a monitor again

martin2345, 2020-06-06:
tail -n 500 /var/log/syslog
Jun  6 17:41:07 raspberrypi bthelper[977]: [#033[0;93mCHG#033[0m] Controller DC:A6:32:5F:31:4A UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
Jun  6 17:41:07 raspberrypi bthelper[977]: [#033[0;93mCHG#033[0m] Controller DC:A6:32:5F:31:4A UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
Jun  6 17:41:07 raspberrypi bthelper[977]: [#033[0;93mCHG#033[0m] Controller DC:A6:32:5F:31:4A Class: 0x00480000
Jun  6 17:41:07 raspberrypi bthelper[977]: [#033[0;93mCHG#033[0m] Controller DC:A6:32:5F:31:4A UUIDs: 00001112-0000-1000-8000-00805f9b34fb
Jun  6 17:41:07 raspberrypi bthelper[977]: [#033[0;93mCHG#033[0m] Controller DC:A6:32:5F:31:4A UUIDs: 00001801-0000-1000-8000-00805f9b34fb
Jun  6 17:41:07 raspberrypi bthelper[977]: [#033[0;93mCHG#033[0m] Controller DC:A6:32:5F:31:4A UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
Jun  6 17:41:07 raspberrypi bthelper[977]: [#033[0;93mCHG#033[0m] Controller DC:A6:32:5F:31:4A UUIDs: 00001800-0000-1000-8000-00805f9b34fb
Jun  6 17:41:07 raspberrypi bthelper[977]: [#033[0;93mCHG#033[0m] Controller DC:A6:32:5F:31:4A UUIDs: 00001200-0000-1000-8000-00805f9b34fb
Jun  6 17:41:07 raspberrypi bthelper[977]: [#033[0;93mCHG#033[0m] Controller DC:A6:32:5F:31:4A UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
Jun  6 17:41:07 raspberrypi bthelper[977]: [#033[0;93mCHG#033[0m] Controller DC:A6:32:5F:31:4A UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
Jun  6 17:41:07 raspberrypi bthelper[977]: [#033[0;93mCHG#033[0m] Controller DC:A6:32:5F:31:4A UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
Jun  6 17:41:07 raspberrypi bthelper[977]: Changing power on succeeded
Jun  6 17:41:07 raspberrypi systemd[1]: bthelper@hci0.service: Succeeded.
Jun  6 17:41:07 raspberrypi dbus-daemon[377]: [system] Successfully activated service 'org.freedesktop.hostname1'
Jun  6 17:41:07 raspberrypi systemd[1]: Started Hostname Service.
Jun  6 17:41:08 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.__main__ Enabled extensions: http, spotify, softwaremixer, m3u, iris, stream, local, file, mpd
Jun  6 17:41:08 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Jun  6 17:41:08 raspberrypi dhcpcd[464]: wlan0: Router Advertisement from fe80::464e:6dff:fe00:e495
Jun  6 17:41:08 raspberrypi dhcpcd[464]: wlan0: adding address 2a02:8071:689:cc00:dea6:32ff:fe5f:3149/64
Jun  6 17:41:08 raspberrypi kernel: [   14.493849] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead
Jun  6 17:41:08 raspberrypi avahi-daemon[421]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::dea6:32ff:fe5f:3149.
Jun  6 17:41:08 raspberrypi avahi-daemon[421]: Joining mDNS multicast group on interface wlan0.IPv6 with address 2a02:8071:689:cc00:dea6:32ff:fe5f:3149.
Jun  6 17:41:08 raspberrypi avahi-daemon[421]: Registering new address record for 2a02:8071:689:cc00:dea6:32ff:fe5f:3149 on wlan0.*.
Jun  6 17:41:08 raspberrypi dhcpcd[464]: wlan0: adding route to 2a02:8071:689:cc00::/64
Jun  6 17:41:08 raspberrypi avahi-daemon[421]: Withdrawing address record for fe80::dea6:32ff:fe5f:3149 on wlan0.
Jun  6 17:41:08 raspberrypi dhcpcd[464]: wlan0: requesting DHCPv6 information
Jun  6 17:41:08 raspberrypi dhcpcd[464]: wlan0: fe80::464e:6dff:fe00:e495 is reachable again
Jun  6 17:41:08 raspberrypi dhcpcd[464]: wlan0: adding default route via fe80::464e:6dff:fe00:e495
Jun  6 17:41:08 raspberrypi dbus-daemon[987]: [session uid=1000 pid=987] Activating via systemd: service name='org.gtk.vfs.Daemon' unit='gvfs-daemon.service' requested by ':1.3' (uid=1000 pid=971 comm="/usr/bin/lxsession -s LXDE-pi -e LXDE ")
Jun  6 17:41:08 raspberrypi systemd[935]: Starting Virtual filesystem service...
Jun  6 17:41:08 raspberrypi dbus-daemon[987]: [session uid=1000 pid=987] Successfully activated service 'org.gtk.vfs.Daemon'
Jun  6 17:41:08 raspberrypi systemd[935]: Started Virtual filesystem service.
Jun  6 17:41:08 raspberrypi kernel: [   15.015919] fuse init (API version 7.27)
Jun  6 17:41:09 raspberrypi systemd[1]: Mounting FUSE Control File System...
Jun  6 17:41:09 raspberrypi systemd[1]: Mounted FUSE Control File System.
Jun  6 17:41:09 raspberrypi mpd[965]: exception: failed to open log file "/var/log/mpd/mpd.log" (config line 40): Permission denied
Jun  6 17:41:10 raspberrypi systemd[935]: mpd.service: Main process exited, code=exited, status=1/FAILURE
Jun  6 17:41:10 raspberrypi systemd[935]: mpd.service: Failed with result 'exit-code'.
Jun  6 17:41:10 raspberrypi systemd[935]: Failed to start Music Player Daemon.
Jun  6 17:41:10 raspberrypi systemd[935]: Reached target Default.
Jun  6 17:41:10 raspberrypi systemd[935]: Startup finished in 3.502s.
Jun  6 17:41:10 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Jun  6 17:41:10 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Jun  6 17:41:10 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Jun  6 17:41:10 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: SpotifyBackend, FileBackend, M3UBackend, StreamBackend, LocalBackend
Jun  6 17:41:10 raspberrypi mopidy[524]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
[... cut ...]
Jun  6 17:46:30 raspberrypi mopidy[2370]: INFO     [MpdSession-83] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:43616
Jun  6 17:46:35 raspberrypi mopidy[2370]: INFO     [MpdSession-84] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:43620
Jun  6 17:46:40 raspberrypi mopidy[2370]: INFO     [MpdSession-85] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:43624
Jun  6 17:46:45 raspberrypi mopidy[2370]: INFO     [MpdSession-86] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:43628

MiczFlor, 2020-06-12:
Can this ticket be closed?

martin2345, 2020-06-12:
Thanks for the Help!

closed 2020-06-12 by martin2345