MiczFlor/RPi-Jukebox-RFID #1001: pirc522 "ModuleNotFound"

deformatik-debug, 2020-06-05:
I would like to use an RFIO-RC522 rfid Reader with phoniebox. After the installation, i have copied the Reader.py.experimental to Reader.py. When I start Reader.py, i get an error ModuleNotFoundError: No module named 'pirc522'. When I start RegisterDevice.py it tells me to import pirc522.
[image]

With my last install of phoniebox 3 months ago, I did the same and when I started RegisterDevice.py I could choose my RC522. Any idea whats wrong here?

Thank you in advance for your efforts.

deformatik-debug, 2020-06-06:
Thank you for your help. I have executed the setup_rc522.sh script. Based on the readme within the RC522 folder the script should setup everything, but after a reboot the rc522 inside phoniebox still doesnt work. If I run RegisterDevice.py again I get a new error. 
[image]

I have done the RC522 tutorial from "pimpmylife" in order to check if the rc522 works, and it works fine. It seems that something with the phoniebox settings is not correct.

Any help would be apreciated.

deformatik-debug, 2020-06-06:
I got it to work. I had to comment out line 17 in the Reader.py script. On line 17 the script tries to import libs
#from py532lib.i2c import *
#from py532lib.mifare import *
that generated the error message.

Not sure if this is the desired behaviour.
Thanks anyway for pointing me to the setup_rc522.sh script, but it doesnt seem to work as expected.

s-martin, 2020-06-06:
Yeah sorry, you need the script AND a change which is already on `develop`, but not yet on `master` branch.

Please see
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/6d52f2741b9e2a4c8daad06500684bce1c07f94c/scripts/Reader.py.experimental#L17-L22

closed 2020-06-06 by deformatik-debug