MiczFlor/RPi-Jukebox-RFID #999: 🐛 | WEB-Ui Volume control failing since new version

Piehti, 2020-06-05:
## Bug

### What I did

Volume up + down not working + SET volume neighter. Actually no interacting with volume on the webinterface works. Didnt test RFiD-card-changes on volume yet. Might (pretty sure) it's the same error as #973

### What happened

After installing the new version on a (almost) clean Pi3b+ the Sound-Control stopped working. Might have to do with that "audiojack" thingy going on and that MIGHT have to do with rpi-update? Dunno.

s-martin, 2020-06-05:
I think that’s the same issue than #973, because the web interface uses the same underlying mechanism.

Piehti, 2020-06-05:
*nods*

s-martin, 2020-06-05:
Closing as duplicate, please follow up in #973

closed 2020-06-05 by s-martin