MiczFlor/RPi-Jukebox-RFID #1005: PIFI DAC+ not working with Gpio buttons enabled

mettyw, 2020-06-09:
## Bug

It seems to me that the PiFi DAC+ does not work with gpio buttons support.

### What I did

I installed the raspberry pi with above mentioned buster image and ran the installer script. I  have a PiFi DAC+ and followed the hifiberry instructions.

### What happened

If I enable the gpio button system service there is no sound. If I disable it and reboot, there is sound.

### I expected this to happen

Sound with gpio enabled.

### Further information that might help

If I change shutGPIO to something else than 3, it works.

DAC+ is using GPIO 3 according to https://pinout.xyz/pinout/pi_dac# 

I am new to this Raspi thing, but I suspect that the DAC+ doesn't like the pull-up of GPIO3.

### Base image and version

10 buster

### Branch / Release

2.0 - ed888da - master

s-martin, 2020-06-09:
That's not a bug, but a configuration issue.

If you don't use GPIO buttons at all, just deactivate phoniebox-gpio-buttons.service.

If you do use GPIO buttons, you have to make sure the pins used in gpio-buttons.py don't interfere with the pins used by your sound card.

So the solution could be to use a different pin than 3 for shutdown button. If you don't use a shutdown button, you could just comment out the lines for the shutdown button.

mettyw, 2020-06-09:
Hm, partly disagree: it's a bug in the documentation at least IMHO.

The hifiberry guide only warns about conflicts with other pins.

And https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hardware-Pinout-Overview#flexible-pin-out in the table has a comment that sounds to me like it is supposed to work together.

So I cannot use a shutdown button in combination with the DAC+ at all?

s-martin, 2020-06-11:
> Hm, partly disagree: it's a bug in the documentation at least IMHO.

Point taken :)

> The hifiberry guide only warns about conflicts with other pins.
> 
> And https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hardware-Pinout-Overview#flexible-pin-out in the table has a comment that sounds to me like it is supposed to work together.
> 
> As a consequence this actually means that I cannot have a soundcard and a single button that simultaneously handles wakeup and shutdown on a raspberry pi (because wakeup exclusively is working on pin 3), is this understanding correct?

Is it pin 3 or GPIO 3 (which should be pin 5, IIRC)?

If your soundcard also needs the same pin, this may not work together with shutdown button, but I'm not entirely sure about the requirements of the shutdown button.

Could you post the needed pins of the soundcard?

mettyw, 2020-06-19:
Yeah it is GPIO3, sorry for that. I am using the PiFi DAC+. The pins used are listed here https://pinout.xyz/pinout/pi_dac# .

However, after some research I found that using GPIO3 and I2C devices is possible after all. I followed the suggestions from that Stackexchange cment here https://raspberrypi.stackexchange.com/a/85316 . So I configured another free GPIO pin for the Phoniebox shutdown and connected the GPIO3 pin and the other pin with a diode (which only lets current flow from GPIO3 to the other pin). The shutdown pin is connected to a normally open pushbutton. This way, I can turn the Raspi on and off with one single button.

I do not need more support. But maybe the info from above can be useful for others. It seems like after all it is actually possible to use a soundcard and a single on/off button using gpio.

jo-me, 2020-06-23:
I ran into the same when setting up with the MiniAmp.
I could not understand why the default GPIO button assigment is not based on the "flexible pinout" assignment if that causes way less conflicts.

s-martin, 2020-06-23:
> I could not understand why the default GPIO button assigment is not based on the "flexible pinout" assignment if that causes way less conflicts.

Probably the only reason is that nobody wrote a PR ;)

closed 2020-12-18 by s-martin