MiczFlor/RPi-Jukebox-RFID #1006: 🐛 | LCD service not working

domu83, 2020-06-10:
## Bug

### What I did

I installed on my buster installation with 2.0 rc7 the LCD components and run throught he installation instructions. However, service was not able to start up.

### What happened
When I run the i2c_lcd.py manually I get the following

`
pi@raspberrypi:~ $ python3 /home/pi/RPi-Jukebox-RFID/components/displays/HD44780-i2c/i2c_lcd.py
  File "/home/pi/RPi-Jukebox-RFID/components/displays/HD44780-i2c/i2c_lcd.py", line 327
SyntaxError: (unicode error) 'utf-8' codec can't decode byte 0xe4 in position 0: unexpected end of data
`

domu83, 2020-07-02:
any ideas anyone?

mriffault, 2020-07-20:
Don't know if it may help, but I had to install (`apt get install`) : `python3-smbus` and `python3-mpd` to make it work on my RPi3 with the lastest **lite** Raspbian release.

s-martin, 2020-10-20:
Do you still experience this issue?

domu83, 2020-10-20:
I've not updated to 2.1 so far, as all of the rest or lets better say spotify is working stable in this config. would you recommend to update to fix this topic?

s-martin, 2020-10-21:
I don't think that this issue has been addressed in 2.1.

s-martin, 2021-02-28:
@domu83 still an issue?

s-martin, 2022-12-24:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/displays/HD44780-i2c/README.md

describes already necessary dependencies.

closing for now as there was no activity for a long time.

please feel free to reopen a new issue, if necessary.

closed 2022-12-24 by s-martin