MiczFlor/RPi-Jukebox-RFID #1008: RFID Question

hawyee125, 2020-06-10:
Hello! I am new to coding and doing Raspberry Pi projects, so this could probably just be my ignorance on my end. I have a question regarding my RFID cards being scanned on the Phoniebox web application. When I follow the instructions to add a new RFID card on the web app, it asks me to scan a new card, and the ID code should pop up and work for "Last Chip Used" The writing at the bottom shows "This will automatically update as you swipe a RFID card." So I assume when I scan my card to the reader, that it should update on the web app, but it doesn't do anything. I hear the beep when the card gets scanned, and the ID shows up on my terminal, but it doesn't do anything. I don't know if I am missing something or if I am supposed to scan the RFID to a particular place to play my playlist or album? I scan it and hear a beep, but it doesn't have any input, just simply the ID number of the card I used, and I don't know how to make it so when I scan it actually plays a song that I want it to on the web app. I don't know if I am missing anything to make this work. After I installed Phoniebox on the Wiki page https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch I then installed samba with 

sudo apt install samba samba-common smbclient

I could possibly be missing something I didn't install and that is why my RFID cards are not scanning on the web application? 

I have this RFID Card Reader: https://www.amazon.com/gp/product/B07TMNZPXK/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1
And these are my RFID cards https://www.amazon.com/gp/product/B01M1L7NHI/ref=ppx_yo_dt_b_asin_title_o02_s01?ie=UTF8&psc=1

s-martin, 2020-06-11:
Could you post the output of
 `tail -n 500 /var/log/syslog`
and
`cat /home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt`?

Thanks

hawyee125, 2020-06-11:
Here is the output for `tail -n 500 /var/log/syslog`

Jun 11 00:41:27 raspberrypi kernel: [    0.707299] mmc-bcm2835 3f300000.mmcnr: could not get clk, deferring probe
Jun 11 00:41:27 raspberrypi kernel: [    0.707707] sdhost-bcm2835 3f202000.mmc: could not get clk, deferring probe
Jun 11 00:41:27 raspberrypi kernel: [    0.707833] sdhci-pltfm: SDHCI platform and OF driver helper
Jun 11 00:41:27 raspberrypi kernel: [    0.709347] ledtrig-cpu: registered to indicate activity on CPUs
Jun 11 00:41:27 raspberrypi kernel: [    0.709537] hidraw: raw HID events driver (C) Jiri Kosina
Jun 11 00:41:27 raspberrypi kernel: [    0.709712] usbcore: registered new interface driver usbhid
Jun 11 00:41:27 raspberrypi kernel: [    0.709717] usbhid: USB HID core driver
Jun 11 00:41:27 raspberrypi kernel: [    0.710654] vchiq: vchiq_init_state: slot_zero = (ptrval), is_master = 0
Jun 11 00:41:27 raspberrypi kernel: [    0.712123] [vc_sm_connected_init]: start
Jun 11 00:41:27 raspberrypi kernel: [    0.720700] [vc_sm_connected_init]: end - returning 0
Jun 11 00:41:27 raspberrypi kernel: [    0.721799] Initializing XFRM netlink socket
Jun 11 00:41:27 raspberrypi kernel: [    0.721823] NET: Registered protocol family 17
Jun 11 00:41:27 raspberrypi kernel: [    0.721952] Key type dns_resolver registered
Jun 11 00:41:27 raspberrypi kernel: [    0.722695] Registering SWP/SWPB emulation handler
Jun 11 00:41:27 raspberrypi kernel: [    0.723338] registered taskstats version 1
Jun 11 00:41:27 raspberrypi kernel: [    0.723355] Loading compiled-in X.509 certificates
Jun 11 00:41:27 raspberrypi kernel: [    0.730141] uart-pl011 3f201000.serial: cts_event_workaround enabled
Jun 11 00:41:27 raspberrypi kernel: [    0.730219] 3f201000.serial: ttyAMA0 at MMIO 0x3f201000 (irq = 81, base_baud = 0) is a PL011 rev2
Jun 11 00:41:27 raspberrypi kernel: [    0.732039] mmc-bcm2835 3f300000.mmcnr: mmc_debug:0 mmc_debug2:0
Jun 11 00:41:27 raspberrypi kernel: [    0.732048] mmc-bcm2835 3f300000.mmcnr: DMA channel allocated
Jun 11 00:41:27 raspberrypi kernel: [    0.757919] sdhost: log_buf @ (ptrval) (fad07000)
Jun 11 00:41:27 raspberrypi kernel: [    0.794435] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
Jun 11 00:41:27 raspberrypi kernel: [    0.796019] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
Jun 11 00:41:27 raspberrypi kernel: [    0.797600] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
Jun 11 00:41:27 raspberrypi kernel: [    0.800404] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
Jun 11 00:41:27 raspberrypi kernel: [    0.805806] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
Jun 11 00:41:27 raspberrypi kernel: [    0.807106] of_cfs_init
Jun 11 00:41:27 raspberrypi kernel: [    0.807296] of_cfs_init: OK
Jun 11 00:41:27 raspberrypi kernel: [    0.807916] Waiting for root device PARTUUID=3f07ff44-02...
Jun 11 00:41:27 raspberrypi kernel: [    0.871824] random: fast init done
Jun 11 00:41:27 raspberrypi kernel: [    0.895223] mmc0: host does not support reading read-only switch, assuming write-enable
Jun 11 00:41:27 raspberrypi kernel: [    0.898475] mmc0: new high speed SDHC card at address aaaa
Jun 11 00:41:27 raspberrypi kernel: [    0.899783] mmcblk0: mmc0:aaaa SL32G 29.7 GiB
Jun 11 00:41:27 raspberrypi kernel: [    0.903094]  mmcblk0: p1 p2
Jun 11 00:41:27 raspberrypi kernel: [    0.914848] mmc1: new high speed SDIO card at address 0001
Jun 11 00:41:27 raspberrypi kernel: [    0.927284] Indeed it is in host mode hprt0 = 00021501
Jun 11 00:41:27 raspberrypi kernel: [    0.933021] EXT4-fs (mmcblk0p2): INFO: recovery required on readonly filesystem
Jun 11 00:41:27 raspberrypi kernel: [    0.933029] EXT4-fs (mmcblk0p2): write access will be enabled during recovery
Jun 11 00:41:27 raspberrypi kernel: [    1.137200] usb 1-1: new high-speed USB device number 2 using dwc_otg
Jun 11 00:41:27 raspberrypi kernel: [    1.137327] Indeed it is in host mode hprt0 = 00001101
Jun 11 00:41:27 raspberrypi kernel: [    1.149500] EXT4-fs (mmcblk0p2): orphan cleanup on readonly fs
Jun 11 00:41:27 raspberrypi kernel: [    1.149655] EXT4-fs (mmcblk0p2): 1 orphan inode deleted
Jun 11 00:41:27 raspberrypi kernel: [    1.149662] EXT4-fs (mmcblk0p2): recovery complete
Jun 11 00:41:27 raspberrypi kernel: [    1.163049] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
Jun 11 00:41:27 raspberrypi kernel: [    1.163116] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
Jun 11 00:41:27 raspberrypi kernel: [    1.164105] devtmpfs: mounted
Jun 11 00:41:27 raspberrypi kernel: [    1.199136] Freeing unused kernel memory: 1024K
Jun 11 00:41:27 raspberrypi kernel: [    1.227576] Run /sbin/init as init process
Jun 11 00:41:27 raspberrypi kernel: [    1.377493] usb 1-1: New USB device found, idVendor=0424, idProduct=9514, bcdDevice= 2.00
Jun 11 00:41:27 raspberrypi kernel: [    1.377508] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Jun 11 00:41:27 raspberrypi kernel: [    1.378138] hub 1-1:1.0: USB hub found
Jun 11 00:41:27 raspberrypi kernel: [    1.378232] hub 1-1:1.0: 5 ports detected
Jun 11 00:41:27 raspberrypi kernel: [    1.697210] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
Jun 11 00:41:27 raspberrypi kernel: [    1.827516] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00, bcdDevice= 2.00
Jun 11 00:41:27 raspberrypi kernel: [    1.827530] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[... cut ...]
Jun 11 09:39:31 raspberrypi mopidy[515]: INFO     [MpdSession-54] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58662
Jun 11 09:39:37 raspberrypi mopidy[515]: INFO     [MpdSession-55] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58666
Jun 11 09:39:37 raspberrypi systemd[1]: Starting Clean php session files...
Jun 11 09:39:38 raspberrypi systemd[1]: phpsessionclean.service: Succeeded.
Jun 11 09:39:38 raspberrypi systemd[1]: Started Clean php session files.

and the output for `cat /home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt`

HID 413d:2107

MiczFlor, 2020-06-12:
Hallo @hawyee125 
schau mal in dieses Ticket: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/551
Vielleicht hilft das?

MiczFlor, 2020-06-13:
Hi @hawyee125 
sorry, that I posted in German :) but plese have a look at this ticket:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/551
In there (if you use e.g. google translate), you will see that your reader might need the kmoon version of the Reader.py script

closed 2020-12-15 by s-martin