MiczFlor/RPi-Jukebox-RFID #1011: 🐛 | Spotify playlists not playing

szaiser, 2020-06-12:
## Bug
While an RFID card registered to play stream URL `spotify:track:##################` or `spotify:album:##############` works like a charm, `spotify:user:<username>:###############`does not work. Even the sample url in the doc [SPOTIFY-INTEGRATION.md#how-to-use-spotify](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/5b2cf742573489db27b117bf9588f2a54d750365/docs/SPOTIFY-INTEGRATION.md#how-to-use-spotify) does not work. This could be connected to [Mopidy-spotiffy issue 182](https://github.com/mopidy/mopidy-spotify/issues/182).

### What I did

I installed via 
```
cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh 
```
Then I swept an RFID card and registered it by giving the stream URL multiple playlist stream urls for testing. One of them `spotify:user:spotify:playlist:37i9dQZF1DWUW2bvSkjcJ6`, which as far as I can tell is a valid spotify url. For each URL which I tried, I selected type Spotify, created a new folder and I swept the RFID card after registering.

### What happened
Nothing.

### I expected this to happen
I expect the first track of the spotify playlist to play, after I sweep the card.

### Further information that might help
pi@raspberrypi:~/phoniebox_logs $ journalctl -u mopidy
-- Logs begin at Fri 2020-06-12 00:17:01 CEST, end at Fri 2020-06-12 22:26:00 CEST. --
Jun 12 00:17:06 raspberrypi systemd[1]: Starting Mopidy music server...
Jun 12 00:17:07 raspberrypi systemd[1]: Started Mopidy music server.
Jun 12 00:17:12 raspberrypi mopidy[487]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Jun 12 00:17:12 raspberrypi mopidy[487]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Jun 12 00:17:12 raspberrypi mopidy[487]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Jun 12 00:17:12 raspberrypi mopidy[487]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Jun 12 00:17:12 raspberrypi mopidy[487]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Jun 12 00:17:12 raspberrypi mopidy[487]: INFO     [MainThread] mopidy.config Loading config from command line options
Jun 12 00:17:12 raspberrypi mopidy[487]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Jun 12 00:17:13 raspberrypi mopidy[487]: INFO     [MainThread] mopidy.__main__ Enabled extensions: file, stream, iris, spotify, local, mpd, http, m3u, softwaremixer
Jun 12 00:17:13 raspberrypi mopidy[487]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Jun 12 00:17:16 raspberrypi mopidy[487]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Jun 12 00:17:16 raspberrypi mopidy[487]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Jun 12 00:17:16 raspberrypi mopidy[487]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Jun 12 00:17:16 raspberrypi mopidy[487]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Jun 12 00:17:16 raspberrypi mopidy[487]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Jun 12 00:17:17 raspberrypi mopidy[487]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Jun 12 00:17:17 raspberrypi mopidy[487]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Authorization failed, not attempting Spotify API request. Please get new credentials from https://www.mopidy.com/authenticate and/or restart Mopidy to resolve this problem.
Jun 12 00:17:17 raspberrypi mopidy[487]: ERROR    [SpotifyBackend-6] mopidy_spotify.web OAuth token refresh failed: invalid_client Client not known.
Jun 12 00:17:17 raspberrypi mopidy[487]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Failed to load Spotify user profile
Jun 12 00:17:17 raspberrypi mopidy[487]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Jun 12 00:17:17 raspberrypi mopidy[487]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: IrisFrontend, HttpFrontend, MpdFrontend
Jun 12 00:17:17 raspberrypi mopidy[487]: INFO     [IrisFrontend-12] mopidy_iris.core Starting Iris 3.49.0
Jun 12 00:17:17 raspberrypi mopidy[487]: INFO     [HttpFrontend-14] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Jun 12 00:17:17 raspberrypi mopidy[487]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Jun 12 00:17:17 raspberrypi mopidy[487]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Jun 12 00:17:20 raspberrypi mopidy[487]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56546
Jun 12 00:17:21 raspberrypi mopidy[487]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56550
Jun 12 00:17:21 raspberrypi mopidy[487]: INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56554
Jun 12 00:17:21 raspberrypi mopidy[487]: INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56558
Jun 12 00:17:26 raspberrypi mopidy[487]: INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56562
Jun 12 00:17:26 raspberrypi mopidy[487]: INFO     [MpdSession-21] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56566
Jun 12 00:17:31 raspberrypi mopidy[487]: INFO     [MpdSession-22] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56570
Jun 12 00:17:31 raspberrypi mopidy[487]: INFO     [MpdSession-23] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56574
Jun 12 21:46:21 raspberrypi mopidy[487]: INFO     [MpdSession-24] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56578
Jun 12 21:46:21 raspberrypi mopidy[487]: INFO     [MpdSession-25] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56582
Jun 12 21:46:26 raspberrypi mopidy[487]: INFO     [MpdSession-26] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56586
Jun 12 21:46:26 raspberrypi mopidy[487]: INFO     [MpdSession-27] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56590
Jun 12 21:46:35 raspberrypi mopidy[487]: INFO     [MpdSession-28] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56594
Jun 12 21:46:35 raspberrypi mopidy[487]: INFO     [MpdSession-29] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56598
Jun 12 21:46:40 raspberrypi mopidy[487]: INFO     [MpdSession-30] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56606
Jun 12 21:46:40 raspberrypi mopidy[487]: INFO     [MpdSession-31] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56610
Jun 12 21:46:46 raspberrypi mopidy[487]: INFO     [MpdSession-32] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56614
Jun 12 21:46:46 raspberrypi mopidy[487]: INFO     [MpdSession-33] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56618
Jun 12 21:46:51 raspberrypi mopidy[487]: INFO     [MpdSession-34] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56628
Jun 12 21:46:51 raspberrypi mopidy[487]: INFO     [MpdSession-35] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56626
Jun 12 21:46:56 raspberrypi mopidy[487]: INFO     [MpdSession-36] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56634
Jun 12 21:46:56 raspberrypi mopidy[487]: INFO     [MpdSession-37] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56638
Jun 12 21:47:01 raspberrypi mopidy[487]: INFO     [MpdSession-38] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56642
Jun 12 21:47:01 raspberrypi mopidy[487]: INFO     [MpdSession-39] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:56646
[... 362 lines cut ...]
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release
master
### Installscript

## Hardware

### RaspberryPi version
Raspberry Pi 3 Model B Rev 1.2
### RFID Reader
KKMoon HID 413d:2107
--> had to follow https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/551
### Soundcard
idVendor           0x0d8c C-Media Electronics, Inc.
idProduct          0x013c CM108 Audio Controller

### Other notable hardware
OLED display, installed SW therefore is:
```
cd; rm o4p_installer.sh; wget https://raw.githubusercontent.com/splitti/oled_phoniebox/master/scripts/install/o4p_installer.sh; chmod +x o4p_installer.sh; ./o4p_installer.sh 
```

s-martin, 2020-06-14:
Are you sure the URI is correct? Because you have the following error: `04 raspberrypi mopidy[487]:     raise ValueError(f"Could not parse {repr(uri)} as a Spotify URI")
Jun 12 21:47:04 raspberrypi mopidy[487]: ValueError: Could not parse 'spotify:user:spotify:playlist:37i9dQZF1DWUVpAXiEPK8P' as a Spotify URI`

szaiser, 2020-06-16:
1. Actually, I am sure that `spotify:user:spotify:playlist:37i9dQZF1DWUVpAXiEPK8P` is a valid URI.  When I copy it into the search field of `https://open.spotify.com/search`desktop application, the playlist is found and correctly opened.  Spotify is also satisfied with `spotify:playlist:37i9dQZF1DWUVpAXiEPK8P`, which points to the same playlist. 
Btw: The second spotify in the URI is the username of the playlist owner (who is "spotify").
2. Is there anybody out there, who successfully uses a current RPi-Jukebox version with spotify playlists in the way I am trying to? To me this seems quite hopeless at this stage, and it would be great to learn otherwise.

s-martin, 2020-06-20:
Did you try the shorter version of the URI, if Spotify is also satisfied with that format?

szaiser, 2020-06-20:
Yes, I tried every URI format which I could come up with. Whenever it referred to a Spotify playlist, then Mopidy complains, that it can not find the playlist.

s-martin, 2020-06-21:
@kingosticks, sorry to bother you again, but maybe you have some info, what mopidy requires here?

kingosticks, 2020-06-21:
That form is old and Spotify are supposed to be moving towards the 'spotify:playlist:blahblahblah' format so I'm a bit surprised they are still giving out the old format. That being said, the old format should still work and the code still handles it. The (updated?) error log in the original post shows the first error is the web API authorization failed.  You must fix that first.

szaiser, 2020-06-23:
Thanks everybody for the help, in particular kingosticks. In hindsight it seems obvious, but the issue indeed was the wep API authorization error. I simply used the wrong config file (I used ~/.config/mopidy/mopidy.conf, see https://docs.mopidy.com/en/latest/running/service/#configuration). 

The reason why I felt safe to ignore this error is, that both albums and tracks played correctly and hence I assumed, this is just wrong startup behavior. I still do not understand, what in this respect is the difference to playlists.

closed 2020-06-23 by szaiser