MiczFlor/RPi-Jukebox-RFID #1013: 🐛 | RFID Card can be assigned to several system commands

jo-me, 2020-06-15:
## Bug

### What I did

Assign a RFID card ID to 2 system commands via the web UI

### What happened

The card ID is assigned to setVolume 75% and setMaxVolume 75% because I chose the wrong command at first.

Swiping the card will only change the maxVolume, so I assume that executing several commands with one swipe is not really working and should therefore be prohibited (at least when it is configured in the UI).

[image]

### I expected this to happen

The ID should've been unassigned from the old system command and assigned to the new system command.

### Further information that might help

## Software

### Base image and version

Raspberry OS

### Branch / Release

2.0 - ed888da - master

### Installscript

buster classic

## Hardware

### RaspberryPi version

2

### RFID Reader

RC522 

### Soundcard

Miniamp

MiczFlor, 2020-06-18:
I @jo-me 
I knew that was the case :) the web UI was a quick hack on my side to allow assigning system stuff to cards at all. Beforehand it had to be done with the editor. But you are right, that should be the way it's working. And I changed it to `enhancement`. Bug is such an ugly word.

skipidar, 2021-01-30:
Hi Micz - I have just set up the Phoniebox project for the kidz and have faced the same behaviour.
Multiple commands are assigned to the same card.
Also something like "unassign card" or "unregister card" from everywhere would be useful I guess.

I would like to enhance the UI a bit - can you point me to the right part of the project?
Where should I look for the relevant code?

Is there an easy way to deploy the modified code to the PI, to test it?
Would copy/paste to the ~/RPi-Jukebox-RFID do it?

s-martin, 2024-05-10:
#2317 tracks the implementation of the desired feature in V3.

We will not implement this for V2, please follow up there.

AlvinSchiller, 2024-05-13:
@s-martin isn't this rather a bug than an enhancement? 
Two Commands on the same id shouldn't be possible, as only the first matched command will be executed. But which one is not visible to the user, as the order in the ui and the backend differs.

s-martin, 2024-05-13:
You are right, didn't read it correctly.

I'm reopening the issue