MiczFlor/RPi-Jukebox-RFID #1014: 🚀 | Confirmation sounds for system commands

jo-me, 2020-06-16:
## Feature Description

### What functionality would you like to see in your phoniebox?

Whenever a system command is executed after swiping a card the system should play a confirmation sound so that it is clear to the user that the swipe action was successful.

In addition to that there could be another (negative) sound when a card is swiped that has no action associated with it (neither music nor system command).

### How do you envision the feature to work from a users perspective?

User gets feedback (positive or negative) for every swiped card so that it is clear that the card was recognized.

### Further information that might help

n/a

s-martin, 2020-06-16:
The Neuftech USB reader has a built-in confirmation sound, but many disable it.

jo-me, 2020-06-17:
I know, but that sound is not an indication whether something was actually executed, just that something was scanned.

MiczFlor, 2020-06-18:
Hi @jo-me 
are you sure? for example using cards to increase decrease volume would beep each time.
What's the benefit?

jo-me, 2020-06-18:
You are right about volume except when there's nothing playing.
In other cases it would be nice, e.g. toggle wifi.

There is a GPIO for this purpose when something is scanned (e.g. to flash a LED), right?

Huseriato, 2020-10-14:
I also like to hear an RFID sound for actions. May be there can be an option for every registered id card (on the panel, where you can register cards). So it's up to the user, if he configures a beep or not. May be there can be a dropdown of some sounds: e.g. "No sound", "Beep", "Double Beep", etc... If I want to create something like this, where would be a good starting point to play the "beep"? Is there any central routine?

MiczFlor, 2020-10-15:
Hi @Huseriato 

One potential problem up front: the soundfile might be too short to be heard when played by a local audio player. I explain: the startup sound for example uses the mpg123 player to play a sound. I ended up adding two seconds of silence at the beginning of the audio file, because by the time the system was ready to play what mpg123 played, the file was over. You can also hear that problem with mpd when playing some of the test samples. So this is experience which might help you not to go crazy if you dive into this task - and all you hear is silence :)

the script which is called each time a RFID chip is swiped is this one:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/scripts/rfid_trigger_play.sh
Starting in line [82](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/7238b9d09cd35096bff5abaeeeb3595645fdad1c/scripts/rfid_trigger_play.sh#L82) the script checks if the RFID is used for a system command. If it finds one, it executes a command, calling another script. 
Starting in line [276](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/7238b9d09cd35096bff5abaeeeb3595645fdad1c/scripts/rfid_trigger_play.sh#L276) all available commands have been checked and not the Phoniebox checks if there is a shortcut set pointing to a folder and then attempts to play that folder content.

So what would be a good place to start...

In my opinion, if you want different sounds for different commands, the script to work with is 
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/scripts/playout_controls.sh
because this is where the different actions take place. 

How would I go about this:

* In line [104](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/7238b9d09cd35096bff5abaeeeb3595645fdad1c/scripts/playout_controls.sh#L104) the actions for each command are checked for and executed. The most obvious place to add the sound thing would be there. BUT...
* if you add it to this script so deep in the code, maintenance would be very difficult, because 
    * each time this script would need to changed if new system sounds would come up
    * if new features are added, each time they checks for playing sounds or not would also be needed to go in there

Instead, what I suggest is:

* Place all your logic in a file `rfid_play_sound.sh` inside the scripts folder
* Create a global settings varialbe like "RFIDPLAYSYSSOUND" with the option TRUE or FALSE or the like.
* In line [76](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/7238b9d09cd35096bff5abaeeeb3595645fdad1c/scripts/playout_controls.sh#L76) add something like the following:
~~~
if [ "${RFIDPLAYSYSSOUND}" == "TRUE" ]; then 
. ${PATHDATA}/rfid_play_syssound.sh
fi
~~~

So if the global var is set to `TRUE` it will be called. 

Inside this file, you can do what you want :) I recommend to use the `case` routine starting in line [104](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/7238b9d09cd35096bff5abaeeeb3595645fdad1c/scripts/playout_controls.sh#L104).
If new features are added to the playout script, your script will be out of date, but if somebody wants to add a new sound for a new feature they can. Everybody can. Without having to alter the playout script.

Does this makes sense?

All the best, micz

Huseriato, 2020-10-15:
Thank you for the good explanation and the perfect starting point. I think the best way to play a short file is to use another file format. However, I just tried out the sox library with a real short beep mp3:

```
root@kidsmusic:/home/pi/soundtest# play beep.mp3
play WARN alsa: can't encode 0-bit Unknown or not applicable

beep.mp3:

 File Size: 16.9k     Bit Rate: 192k
  Encoding: MPEG audio
  Channels: 2 @ 16-bit
Samplerate: 44100Hz
Replaygain: off
  Duration: 00:00:00.71

In:96.3% 00:00:00.68 [00:00:00.03] Out:30.0k [      |      ] Hd:0.2 Clip:0
Done.
root@kidsmusic:/home/pi/soundtest#
```

This played instantly. What sound file did you use to test and which player? Or did to play it via mopidy? I'm not sure about the used audiodevice when using sox. It would be nice to have just one configuration for the audio setup.

Huseriato, 2020-10-16:
Adding the line `mpg123 $PATHDATA/../sounds/beep.mp3` (I downloaded this soundfile from a random website with some soundfiles) in  `:/scripts/playout_controls.sh` line 79 (directly after `echo "$CARDID" > $PATHDATA/../settings/Latest_RFID`) works fine as a quick test. The sound is instantly played. But it tooks 1 to 5 seconds before the track is played or the system command is executed. 

May be this beep debugger will help me finding the problem of https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1087 - I also found out, that sometimes my card is not recognized on the first try as described here: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1083

Huseriato, 2020-10-21:
So I tested this in my fork and it did not play well. But the problem is not playing the sound files. I had a perfect detection for system command and play album. I also checked, if the command worked. And there is the problem: You can have a beep, when the card is recognized, but the time it tooks until it's checked is to long. So the "okay beep" and the playback are starting at the same time. Before executing the command you don't know if it will run. So there might be a beep, but no beep that is indicating success or error. Currently success is already indicated by action/play start. And an error is indicated by doing nothing.

So I think this feature is not working very well. I reverted my personal changes I made to test.