MiczFlor/RPi-Jukebox-RFID #1015: Shutdown button doesn't work

thisisstevi, 2020-06-16:
Hey Guys,

I've got a little problem. Everything works fine for me except the shutdown button. If i press the button for 2 or more secondes nothing happens. If the system is turned off and i press the button, it starts fine. A linked RFID Tag with the shutdown command works fine. Only the Button for shutdown doesnt work.

It's an RPi4 B with 1GB Ram with the latest Raspberry Pi OS (32-bit) Lite from 2020-05-27. I've installed the software with the "One line install command" (without Spotify integration).

I've tried already a complete reinstall of the OS and the Jukebox software, but the probleme is still there. Maybe someone can help me with my issue.

Thanks

thisisstevi, 2020-06-18:
I think i found the problem. I was checking the hardware again today when I suddenly had the USB-C connector in my hand!

I orderd a new one. It will arrive on saturday. I'll let you know when the problem with the new device is gone.

MiczFlor, 2020-06-18:
Hm... why would that be the issue?

thisisstevi, 2020-06-18:
Maybe i'm holding the  USB-C conecctor in my hand?

Have a look yourself:
https://ibb.co/s2sQW72

But you're also right. I have reinstalled my other phoniebox in the last hour and now i have the same problem here.

Did you have any idea for this problem?

s-martin, 2020-06-20:
> Maybe i'm holding the USB-C conecctor in my hand?
> 
> Have a look yourself:
> https://ibb.co/s2sQW72

Yes, looks like USB-C

> 
> But you're also right. I have reinstalled my other phoniebox in the last hour and now i have the same problem here.
> 
> Did you have any idea for this problem?

Could you post a photo or maybe a drawing of your wiring and the content of `gpio-buttons.py`?

tobiasnix, 2020-06-22:
Same here. After the upgrade from 1.1.9 to 2.0 the shutdown button stopped working. If the system is turned off and i press the button, it starts fine. Here is wiring parts photo: https://ibb.co/cY2nwgQ

thisisstevi, 2020-06-22:
@s-martin 

I used in all of my Phonieboxes the classic Layout for the wiring and i had never a problem with it until now.

I've build now the forth Phoniebox.

Now after the upgrade to the latest software version of the Phoniebox no one of them can't be shutdown 

Here's the wiring:
    Volume Down GPIO19 (PIN35) and GND (PIN39)
    Volume Up GPIO16 (PIN36) and GND (PIN34)
    Play/Pause/Halt (or how you call it) GPIO21 (PIN40) and GND (PIN25)
    Next GPIO26 (PIN37) and GND (PIN30)
    Previous GPIO20 (PIN38) and GND (PIN20)
    Shutdown (you need to hold button for 2 secs for shutdown) GPIO3 (PIN5) and GND (PIN6)

hlgr, 2020-06-25:
Same here, using the new button layout (but shutdown is the same on Pin3), all the buttons worked fine in 1.19, all other buttons work fine in 2.0, except it won't shutdown. Shutdown from webapp is fine, Startup using the power button is fine, only shutdown won't work. 

I've looked at the scripts, but unfortunately I'm not good enough to find the mistake :(

tobiasnix, 2020-07-01:
Can anyone give a hint where to start debugging?

thoto77, 2020-07-09:
Same problem here. Web GUI and SSH buggy as well since upgrade to 2.0...

s-martin, 2020-07-09:
> Same problem here. Web GUI and SSH buggy as well since upgrade to 2.0...

Could you give some details what's buggy?

Unclear how the 2.0 upgrade would impact SSH.

ElectronicTom, 2020-10-28:
I had the same problem with my box (RPi Zero, software version 2.0). All buttons worked fine, but the shutdown button didn't work.
The buttons have a 1k resistor in series to GND. Measuring the voltage while pressing a button showed nearly 0V at the GPIO-pins except for the shutdown. The voltage remained above 1V when pressing the button. So i removed the resistor and connected direct to GND. Now the shutdown-button woks fine.

MiczFlor, 2020-10-29:
Hi @ElectronicTom 
if this is an issue that many Phonieboxers could run into, could you add it to the wiki under trouble shooting?

MiczFlor, 2020-10-29:
@ElectronicTom 
I quickly did it: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#shutdown-button-gpio-doesnt-work
Apoligies if I made you start writing it up. 
Thanks again for the contribution

s-martin, 2020-11-21:
I close this as it seems to be fixed. Feel free to reopen or open a new issue, if you still experience pr.

closed 2020-11-21 by s-martin

antiwarhol, 2021-09-13:
I had the same issue, but I was not using a resistor between the GPIO pins and the Button in the first place. I wired everything up as described [here](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-buttons), every button worked but the Shutdown button would only wake the Pi up, not shutting it down.
 
Let me document this for other lost souls. 
The solution in my case was **painfully simple**:

In the **gpio_settings.ini** file, for some reason the shut down button was the only one that was **not activated** after the initial install.

Here is the part from the gpio_settings.ini
Setting them to "True" (like in "after") fixed it for me.

Before:
[Shutdown]
enabled : False
type : Button
pin : 3
hold_time : 2
functioncall : functionCallShutdown
pull_up_down : pull_up

After:
[Shutdown]
enabled : True
type : Button
pin : 3
hold_time : 2
functioncall : functionCallShutdown
pull_up_down : pull_up

Newbies: Just open the file ~/RPi-Jukebox-RFID/settings/gpio_settings.ini with e.g. a right click in the file manager, make backup copy and then edit the value mentioned above from "False" to "True". 
Save it, restart the Raspberry Pi and the button worked in my case.

Why it was not set to "True" like the other buttons is a mystery to me, could not find any post in the web that mentioned this as a necessary step after the install process.

**My setup:**
Raspberry Pi 4 Model B Rev 1.4
	
**Version**
2.3 - 2425890 - master
Edition
Plus edition (feat. Spotify integration)