MiczFlor/RPi-Jukebox-RFID #1017: 🐛 | Unable to play cards via swiping after upgrade from 1.1.9 to 2.0

tobiasnix, 2020-06-17:
## Bug

Its not possible to use cards to play music. 

### What I did

I upgraded from 1.1.9-rc to a 2.0 - ed888da - master Classic edition (barebones). A backup of the phoniebox folder was done and restored after buster installation was finished.

### What happened

Its not possible to play any song via card swiping. There is always an error like `...[: ==: unary operator expected...` in the logs.
It is possible to start playback via script eg `sudo ./rfid_trigger_play.sh -i=0012805484`.

### I expected this to happen

Playback of cards via swipinig.

### Further information that might help

Logs from swiping two different cards:

```
# tail -f /var/log/syslog
...
Jun 17 18:12:57 raspberrypi daemon_rfid_reader.py[474]: 2020-06-17 18:12:57,200 - root - INFO - Trigger Play Cardid=0012805484
Jun 17 18:12:57 raspberrypi daemon_rfid_reader.py[474]: /home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh: line 76: [: ==: unary operator expected
Jun 17 18:12:57 raspberrypi daemon_rfid_reader.py[474]: volume: n/a   repeat: off   random: off   single: off   consume: off
Jun 17 18:12:57 raspberrypi daemon_rfid_reader.py[474]: volume: n/a   repeat: off   random: off   single: off   consume: off
Jun 17 18:12:57 raspberrypi daemon_rfid_reader.py[474]: loading: Kinderlieder % Hoppe-hoppe-Reiter
Jun 17 18:12:57 raspberrypi daemon_rfid_reader.py[474]: /home/pi/RPi-Jukebox-RFID/scripts/single_play.sh: line 71: [: ==: unary operator expected
Jun 17 18:12:57 raspberrypi daemon_rfid_reader.py[474]: volume: n/a   repeat: off   random: off   single: off   consume: off
Jun 17 18:12:57 raspberrypi daemon_rfid_reader.py[474]: volume: n/a   repeat: off   random: off   single: off   consume: off
Jun 17 18:12:58 raspberrypi daemon_rfid_reader.py[474]: /home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh: line 109: [: ==: unary operator expected
Jun 17 18:12:58 raspberrypi daemon_rfid_reader.py[474]: OK MPD 0.21.4
Jun 17 18:12:58 raspberrypi daemon_rfid_reader.py[474]: OK
# ...
# another card
# ...
Jun 17 18:18:20 raspberrypi daemon_rfid_reader.py[474]: 2020-06-17 18:18:20,000 - root - INFO - Trigger Play Cardid=0007804907
Jun 17 18:18:20 raspberrypi daemon_rfid_reader.py[474]: /home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh: line 76: [: ==: unary operator expected
Jun 17 18:18:20 raspberrypi daemon_rfid_reader.py[474]: volume: n/a   repeat: off   random: off   single: off   consume: off
Jun 17 18:18:20 raspberrypi daemon_rfid_reader.py[474]: volume: n/a   repeat: off   random: off   single: off   consume: off
Jun 17 18:18:20 raspberrypi daemon_rfid_reader.py[474]: loading: Die_30_besten_Spiel_und_Bewegungslieder
Jun 17 18:18:20 raspberrypi daemon_rfid_reader.py[474]: /home/pi/RPi-Jukebox-RFID/scripts/single_play.sh: line 71: [: ==: unary operator expected
Jun 17 18:18:20 raspberrypi daemon_rfid_reader.py[474]: volume: n/a   repeat: off   random: off   single: off   consume: off
Jun 17 18:18:20 raspberrypi daemon_rfid_reader.py[474]: volume: n/a   repeat: off   random: off   single: off   consume: off
Jun 17 18:18:20 raspberrypi daemon_rfid_reader.py[474]: /home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh: line 109: [: ==: unary operator expected
Jun 17 18:18:20 raspberrypi daemon_rfid_reader.py[474]: OK MPD 0.21.4
Jun 17 18:18:20 raspberrypi daemon_rfid_reader.py[474]: OK

# journalctl -u mopidy
-- Logs begin at Thu 2019-02-14 10:11:59 GMT, end at Wed 2020-06-17 18:17:15 BST. --
-- No entries --
```
Starting playback manually

```
# starting manually 

# sudo ./rfid_trigger_play.sh -i=0012805484
/home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh: line 76: [: ==: unary operator expected
volume: n/a   repeat: off   random: off   single: off   consume: off
volume: n/a   repeat: off   random: off   single: off   consume: off
loading: Kinderlieder % Hoppe-hoppe-Reiter
/home/pi/RPi-Jukebox-RFID/scripts/single_play.sh: line 71: [: ==: unary operator expected
volume: n/a   repeat: off   random: off   single: off   consume: off
volume: n/a   repeat: off   random: off   single: off   consume: off
/home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh: line 109: [: ==: unary operator expected
OK MPD 0.21.4
OK
```

## Software

### Base image and version

`2020-05-27-raspios-buster-lite-armhf.zip`

### Branch / Release
`2.0 - ed888da - master Classic edition (barebones)`

### Installscript

`cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh`

## Hardware

### RaspberryPi version

`Raspberry Pi 2 Model B Rev 1.1`

s-martin, 2020-06-18:
Just to understand you correctly, if you manually invoke `sudo ./rfid_trigger_play.sh -i=0012805484` it works, despite the similar errors in the log?

tobiasnix, 2020-06-18:
I am confused. Its not working in both cases today. Here is a fresh log from startup, playing a file via swiping and several tries playing via webui.

```
pi@raspberrypi:~ $ tail -n 500 /var/log/syslog
Jun 18 09:43:48 raspberrypi kernel: [    0.280153] io scheduler noop registered
Jun 18 09:43:48 raspberrypi kernel: [    0.280189] io scheduler deadline registered (default)
Jun 18 09:43:48 raspberrypi kernel: [    0.280508] io scheduler cfq registered
Jun 18 09:43:48 raspberrypi kernel: [    0.280543] io scheduler mq-deadline registered (default)
Jun 18 09:43:48 raspberrypi kernel: [    0.280572] io scheduler kyber registered
Jun 18 09:43:48 raspberrypi kernel: [    0.283123] bcm2708_fb soc:fb: FB found 1 display(s)
Jun 18 09:43:48 raspberrypi kernel: [    0.293547] Console: switching to colour frame buffer device 82x26
Jun 18 09:43:48 raspberrypi kernel: [    0.301397] bcm2708_fb soc:fb: Registered framebuffer for display 0, size 656x416
Jun 18 09:43:48 raspberrypi kernel: [    0.309492] bcm2835-rng 3f104000.rng: hwrng registered
Jun 18 09:43:48 raspberrypi kernel: [    0.312561] vc-mem: phys_addr:0x00000000 mem_base=0x3ec00000 mem_size:0x40000000(1024 MiB)
Jun 18 09:43:48 raspberrypi kernel: [    0.318850] vc-sm: Videocore shared memory driver
Jun 18 09:43:48 raspberrypi kernel: [    0.322094] gpiomem-bcm2835 3f200000.gpiomem: Initialised: Registers at 0x3f200000
Jun 18 09:43:48 raspberrypi kernel: [    0.343620] brd: module loaded
Jun 18 09:43:48 raspberrypi kernel: [    0.360874] loop: module loaded
Jun 18 09:43:48 raspberrypi kernel: [    0.364488] Loading iSCSI transport class v2.0-870.
Jun 18 09:43:48 raspberrypi kernel: [    0.368365] libphy: Fixed MDIO Bus: probed
Jun 18 09:43:48 raspberrypi kernel: [    0.371338] usbcore: registered new interface driver lan78xx
Jun 18 09:43:48 raspberrypi kernel: [    0.374152] usbcore: registered new interface driver smsc95xx
Jun 18 09:43:48 raspberrypi kernel: [    0.376827] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
Jun 18 09:43:48 raspberrypi kernel: [    0.407519] dwc_otg 3f980000.usb: base=(ptrval)
Jun 18 09:43:48 raspberrypi kernel: [    0.610652] Core Release: 2.80a
Jun 18 09:43:48 raspberrypi kernel: [    0.613356] Setting default values for core params
Jun 18 09:43:48 raspberrypi kernel: [    0.616135] Finished setting default values for core params
Jun 18 09:43:48 raspberrypi kernel: [    0.819333] Using Buffer DMA mode
Jun 18 09:43:48 raspberrypi kernel: [    0.822049] Periodic Transfer Interrupt Enhancement - disabled
Jun 18 09:43:48 raspberrypi kernel: [    0.824799] Multiprocessor Interrupt Enhancement - disabled
Jun 18 09:43:48 raspberrypi kernel: [    0.827485] OTG VER PARAM: 0, OTG VER FLAG: 0
Jun 18 09:43:48 raspberrypi kernel: [    0.830173] Dedicated Tx FIFOs mode
Jun 18 09:43:48 raspberrypi kernel: [    0.833399] WARN::dwc_otg_hcd_init:1074: FIQ DMA bounce buffers: virt = bad14000 dma = 0xfad14000 len=9024
Jun 18 09:43:48 raspberrypi kernel: [    0.838649] FIQ FSM acceleration enabled for :
Jun 18 09:43:48 raspberrypi kernel: [    0.838649] Non-periodic Split Transactions
Jun 18 09:43:48 raspberrypi kernel: [    0.838649] Periodic Split Transactions
Jun 18 09:43:48 raspberrypi kernel: [    0.838649] High-Speed Isochronous Endpoints
Jun 18 09:43:48 raspberrypi kernel: [    0.838649] Interrupt/Control Split Transaction hack enabled
Jun 18 09:43:48 raspberrypi kernel: [    0.850992] dwc_otg: Microframe scheduler enabled
Jun 18 09:43:48 raspberrypi kernel: [    0.851081] WARN::hcd_init_fiq:457: FIQ on core 1
Jun 18 09:43:48 raspberrypi kernel: [    0.853609] WARN::hcd_init_fiq:458: FIQ ASM at 8068fda0 length 36
Jun 18 09:43:48 raspberrypi kernel: [    0.856116] WARN::hcd_init_fiq:497: MPHI regs_base at bb810000
Jun 18 09:43:48 raspberrypi kernel: [    0.858677] dwc_otg 3f980000.usb: DWC OTG Controller
Jun 18 09:43:48 raspberrypi kernel: [    0.861279] dwc_otg 3f980000.usb: new USB bus registered, assigned bus number 1
Jun 18 09:43:48 raspberrypi kernel: [    0.863925] dwc_otg 3f980000.usb: irq 56, io mem 0x00000000
Jun 18 09:43:48 raspberrypi kernel: [    0.866531] Init: Port Power? op_state=1
Jun 18 09:43:48 raspberrypi kernel: [    0.869122] Init: Power Port (0)
Jun 18 09:43:48 raspberrypi kernel: [    0.871952] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 4.19
Jun 18 09:43:48 raspberrypi kernel: [    0.877046] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
Jun 18 09:43:48 raspberrypi kernel: [    0.879812] usb usb1: Product: DWC OTG Controller
Jun 18 09:43:48 raspberrypi kernel: [    0.882492] usb usb1: Manufacturer: Linux 4.19.118-v7+ dwc_otg_hcd
Jun 18 09:43:48 raspberrypi kernel: [    0.885234] usb usb1: SerialNumber: 3f980000.usb
Jun 18 09:43:48 raspberrypi kernel: [    0.888805] hub 1-0:1.0: USB hub found
Jun 18 09:43:48 raspberrypi kernel: [    0.891419] hub 1-0:1.0: 1 port detected
Jun 18 09:43:48 raspberrypi kernel: [    0.894701] dwc_otg: FIQ enabled
Jun 18 09:43:48 raspberrypi kernel: [    0.894710] dwc_otg: NAK holdoff enabled
Jun 18 09:43:48 raspberrypi kernel: [    0.894716] dwc_otg: FIQ split-transaction FSM enabled
Jun 18 09:43:48 raspberrypi kernel: [    0.894731] Module dwc_common_port init
Jun 18 09:43:48 raspberrypi kernel: [    0.895143] usbcore: registered new interface driver usb-storage
Jun 18 09:43:48 raspberrypi kernel: [    0.898035] mousedev: PS/2 mouse device common for all mice
[... cut ...]
Jun 18 09:46:12 raspberrypi lighttpd[748]: /home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh: line 109: [: ==: unary operator expected
Jun 18 09:46:14 raspberrypi lighttpd[748]: /home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh: line 76: [: ==: unary operator expected
Jun 18 09:46:14 raspberrypi lighttpd[748]: /home/pi/RPi-Jukebox-RFID/scripts/single_play.sh: line 71: [: ==: unary operator expected
Jun 18 09:46:15 raspberrypi lighttpd[748]: /home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh: line 109: [: ==: unary operator expected
pi@raspberrypi:~ $ 

```

MiczFlor, 2020-06-18:
Hi @tobiasnix 
if it works now (sometimes there is magic, we just have to accept that), can we close the ticket?

tobiasnix, 2020-06-18:
Error was still present. I did a fresh install and reassigned the cards. Everything is working now.

closed 2020-06-18 by tobiasnix

varac, 2020-10-26:
I had  the same, and the problem was that the `folder.conf` files in each music dir was missing a variable only present in 2.0. That's why the `if` statement failed. I removed all old  `folder.conf` files and it worked afterwards.