MiczFlor/RPi-Jukebox-RFID #1018: Question: Spotify Edition and Raspberry Zero W unusable?

MoJo2600, 2020-06-18:
Hi,

I installed the spotify edition on a Raspberry Zero W and it seems that the Zero W is under powered for this task. The music stops every few seconds for a short time. Local mp3 files play fine. I did a lot of tests, because i did want to make it work on the Zero W but it just seems to not work with the combination of spotify.

What i tried:

- Zero W + Spotify + RC522 -> stutters
- Zero W + RC522 -> works perfectly
- 3b + Spotify + RC522 -> works perfectly
- 3b + RC522 -> works perfectly

I tried the stable version and develop version on raspberry pi os with a class 10 sd card. Power management of the wifi interface is off. Maybe there should be a warning not to try spotify on Zero W - or am I alone with the issues and it is working but i did something wrong?

s-martin, 2020-06-18:
I didn't find it yet, but another user reported in one of the recent issues also problems with Zero and Spotify.

MoJo2600, 2020-06-18:
Yes, I recorded a video for you. Every peak at 100% CPU leads to a short break in playback. So i assume it is an issue with mopidy.
[https://imgur.com/a/5TTrwqd](https://imgur.com/a/5TTrwqd)

MoJo2600, 2020-06-18:
So, I did a test with mopidy on a fresh Raspberry OS installation. The load spikes are there as well, but they do not exceed 60% the sound is not stopping. I think the base load of phoniebox is too high to work smoothly with mopidy on a Raspberry Zero.

MoJo2600, 2020-06-19:
Ok, I did some more testing and will close this question. If you do not use the WebUI of phoniebox and Iris, it works better. If you open the WebUI it starts to stutter on the raspbery Pi Zero.

closed 2020-06-19 by MoJo2600