MiczFlor/RPi-Jukebox-RFID #1020: RPi-Jukebox - unable to set volume

Fips03, 2020-06-21:
Hi everybody, 

I am rather on the user side of things, but able to think myself into technical stuff, so setting up a Phoniebox worked out well. Everything seems to be functional - except volume control. 

The web interface does not show the initial volume although startup_volume is set to 65.
When I manually change the volume in the interface, it actually shows the new value in the now blueish circle, but neither the audible volume not the PCM value does change. 
Changing the volume via programmed RFID cards doesn't work, either. 

What can I do? What can I check? What can I change?

Thanks in advance for your help,
Phillip

s-martin, 2020-06-21:
Hi, welcome to the Phoniebox community.

Please check issue #973, seems to be a similar issue.

If it's the same issue, please continue there and close this Issue here.

Fips03, 2020-06-21:
Issue solved. Thanks for guidance!

closed 2020-06-21 by Fips03