MiczFlor/RPi-Jukebox-RFID #1023: no sound from usb sound card

IZooRz, 2020-06-23:
Dear all thank you for your exceptional work in this project. 

I just installed version 2.0 via the one line install script and everything seems to work except the sound from the usb sound card.

I already tried:
I installed version 2.0 and everything seems to run except my usb sound card
I tried all hints from https://learn.adafruit.com/usb-audio-cards-with-a-raspberry-pi/testing-audio but the sound card does not appear when trying the commands dsmeg ...

then I change sound usb from 2 to 0 as explained in the install section but it still does not work

Does anyone have an idea where to fix?

Thanks in advance!

IZooRz

s-martin, 2020-06-23:
Please check #973 as Raspberry Pi changed the audio behavior recently.

IZooRz, 2020-06-23:
> Please check #973 as Raspberry Pi changed the audio behavior recently.

Thanks s-martin. Honestly i am a litte lost in the #973. Do you have a hint what to try exactly?

Thank you!

IZooRz, 2020-06-23:
I just tried https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working and have now a starting up sound ;-)

The problem was, that in in the example one would have to change the output to 1 but my usb card has the 0 and now I have change that and it works!!! 

Thank you very very much for the impressive project.

IZooRz

IZooRz, 2020-06-23:
The Box is running

closed 2020-06-23 by IZooRz