MiczFlor/RPi-Jukebox-RFID #1024: 🐛 | Startup Volume not working

princemaxwell, 2020-06-23:
## Bug

### What I did

I set the startup volume to 5 in the settings, but after restart phoniebox starts with 30.

### Further information that might help

Link: https://askubuntu.com/questions/1072165/systemd-how-to-start-service-b-after-service-a-runs-to-completion

While installation of phoniebox, the startup-service will be installed.
`[Unit]
Description=Phoniebox Startup
After=mpd.service

[Service]
User=pi
Group=pi
Type=oneshot
RemainAfterExit=true
WorkingDirectory=/home/pi/RPi-Jukebox-RFID
ExecStart=/bin/bash /home/pi/RPi-Jukebox-RFID/scripts/startup-scripts.sh

[Install]
WantedBy=multi-user.target`

The "After"-command seems to be not working. So "startup-scripts.sh" is being executed BEFORE mpd.service is started.

`/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=setvolumetostartup

####################
# play startup sound
# after some sleep
/bin/sleep 2
/usr/bin/mpg123 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3`

The "startup-scripts.sh" is executing "playout_controls.sh". I tried to put a "sleep 60" command before this, but the sleep command is being ignored completely. Don't know why...

My solution would be to put a sleep command into "playout_controls.sh", line 356:
`    setvolumetostartup)
        if [ "${DEBUG_playout_controls_sh}" == "TRUE" ]; then echo "   ${COMMAND}" >> ${PATHDATA}/../logs/debug.log; fi
        # check if startup-volume is disabled
        if [ "${AUDIOVOLSTARTUP}" == 0 ]; then
            exit 1
        else
            # set volume level in percent
			sleep 60
            echo -e setvol ${AUDIOVOLSTARTUP}\\nclose | nc -w 1 localhost 6600
        fi
        ;;`

The box is waiting 60 seconds before the startup-volume is being set. This is enough time for mpd.service to start.
Another good effect coming with this solution is, that the startup sound appears AFTER that, so the user knows that the services are loaded...

Can please someone confirm this?

## Software

### Base image and version

Raspbian GNU/Linux 10 (buster)

### Branch / Release

master

## Hardware

### RaspberryPi version

Raspberry Pi 3 Model B Rev 1.2

### RFID Reader

13ba:0018 PCPlay Barcode PCP-BCG4209

### Soundcard

08bb:2902 Texas Instruments PCM2902 Audio Codec

s-martin, 2020-06-24:
Additional 60secs during boot sounds quite long until the box is ready.

princemaxwell, 2020-06-24:
You're right, this is only a quick and dirty fix.
The problem is, that mopidy.SERVICE (Spotify) is starting very quick. But while the service is already running, mopidy is establishing the SERVER connection to localhost:6600 in the background. This need 45-60 seconds, depending on how much playlists the user has.

Only when the server is running, you can fire commands.

By the way, in the information tab you can still see, the mopidy server status. But this is only the status of the service.

I am working on a while-loop insteatd of "sleep 60" to check the localhost status and hold on until the server is up before firing commands and startupsound. This has the charme, that spotify user know, that they can swipe the card. So far, many user have issues saying spotify is not working, but i think they are swiping cards before server is up.

princemaxwell, 2020-06-24:
I fixed this bug, will make a pull request.

PR #1026 (2020-06-28): Wait until server is running
The problem is, that mopidy.SERVICE (Spotify) is starting very quick. But while the SERVICE is already running, mopidy is establishing the SERVER connection to localhost:6600 in the background. This needs ~ 30-60 seconds, depending on how much playlists the user has.

Only when the server is running, you can give commands (volume, starting songs from spotify, ...)

- I added seperate informations about service and server to the info-tab.
- Adding a while-loop for checking the mopidy/MPD server status in the startup-script.

Fixes #1024
closed 2020-06-24 by princemaxwell