MiczFlor/RPi-Jukebox-RFID #1029: 🐛 | Audio Play  with RFID only possible after manual update of mpc

matzeglatze, 2020-06-26:
## Bug

Hi @all
I am new with Phonie Box and I learned a lot  the last days.
 I installed the raspberry pi with the Phonie Software

 What happened:

new some Folders in the app is working
The upload of files with the app is working 
also the Registration of the RFID to the Media.

Testing the registered Card Shows the cover but without Music playing

I have to `Update mpc` using the console (Putty)
 after that the RFID starts the files.

any idea what i did wrong . Maybe wrong Setup of a config file?

The `update  mpc` is Always necessary when i upload new files and Folders.

at the Raspberry Forum they say i have to push the button for "scan Music Library", 
but i dont have this button at the WebApp

sorry for my bad english ( i swear my german is better)

[Raspberry 4 with USB Soundcard spotyfy Edition one line installer ]

kawuppdich, 2020-06-26:
Hi matzeglatze,

seems I have same issue, installed Rasperry Buster via raspberry Pi imager and then from one line script the Phonibox stuff

I am not able to play anything via Phonibox, there is no sound. When I try using spotify via Iris the same. have you tried via Iris?
Does not matter if local mp3 or youtube or spotify via iris

In general sound is working like youtube via browser, connected direct to pi via VNC

[Raspberry 3B, Buster, Sound via PCM]
two boxes on stretch running fine

s-martin, 2020-06-26:
Hi @matzeglatze, @kawuppdich, did you check the hints in https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working ?

s-martin, 2020-06-26:
> Testing the registered Card Shows the cover but without Music playing
> I have to `Update mpc` using the console (Putty)
> after that the RFID starts the files.
> The `update mpc` is Always necessary when i upload new files and Folders.

After you added folder you have to update mpc, that’s sort of expected.

> 
> at the Raspberry Forum they say i have to push the button for "scan Music Library",
> but i dont have this button at the WebApp

This button is only available, if you installed the Spotify edition, but you are right, that should be available for classic edition, too. What do you think @MiczFlor?

matzeglatze, 2020-06-28:
ok thank you ...i seams i did not install the Spotify Edition ….
[image]
 in that case:

> This button is only available, if you installed the Spotify edition, but you are right, that should be available for classic edition, t...

again THANK YOU for help

kawuppdich, 2020-06-29:
> 
> 
> Hi @matzeglatze, @kawuppdich, did you check the hints in https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working ?

OK different topic to mine, nevertheless I tried changing all relevant conf files as discribed in link and acording mpd doc but no success, will try a clean installation from formatted sd card...
Thanks so far

closed 2020-10-15 by s-martin