MiczFlor/RPi-Jukebox-RFID #1031: 🐛 | mpd / seekcur -15 is not working

PsychoD, 2020-06-29:
## Bug

### What I did

Installed the buster version with one line script, configured gpio buttons: 

def def_next():
    check_call("./scripts/playout_controls.sh -c=playerseek -v=+15", shell=True)

def def_prev():
    check_call("./scripts/playout_controls.sh -c=playerseek -v=-15", shell=True)

### What happened
The next button works as expected, the prev button does not. You can hear a little skip, but no jumping back in the track. 

### I expected this to happen

Pressing prev jumps -15 sec in the current track

### Further information that might help

I debugged this down to mpd: 
`echo seekcur 15 | nc -w 1 localhost 6600
echo status | nc -w 1 localhost 6600 | grep -o -P '(?<=elapsed: ).*'
echo seekcur +15 | nc -w 1 localhost 6600
echo status | nc -w 1 localhost 6600 | grep -o -P '(?<=elapsed: ).*'
echo seekcur -15 | nc -w 1 localhost 6600
echo status | nc -w 1 localhost 6600 | grep -o -P '(?<=elapsed: ).*'`

results in: 
`OK MPD 0.21.4
OK
16.782
OK MPD 0.21.4
OK
34.570
OK MPD 0.21.4
OK
37.358`

Also, I wonder why it says version 0.21.4, while it should say 0.21.5, right? 
` 
pi@phoniebox:~ $ which mpd
/usr/bin/mpd
pi@phoniebox:~ $ mpd --version
Music Player Daemon 0.21.5 (0.21.5)`

## Software

### Base image and version
cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release

master

### Installscript
I used the latest buster one line install script to update my installation from october 2019 (where this bug already was present)

## Hardware

### RaspberryPi version

Raspberry Pi 2 Model B Rev 1.1

### Soundcard
Hifiberry MiniAmp

pinguin42, 2020-07-14:
I have the same problem here. My phoniebox doesn't rewind either.

PsychoD, 2020-07-15:
Does anyone know where I can specify / look up which MPD the phoniebox uses? It seems my system wide installed version is not used.

s-martin, 2020-07-31:
> Does anyone know where I can specify / look up which MPD the phoniebox uses? It seems my system wide installed version is not used.

You could search for multiple `mpd` binaries...

s-martin, 2020-08-16:
Please check https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/878#issuecomment-672283454

MiczFlor, 2020-08-26:
Hi @PsychoD 
I added your fix to the develop branch. You can see it here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/d6408b4d574a1f0f216c4fd5f6d283ce57569800
Could you check if it works? I am only on my laptop. Reading this thread, I thought: why not just seek to the absolute position if the *minus* creates... and before I could finish that thought: here you posted it :) Thanks
If it works, please close the ticket.

PsychoD, 2020-08-28:
Hi, thanks for including the fix! 
I currently have some issues updating my version so I can't test the development branch quickly, but will as soon as I manage to get the update running. From the code it looks exactly like my playout-controls (I only am lacking the whole "VOLFILE" part, but I guess that has nothing to do with the fix). Also I used awk, and am not 100% sure if this dependency is already met by any raspberry pi os.

closed 2020-10-15 by s-martin