MiczFlor/RPi-Jukebox-RFID #1032: 🐛 | Oneline install script does not keep rotary-encoder script

PsychoD, 2020-06-30:
## Bug

### What I did
I upgraded my installation using the one-line-installer script. During installation I said "yes" when I was asked if I wanted to keep my GPIO configuration. 

### What happened
The upgrade was performed, but the rotary-encoder scripts were missing afterwards. 

### I expected this to happen
Rotary controller works as it did before. 

### Further information that might help
Reviewing the script revealed that rotary controller is not regarded.

s-martin, 2020-06-30:
Yes, you are right. The new `components/gpio_control` covers the rotary encoder now, but it’s not correctly integrated yet and also not really documented.

Please see #991.

PsychoD, 2020-06-30:
Thanks! I found the new gpio settings file, but could not make any sense from it. Now it is clearer.

closed 2020-06-30 by PsychoD