MiczFlor/RPi-Jukebox-RFID #1033: 🐛 | Double occupancy of RFID ID's. Bug or expected behaviour?

themorlan, 2020-07-02:
## Bug

### What I did

I linked the system command "Volume up by one step" to an RFID-Tag via the WebUI. Afterwards I wanted to links the same ID to a stream URL. 

### What happened

Afterwards swiping the tag does not lead to stream being played.

### I expected this to happen

I expected the system command "Volume up by one step" to be removed from the ID and the ID assigned to the stream URL.
These were my expectations. Not sure if the double occupancy is intended.

### Further information that might help

It seems the ID is not being removed from the system command. My `~/RPi-Jukebox-RFID/settings/rfid_trigger_play.conf` shows:
```
## Audio player
# The following commands control audio playout
# next and prev play the previous or next track in the playlist (== folder & spotify)
### Volume up by one step
CMDVOLUP="0005829700"
```

and the content of `~/RPi-Jukebox-RFID/shared/shortcuts/0005829700` is:
`DLF_Der_Tag`

When the tag is swiped the output of debug.log is:
``` ########### SCRIPT rfid_trigger_play.sh (2020-07-02.15:26:21) ##
Card ID '0005829700' was used
########### SCRIPT playout_controls.sh (2020-07-02.15:26:21) ##
VAR COMMAND: volumeup
VAR VALUE:
   volumeup
# Attempting to play: /home/pi/RPi-Jukebox-RFID/shared/audiofolders/
# Type of play $VALUE:
Path not found /home/pi/RPi-Jukebox-RFID/shared/audiofolders/ 
```
### Base image and version
`2020-05-27-raspios-buster-lite-armhf.img`
### Branch / Release
 `master`
### Installscript
`scripts/installscripts/buster-install-default.sh`
### RaspberryPi version
`3 B+`

themorlan, 2020-07-02:
Just saw #1013 , more or less the same issue

s-martin, 2020-07-02:
Please close this issue, if you think it's the same.

closed 2020-07-02 by themorlan